me-no-dev/ESPAsyncWebServer #1122: Support for FATFS filesystem

Vishnu-Karthikeyan, 2022-02-20:
Is there any way that we can use ESPAsyncWebServer with FATFS filesystem.
Trying to call images or files from flash(FATFS) using "server.on  request->send" does not work as there is no support for it.

zekageri, 2022-02-21:
I don't know about that but LITTLEFS is faster. ^^

Vishnu-Karthikeyan, 2022-02-22:
What if the Flash size is 16MB. We do not have an option for SPIFFS/ LITTLEFS.
.
.
.
.
.
[image]

zekageri, 2022-02-22:
In theory you can use any file system with async webserver, it is a little weird that you can not use it. 

You can build your own partition.csv and you can configure any size for the file system you want. 

I don't know how to import this to Arduino but on platform IO it is fairly easy

Vishnu-Karthikeyan, 2022-03-08:
This method works for me 
       File file = FFat.open("/abc.txt");
       request->send(file, "/abc.txt", "text/plain", true);  //downloading
       request->send(file, "/abc.txt", "text/plain");          // to load
       file.close();

Thank you :)

closed 2022-03-08 by Vishnu-Karthikeyan