me-no-dev/ESPAsyncWebServer #1131: Help Please Sending BMP Fails

ispybadguys, 2022-03-17:
This is probably my own total lack of understanding but I don't get an image when I use: 

server.on("/thermal", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(200,"image/bmp",image);
  });

when image is : char image[2358]; and image is filled with the bytes of a BMP. The BMP is exactly 2358 bytes long and there is no terminator. There is not a zero terminator in the array image if that is needed but then again there are a lot of zeros in image before the end.

I do get a image when I write the image to SPIFFS and use:
server.on("/thermal", HTTP_GET, [](AsyncWebServerRequest *request){
  request->send(SPIFFS, "/thermal.bmp", "image/bmp", false);
 });

Thanks

Kurt

NC700Rider, 2022-03-24:
Hi Kurt,
I do not understand why you need to send back an image. I have some images in my ESP SPIFFS that I upload into the Web page ESP serve but I do not use an end point to send back the image as a response.
Please, could you explain why you need to response back with the image?
BR
Juan

ispybadguys, 2022-03-24:
I have some code that reads a thermal camera and writes it to SPIFFS and then has the client request the file. The problem is that if I write to SPIFFS at 2 frames per second it will wear out the SPIFFS memory so I wanted to same the image in RAM and not use SPIFFS. The SPIFFS is setup with:

server.on("/thermal", HTTP_GET, [](AsyncWebServerRequest *request){       // Thermal.bmp
    request->send(SD, "/thermal.bmp", "image/bmp", false);
  });

The web page contain the following to refresh the image.
setInterval(function ( ) {
  var xhttp = new XMLHttpRequest();
  xhttp.onreadystatechange = function() {
    //if (this.readyState == 4 && this.status == 200) {
      var buffer = document.getElementById("thermalbuffer");
      //buffer.src = this.response;
      buffer.src = "thermal";
      //document.getElementById("thermal") = this.response;
    //}

I must confess that I understand almost nothing about HTML.

As a work around I wrote the image to an SD card. At least I can replace that and it is bigger so it will wear out slower.

Any help is appreciated.

Thanks

Kurt

ispybadguys, 2022-04-08:
I guess I didn't understand your response until I re-read it. I want to send the file from memory instead of SPIFFS.

NC700Rider, 2022-04-08:
Hi Kurt,

I think now we understand the problem. I ave not implemented a use case like that in my project. But I have a look in ESPAsyncWebServer docuumentation and looks there is a use case to send back a binary content stored in PROGEM. This may force you to store the variable where you store the thermal image in PROGEM rather than SRAM. This is the use case in GitHub:

//File: favicon.ico.gz, Size: 726
#define favicon_ico_gz_len 726
const uint8_t favicon_ico_gz[] PROGMEM = {
 0x1F, 0x8B, 0x08, 0x08, 0x0B, 0x87, 0x90, 0x57, 0x00, 0x03, 0x66, 0x61, 0x76, 0x69, 0x63, 0x6F,
 0x6E, 0x2E, 0x69, 0x63, 0x6F, 0x00, 0xCD, 0x53, 0x5F, 0x48, 0x9A, 0x51, 0x14, 0xBF, 0x62, 0x6D,
 0x86, 0x96, 0xA9, 0x64, 0xD3, 0xFE, 0xA8, 0x99, 0x65, 0x1A, 0xB4, 0x8A, 0xA8, 0x51, 0x54, 0x23,
 0xA8, 0x11, 0x49, 0x51, 0x8A, 0x34, 0x62, 0x93, 0x85, 0x31, 0x58, 0x44, 0x12, 0x45, 0x2D, 0x58,
 0xF5, 0x52, 0x41, 0x10, 0x23, 0x82, 0xA0, 0x20, 0x98, 0x2F, 0xC1, 0x26, 0xED, 0xA1, 0x20, 0x89,
 0x04, 0xD7, 0x83, 0x58, 0x20, 0x28, 0x04, 0xAB, 0xD1, 0x9B, 0x8C, 0xE5, 0xC3, 0x60, 0x32, 0x64,
 0x0E, 0x56, 0xBF, 0x9D, 0xEF, 0xF6, 0x30, 0x82, 0xED, 0xAD, 0x87, 0xDD, 0x8F, 0xF3, 0xDD, 0x8F,
 0x73, 0xCF, 0xEF, 0x9C, 0xDF, 0x39, 0xBF, 0xFB, 0x31, 0x26, 0xA2, 0x27, 0x37, 0x97, 0xD1, 0x5B,
 0xCF, 0x9E, 0x67, 0x30, 0xA6, 0x66, 0x8C, 0x99, 0xC9, 0xC8, 0x45, 0x9E, 0x6B, 0x3F, 0x5F, 0x74,
 0xA6, 0x94, 0x5E, 0xDB, 0xFF, 0xB2, 0xE6, 0xE7, 0xE7, 0xF9, 0xDE, 0xD6, 0xD6, 0x96, 0xDB, 0xD8,
 0xD8, 0x78, 0xBF, 0xA1, 0xA1, 0xC1, 0xDA, 0xDC, 0xDC, 0x2C, 0xEB, 0xED, 0xED, 0x15, 0x9B, 0xCD,
 0xE6, 0x4A, 0x83, 0xC1, 0xE0, 0x2E, 0x29, 0x29, 0x99, 0xD6, 0x6A, 0xB5, 0x4F, 0x75, 0x3A, 0x9D,
 0x61, 0x75, 0x75, 0x95, 0xB5, 0xB7, 0xB7, 0xDF, 0xC8, 0xD1, 0xD4, 0xD4, 0xF4, 0xB0, 0xBA, 0xBA,
 0xFA, 0x83, 0xD5, 0x6A, 0xFD, 0x5A, 0x5E, 0x5E, 0x9E, 0x28, 0x2D, 0x2D, 0x0D, 0x10, 0xC6, 0x4B,
 0x98, 0x78, 0x5E, 0x5E, 0xDE, 0x95, 0x42, 0xA1, 0x40, 0x4E, 0x4E, 0xCE, 0x65, 0x76, 0x76, 0xF6,
 0x47, 0xB5, 0x5A, 0x6D, 0x4F, 0x26, 0x93, 0xA2, 0xD6, 0xD6, 0x56, 0x8E, 0x6D, 0x69, 0x69, 0xD1,
 0x11, 0x36, 0x62, 0xB1, 0x58, 0x60, 0x32, 0x99, 0xA0, 0xD7, 0xEB, 0x51, 0x58, 0x58, 0x88, 0xFC,
 0xFC, 0x7C, 0x10, 0x16, 0x02, 0x56, 0x2E, 0x97, 0x43, 0x2A, 0x95, 0x42, 0x2C, 0x16, 0x23, 0x33,
 0x33, 0x33, 0xAE, 0x52, 0xA9, 0x1E, 0x64, 0x65, 0x65, 0x71, 0x7C, 0x7D, 0x7D, 0xBD, 0x93, 0xEA,
 0xFE, 0x30, 0x1A, 0x8D, 0xE8, 0xEC, 0xEC, 0xC4, 0xE2, 0xE2, 0x22, 0x6A, 0x6A, 0x6A, 0x40, 0x39,
 0x41, 0xB5, 0x38, 0x4E, 0xC8, 0x33, 0x3C, 0x3C, 0x0C, 0x87, 0xC3, 0xC1, 0x6B, 0x54, 0x54, 0x54,
 0xBC, 0xE9, 0xEB, 0xEB, 0x93, 0x5F, 0x5C, 0x5C, 0x30, 0x8A, 0x9D, 0x2E, 0x2B, 0x2B, 0xBB, 0xA2,
 0x3E, 0x41, 0xBD, 0x21, 0x1E, 0x8F, 0x63, 0x6A, 0x6A, 0x0A, 0x81, 0x40, 0x00, 0x94, 0x1B, 0x3D,
 0x3D, 0x3D, 0x42, 0x3C, 0x96, 0x96, 0x96, 0x70, 0x7E, 0x7E, 0x8E, 0xE3, 0xE3, 0x63, 0xF8, 0xFD,
 0xFE, 0xB4, 0xD7, 0xEB, 0xF5, 0x8F, 0x8F, 0x8F, 0x5B, 0x68, 0x5E, 0x6F, 0x05, 0xCE, 0xB4, 0xE3,
 0xE8, 0xE8, 0x08, 0x27, 0x27, 0x27, 0xD8, 0xDF, 0xDF, 0xC7, 0xD9, 0xD9, 0x19, 0x6C, 0x36, 0x1B,
 0x36, 0x36, 0x36, 0x38, 0x9F, 0x85, 0x85, 0x05, 0xAC, 0xAF, 0xAF, 0x23, 0x1A, 0x8D, 0x22, 0x91,
 0x48, 0x20, 0x16, 0x8B, 0xFD, 0xDA, 0xDA, 0xDA, 0x7A, 0x41, 0x33, 0x7E, 0x57, 0x50, 0x50, 0x80,
 0x89, 0x89, 0x09, 0x84, 0xC3, 0x61, 0x6C, 0x6F, 0x6F, 0x23, 0x12, 0x89, 0xE0, 0xE0, 0xE0, 0x00,
 0x43, 0x43, 0x43, 0x58, 0x5E, 0x5E, 0xE6, 0x9C, 0x7D, 0x3E, 0x1F, 0x46, 0x47, 0x47, 0x79, 0xBE,
 0xBD, 0xBD, 0x3D, 0xE1, 0x3C, 0x1D, 0x0C, 0x06, 0x9F, 0x10, 0xB7, 0xC7, 0x84, 0x4F, 0xF6, 0xF7,
 0xF7, 0x63, 0x60, 0x60, 0x00, 0x83, 0x83, 0x83, 0x18, 0x19, 0x19, 0xC1, 0xDC, 0xDC, 0x1C, 0x8F,
 0x17, 0x7C, 0xA4, 0x27, 0xE7, 0x34, 0x39, 0x39, 0x89, 0x9D, 0x9D, 0x1D, 0x6E, 0x54, 0xE3, 0x13,
 0xE5, 0x34, 0x11, 0x37, 0x49, 0x51, 0x51, 0xD1, 0x4B, 0xA5, 0x52, 0xF9, 0x45, 0x26, 0x93, 0x5D,
 0x0A, 0xF3, 0x92, 0x48, 0x24, 0xA0, 0x6F, 0x14, 0x17, 0x17, 0xA3, 0xB6, 0xB6, 0x16, 0x5D, 0x5D,
 0x5D, 0x7C, 0x1E, 0xBB, 0xBB, 0xBB, 0x9C, 0xD7, 0xE1, 0xE1, 0x21, 0x42, 0xA1, 0xD0, 0x6B, 0xD2,
 0x45, 0x4C, 0x33, 0x12, 0x34, 0xCC, 0xA0, 0x19, 0x54, 0x92, 0x56, 0x0E, 0xD2, 0xD9, 0x43, 0xF8,
 0xCF, 0x82, 0x56, 0xC2, 0xDC, 0xEB, 0xEA, 0xEA, 0x38, 0x7E, 0x6C, 0x6C, 0x4C, 0xE0, 0xFE, 0x9D,
 0xB8, 0xBF, 0xA7, 0xFA, 0xAF, 0x56, 0x56, 0x56, 0xEE, 0x6D, 0x6E, 0x6E, 0xDE, 0xB8, 0x47, 0x55,
 0x55, 0x55, 0x6C, 0x66, 0x66, 0x46, 0x44, 0xDA, 0x3B, 0x34, 0x1A, 0x4D, 0x94, 0xB0, 0x3F, 0x09,
 0x7B, 0x45, 0xBD, 0xA5, 0x5D, 0x2E, 0x57, 0x8C, 0x7A, 0x73, 0xD9, 0xED, 0xF6, 0x3B, 0x84, 0xFF,
 0xE7, 0x7D, 0xA6, 0x3A, 0x2C, 0x95, 0x4A, 0xB1, 0x8E, 0x8E, 0x0E, 0x6D, 0x77, 0x77, 0xB7, 0xCD,
 0xE9, 0x74, 0x3E, 0x73, 0xBB, 0xDD, 0x8F, 0x3C, 0x1E, 0x8F, 0xE6, 0xF4, 0xF4, 0x94, 0xAD, 0xAD,
 0xAD, 0xDD, 0xDE, 0xCF, 0x73, 0x0B, 0x0B, 0xB8, 0xB6, 0xE0, 0x5D, 0xC6, 0x66, 0xC5, 0xE4, 0x10,
 0x4C, 0xF4, 0xF7, 0xD8, 0x59, 0xF2, 0x7F, 0xA3, 0xB8, 0xB4, 0xFC, 0x0F, 0xEE, 0x37, 0x70, 0xEC,
 0x16, 0x4A, 0x7E, 0x04, 0x00, 0x00
};

AsyncWebServerResponse *response = request->beginResponse_P(200, "image/x-icon", favicon_ico_gz, favicon_ico_gz_len);
response->addHeader("Content-Encoding", "gzip");
request->send(response);

BR
Juan

zekageri, 2022-04-08:
Or you could just stream the raw content with websockets and process every other stuff on the client side with JavaScript

ispybadguys, 2022-04-12:
I am reading data from a MLX90640 thermal camera and converting it to .bmp format. I can’t use a string because the zeros in the data are misinterpreted as the end of string.

RadianM, 2022-05-16:
You could trigger a _download_ of a .bmp file from a **char bitmap[bitmapsize]** array in SRAM with e.g...

 server.on("/image.bmp", HTTP_GET, [](AsyncWebServerRequest *request) {
    AsyncResponseStream *response = request->beginResponseStream("image/x-windows-bmp");
    response->addHeader("Content-Disposition", "inline; filename=image.bmp");
    response->write(bitmap, bitmapsize);
    request->send(response);
  });

At least it's a sanity check that the bitmap is valid. Not as convenient as having display directly in the browser though.

I still don't get why it can't be served directly out of SRAM. While the example given of sending the favicon from PROGMEM works, it doesn't get around the flash memory wear issue if the image is created dynamically. So it seems there's no way of dynamically creating an image and serving it 'as is'.

If we upload a bitmap image file to a typical LAMP server and point a browser at the corresponding URL it just renders. This is also true for an image file held in SPIFFS using ESPAsyncWebServer but surely it needn't _depend_ on being held on a file system if it's only small?

I've also used the technique of dynamically building small .bmp bitmaps in memory and serving _sans_ HTML on Electric Imps for visual analysis - a very useful visual debugging aid for certain projects. I've therefore struggled with this on ESP myself.

RadianM, 2022-05-16:
Or, (like me, seven hours later) finally make it work (I don't know why) by chunking the response as described here:

[https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/README.md#chunked-response](url)

This allows the prepared bitmap image in SRAM to be memcpy'd into the **buffer** provided by the callback, in **maxLen** chunks - even if the entire bitmap would easily fit in the basic response.

Implementation will be specific to requirements and can even allow the bitmap to be composed 'on the 'fly' but in my case I had the entire image sitting in a buffer so just kept moving a pointer through the buffer and used memcpy to transfer the chunks until all written.

It makes for a more general solution (read bigger images) and can also be accessed via a little PROGMEM static html to scale and position the bitmap as per:

[https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/README.md#send-large-webpage-from-progmem](url)

something like:
`const char index_html[] PROGMEM = "<center>[image]</center>";
`

closed 2022-11-02 by stale[bot]