me-no-dev/ESPAsyncWebServer #1138: Seeking help to clear previous route on page reload

dduehren, 2022-04-15:
I have a few forms where a button causes an action,  e.g. 
` <form action="/W_OK" method ="get" autocomplete="off">
        <p><label id= wifi>%WIFISTATE%</label><p>
        <p><label><a href="wCrd.html">Wifi</a><label></p><br>
        <p><label id = ssid>%SSID%<label><p><br>
        <p><label id = ipaddr>%IPADDR%</label><p>
        <p><button type="submit" class="submitBtn">Accept</button></p>
</form>`
When the submit button is pressed, the URL in the browser is 192.168.0.28\W_OK

And the route at /W_OK is processed like this:
`  server.on("/W_OK", HTTP_GET, [](AsyncWebServerRequest *request){
      wifiMgr.useCredentials();
      request->send(LittleFS, FPSTR(string_indexHTML), String(),false, processor);
      });`

The problem I have is that when the index HTML is loaded with the request->send, the URL is still 192.168.0.28\W_OK rather than just 192.168.0.28. 

My question is what can I do to clear the extension?  Why doesn't reloading the index page, clear the previous request?  Is this a browser issue, an HTML issue, or something about how the request is handled?

Thanks in advance

zekageri, 2022-04-15:
Use a POST  request instead of GET

dduehren, 2022-04-20:
I'm having trouble getting POST to work.   I'm still using get on  the HTML above because that's a less important concern, although it didn't work either when I tried to make it a POST.  The problem is that it's not even executing the action path and I get error 500.   Here's the HTML where I'm using the POST
```
<form action="/wCRD" method ="post">
    <p>Please enter LAN wifi credentials.</p>
  	<label for="tempAdj"><b>SSID</b></label>
    <input type="text" name="NewSSID" id="NewSSID">
  	<label for="humAdj"><b>NewPass</b></label>
    <input type="text" name="NewPass" id="NewSSID">
    <button type="submit" class="submitBtn">Save</button>
  </form>
```
and here's the route handling code
```
    server.on("/wCRD", HTTP_GET, [](AsyncWebServerRequest *request){
    Debug.print(DBG_INFO, "wCRD route");
    StaticJsonDocument<200> doc;
    if(request->hasParam(F("NewSSID"),true)){
      inputMessage=request->getParam(F("NewSSID"),true)->value();
      doc["ssid"] = inputMessage;
//      Serial.printf("Wifi Cred route. SSID: %s", inputMessage.c_str());
       }
    if (request->hasParam(F("NewPass"),true)){
      inputMessage=request->getParam(F("NewPass"),true)->value();
      doc["password"] = inputMessage;
//      Serial.printf(" Password: %s\n", inputMessage.c_str());
      }
```
The debug statement isn't executing

dduehren, 2022-04-20:
Didn't solve the problem.  Still displaying the last URL when it loads index.html

zekageri, 2022-04-20:
Hmm... The whole point of a post request is to not display the requested url and to be able to send more data thought. Can you show us how you try to use post request?

dduehren, 2022-04-22:
All I can do you is show the HTML and the full response.  
```
<form action="/wCRD" method ="post" autocomplete="off">
    <p>Please enter LAN wifi credentials.</p>
  	<label for="ssid"><b>SSID</b></label>
    <input type="text" name="NewSSID" id="NewSSID">

  	<label for="passwd"><b>NewPass</b></label>
    <input type="text" name="NewPass" id="NewSSID">

    <button type="submit" class="submitBtn">Save</button>
  </form>
```
And here' the code at /wCRD
```
    server.on("/wCRD", HTTP_POST, [](AsyncWebServerRequest *request){
    Debug.print(DBG_INFO, "wCRD route");
    StaticJsonDocument<200> doc;
    if(request->hasParam(F("NewSSID"),true)){
      inputMessage=request->getParam(F("NewSSID"),true)->value();
      doc["ssid"] = inputMessage;
//      Serial.printf("Wifi Cred route. SSID: %s", inputMessage.c_str());
       }
    if (request->hasParam(F("NewPass"),true)){
      inputMessage=request->getParam(F("NewPass"),true)->value();
      doc["password"] = inputMessage;
//      Serial.printf(" Password: %s\n", inputMessage.c_str());
      }
      File configFile = LittleFS.open("/wifiCredentials.json", "w");
      if (!configFile) {
      Debug.print(DBG_ERROR,F("Failed to open wifi config file for writing"));
        }
      serializeJson(doc, configFile);
      configFile.close();
//      Serial.println("WifiCred just before html request response call");
      wifiMgr.wifiConnect(); // initiate wifi connect with new parameters.);  // start wifi after sending request
      // should be time to switch.  In case of AP mode, it will just switch modes, in case of STA mode, it will stay STA for a second.
      request->send(LittleFS, FPSTR(string_indexHTML), String(), false, processor);
      });
```

dduehren, 2022-04-25:
I'm at a complete loss about how to debug this.  The request-send does load the index.html page but the URL in the browser is still .../wCRD.

dduehren, 2022-04-25:
using POST with redirect does work.  So the last line becomes
```
      request->redirect(FPSTR(string_indexHTML));
```

closed 2022-04-25 by dduehren