me-no-dev/ESPAsyncWebServer #1139: AsyncWebHandler memory leak

Di-Strix, 2022-04-16:
Hardware: Wemos D1 mini 4mb.
I used PlatformIO with Arduino framework.
ESP8266 core version: 3.2.0

<details>
<summary>Sketch to reproduce:</summary>

```c++
#include <Arduino.h>
#include <ESPAsyncWebServer.h>

AsyncWebServer server(80);

void setup()
{
  Serial.begin(115200);
  Serial.println("\n\n\n\n\n");

  Serial.println("Free heap at start: " + (String)ESP.getFreeHeap());

  for (size_t i = 0; i < 100; i++) {
    auto h = server.on("*", [](AsyncWebServerRequest* request) {});
    server.removeHandler(&h);
  }

  Serial.println("Free heap after 100 creations and removals: " + (String)ESP.getFreeHeap());

  for (size_t i = 0; i < 100; i++) {
    auto h = server.on("*", [](AsyncWebServerRequest* request) {});
    server.removeHandler(&h);
  }

  Serial.println("Free heap after 200 creations and removals: " + (String)ESP.getFreeHeap());
}

void loop()
{
}
```

</details>

Sketch run results:
```
Free heap at start: 37592
Free heap after 100 creations and removals: 22392
Free heap after 200 creations and removals: 7192
```

It appears that `LinkedList::remove` could not find just registered handler and as a result - it is not deleted. According to the docs I should not do anything after calling `removeHandler`. Am I missing something?
<details>
<summary>Part from doc's 'Remove handlers and rewrites' section</summary>

```c++
// save callback for particular URL path
auto handler = server.on("/some/path", [](AsyncWebServerRequest *request){
  //do something useful
});
// when you don't need handler anymore remove it
server.removeHandler(&handler);
```
</details>

Di-Strix, 2022-04-16:
I see that callback-register functions return a reference to `AsyncCallbackWebHandler`. It would be much easier to store and operate them if they are pointers, isn't it?

blikkk, 2022-07-19:
UP

Di-Strix, 2022-08-04:
I will be really disappointed if it turns out that such a cool library is dead😕

But its current activity doesn't live up to the best expectations...

kmomberg, 2024-02-12:
Found that same issue. Any news or pull request around we can apply as a patch?