me-no-dev/ESPAsyncWebServer #1141: WebAuthentication.cpp undefined reference to `mbedtls_md5_starts'

m-menon, 2022-04-25:
I have the latest version of all the libraries and also PlatformIO. Suddenly this error started popping up

```
Archiving .pio\build\esp32dev\libFrameworkArduino.a
Linking .pio\build\esp32dev\firmware.elf
c:/users/dev/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\libf99\libESP Async WebServer.a(WebAuthentication.cpp.o):(.literal._ZL6getMD5PhtPc+0x10): undefined reference to `mbedtls_md5_starts'
c:/users/dev/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\libf99\libESP Async WebServer.a(WebAuthentication.cpp.o): in function `getMD5(unsigned char*, unsigned short, char*)':
C:\PlatformIO\Projects\Know\BTM/.pio/libdeps/esp32dev/ESP Async WebServer/src/WebAuthentication.cpp:74: undefined reference to `mbedtls_md5_starts'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1
```

I did some search for this
WebAuthentication.cpp undefined reference to `mbedtls_md5_starts' and the solution that I came across mentions updating to latest version of ESPAsyncWebServer. I have the very latest version and I even tried completely reinstalling everything ( Visual Code Studio, PlatformIO, all the libraries (latest versions) etc. Any ideas what could be wrong? I see that mbedtls_md5_starts is deprecated and superseded by mbedtls_md5_starts_ret(). I didn't want to just replace the function with the newer one as it might break something else. What would be the right way to fix this? Pls let me know if any additional information is required that might help narrow down on the issue. Thanks.

alex-code, 2022-04-26:
I got this error with an update to Espressif32 IDF 4.1
It has been updated, https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/WebAuthentication.cpp#L72 but I don't think there's a release for it.

If you reference the git url it'll download the latest, https://github.com/me-no-dev/ESPAsyncWebServer#using-platformio
```
lib_deps = https://github.com/me-no-dev/ESPAsyncWebServer.git
```

closed 2022-04-26 by m-menon

Keanu-Type, 2022-05-04:
hi im also encountering same problem. any update?

m-menon, 2022-05-04:
probably the same reason I guess. You can try this - change the line
mbedtls_md5_starts(&_ctx);
in WebAuthentication.cpp (ESP Async Web Server) src to
mbedtls_md5_starts_ret(&_ctx);

I think that will fix the problem for now - till the updated libraries are linked / picked up.

RoboticNerd, 2022-09-28:
How can this still be an issue? I mean I'm running stock everything just trying to run a tutorial and it still explodes from an obscure bug?

seconika, 2023-12-04:
Same here..   Can't get around it...