me-no-dev/ESPAsyncWebServer #1143: Redirect HTTPS -> HTTP

VNovytskyi, 2022-04-26:
I have a simple web server on ESP32:
```c
#include <WiFi.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>

AsyncWebServer server(80);

void setup() {
  Serial.begin(115200);

  WiFi.mode(WIFI_STA);
  WiFi.begin("*******", "********");
  if (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.printf("WiFi Failed!\n");
    return;
  }

  Serial.print("IP Address: ");
  Serial.println(WiFi.localIP());

  server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(200, "text/plain", "Hello, world");
  });

  server.onNotFound([](AsyncWebServerRequest *request) {
    request->send(404, "text/plain", "Not found");
  });

  server.begin();
}

void loop() {
  delay(100);
}
```
I can execute the request: `http://xxx.xxx.xxx.xxx/` and got a "Hello world" message on the web page, it's good. My server has no SSL Certificate. Can I execute the request `https://xxx.xxx.xxx.xxx/`, which will be forces redirected to `http://xxx.xxx.xxx.xxx/`?

IronBamBam1990, 2022-05-06:
> My server has no SSL Certificate

first you must create cert.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_https_server.html

`Can I execute the request https://xxx.xxx.xxx.xxx/, which will be forces redirected to http://xxx.xxx.xxx.xxx/?`

```
  server.on("/hello", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(200, "text/plain", "Hello from ESP32 server route");
  });
 
  server.on("/redirect/internal", HTTP_GET, [](AsyncWebServerRequest *request){
    request->redirect("/hello");
  });
 
  server.on("/redirect/external", HTTP_GET, [](AsyncWebServerRequest *request){
    request->redirect("https://techtutorialsx.com/");
  });
```
> Is there a way to reject the SSL handshake and continue the request in an unsecured context without ERR_SSL_PROTOCOL_ERROR ?

Yes. Generate Certs and there is something like root certs and server certs.

IronBamBam1990, 2022-05-06:
https://www.boost.org/doc/libs/develop/libs/beast/example/common/server_certificate.hpp
https://www.boost.org/doc/libs/develop/libs/beast/example/common/
OpenSSL

VNovytskyi, 2022-05-16:
Ok, thanks!

closed 2022-05-16 by VNovytskyi