me-no-dev/ESPAsyncWebServer #1147: WebAuthentication.cpp:73: undefined reference to `mbedtls_md5_starts'

41linea45, 2022-05-01:
Hi,
I am using Visual Studio code and added last library to my project.
platform.ini:
`env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps = 	me-no-dev/ESP Async WebServer@^1.2.3
			me-no-dev/AsyncTCP@^1.1.1
			ayushsharma82/AsyncElegantOTA@^2.2.6
`
It gives error:

c:/users/delphi/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\lib299\libESP Async WebServer.a(WebAuthentication.cpp.o):(.literal._ZL6getMD5PhtPc+0x4): undefined reference to `mbedtls_md5_starts'
c:/users/delphi/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\lib299\libESP Async WebServer.a(WebAuthentication.cpp.o): in function `getMD5(unsigned char*, unsigned short, char*)':
D:\Yazilim\Off_Road_2\ESP32\Inclinometer2/.pio/libdeps/esp32dev/ESP Async WebServer/src/WebAuthentication.cpp:73: undefined reference to `mbedtls_md5_starts'

How can i solve this problem? I used just sample code.

QuangKhuongIoT, 2022-05-02:
I had the same issue, waiting for the solution.

csetera, 2022-05-02:
I just ran into this as well.  I forced an older version of the expressif framework in platform.io for now:
```
  framework = arduino
  platform = espressif32@3.5.0
  board = esp32doit-devkit-v1
```

Keanu-Type, 2022-05-04:
> I just ran into this as well. I forced an older version of the expressif framework in platform.io for now:
> 
> ```
>   framework = arduino
>   platform = espressif32@3.5.0
>   board = esp32doit-devkit-v1
> ```

is there a way to force older version on Arduino IDE? currently my Platform IO is not running in VS Code(separate problem) so i can barely even try

MartinVerges, 2022-05-08:
super annoying issue, I used the workaround from #1085 to be able to compile my project. Please fix

82andre, 2022-05-08:
I think there may be a versioning issue, v1.2.3 was released 3 years ago
https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/library.properties  2 Oct 2019
platformio get this version 
https://registry.platformio.org/libraries/me-no-dev/ESP%20Async%20WebServer/versions
then after that at 26 Jul 2021 have a new file without new version
https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/WebAuthentication.cpp 

My bet is, that platformio will only update their files if you release a new version like 1.2.4

82andre, 2022-05-08:
https://docs.platformio.org/en/latest/librarymanager/creating.html

IronBamBam1990, 2022-05-08:
https://github.com/me-no-dev/ESPAsyncWebServer/pull/970

tmattukat, 2022-05-11:
Same issue, won´t write to esp until you switch to expressiv 3.5.0; If you use expressiv 3.5.0, then everything should be fine...

DanteNahuel, 2022-05-17:
PlatformIO still gives problems because the version in the json shows as 1.2.3 so Platform doesn't try to update the files.

JellevanKraaij, 2022-05-25:
Any updates on when this is fixed?

41linea45, 2022-05-26:
Hello again 
Thank you very much for replies. I apoligize for late feedback.
I solved problem in platformIO like that:

`lib_deps = 
	https://github.com/me-no-dev/ESPAsyncWebServer.git`

MartinVerges, 2022-05-26:
That's not solving, that's just a workaround. We need a simple version upgrade and a clean update of the registry files. This can only be done by the maintainer and it takes as little as a minute to solve the whole problem.

Erriez, 2022-08-17:
Can someone update the library / registry to fix other projects using the old style `lib_deps = ESP Async WebServer`?
Thanks!

maverck-, 2022-09-17:
> I think there may be a versioning issue, v1.2.3 was released 3 years ago 
https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/library.properties 2 Oct 2019 
platformio get this version 
https://registry.platformio.org/libraries/me-no-dev/ESP%20Async%20WebServer/versions 
then after that at 26 Jul 2021 have a new file without new version 
https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/WebAuthentication.cpp
> 
> My bet is, that platformio will only update their files if you release a new version like 1.2.4

Following this logic by @82andre (thanks) I was able to solve the problem.
By downloading the library directly (Code -> Download ZIP) and copying it to the libraries folder of my PlatformIO project (Project/lib/ESPAsyncWebServer).

RalphBacon, 2022-09-19:
PlatformIO.ini entry

`lib_deps = https://github.com/me-no-dev/ESPAsyncWebServer.git`

This solved it for me (rather then the usual `me-no-dev/ESP Async WebServer@^1.2.3` entry) but it means this still has not been "fixed" natively. Still is version 1.2.3.

me-chiel, 2022-09-26:
Thanks for the quick fix:

lib_deps = https://github.com/me-no-dev/ESPAsyncWebServer.git

this works for me, took me a while to figure that out....

sanjayraojjn, 2022-11-29:
replacing 
//platfromio.ini
lib_deps = ESP Async WebServer

with

//platfromio.ini
lib_deps = https://github.com/me-no-dev/ESPAsyncWebServer.git

worked. 
Is platformIO repo using an older version of the server?

h0l0gram, 2023-01-12:
@me-no-dev Please see https://github.com/me-no-dev/ESPAsyncWebServer/issues/1147#issuecomment-1120415957.
It seems to be a quickfix!

RalphBacon, 2023-01-30:
You can remove this error in PlatformIO by editing

```.pio\libdeps\esp32dev\ESP Async WebServer\src\WebAuthentication.cpp```

and remove (comment out) the call to ```mbedtls_md5_starts(&_ctx);``` (line 74, approx).

Or just get the library another way like this:
```
https://github.com/me-no-dev/ESPAsyncWebServer.git
```
which seems to force the download of the correct version of the library.

mjechow, 2023-02-02:
better use mbedtls_md5_starts_ret() instead of commenting out a deprecated method.

RalphBacon, 2023-02-02:
Yes, that seems to be the update in the GitHub:  

[image]

Oddly, I've had no trouble with the call not being there! 😲

SebastianKotstein, 2023-04-21:
I ran into same issue. However, only removing/commenting `mbedtls_md5_starts_ret()` out worked in my case. If I replace `me-no-dev/ESP Async WebServer@^1.2.3` with `https://github.com/me-no-dev/ESPAsyncWebServer.git` in platformio.ini the web server could not resolve any endpoint (I get a `404 Not Found` for any URI).

acsadamjozsef, 2023-06-27:
I could compiled, but it did not work for me with: 
"lib_deps = https://github.com/me-no-dev/ESPAsyncWebServer.git"

it just works like changing the decrapted lines to the new versions :
  mbedtls_md5_starts_ret(&_ctx);
  mbedtls_md5_update_ret(&_ctx, data, len);
  mbedtls_md5_finish_ret(&_ctx, _buf);

sseruzaifa, 2023-11-01:
> I had the same issue, waiting for the solution.
> 
> update: I had check the esp32 framework tool, and realized that the md5.h file had a small changing, the mbedtls_md5_starts() was removed. I think the radical solution is on the next update on the AsyncWebServer library.
> 
> update: the WebServer file was use new function mbedtls_md5_starts_ret() instead mbedtls_md5_starts().

thank you, saved my day

Westcott1, 2023-11-17:
Thanks, worked for me too!
Only mbedtls_md5_starts needed to be changed

andyb749, 2023-11-21:
I've not tried this.... but if you do a clean, then PlatformIO will go to the repositority to get the latest available version and not use the out of date one in the project. 

Failing that, fork the repository to your own Github account and link to that instead.

A

mortie, 2024-02-03:
@me-no-dev this is still an issue. Is ESPAsyncWebServer unmaintained?

mjechow, 2024-02-04:
This fork seems to be maintained: https://github.com/esphome/ESPAsyncWebServer

mortie, 2024-02-04:
@mjechow That's what I eventually found as well :) There's a bunch of different forks, but the esphome one seems like the most serious one. I added `esphome/ESPAsyncWebServer-esphome@3.1.0` and `esphome/AsyncTCP-esphome@2.1.1` to my platformio lib_deps and it works flawlessly so far.

talhaahmii, 2024-03-16:
lib_deps = https://github.com/me-no-dev/ESPAsyncWebServer.git

this worked in platform IO