me-no-dev/ESPAsyncWebServer #1148: websocket: how to execute long-running tasks and reply to the requestor

guestisp, 2022-05-02:
I'm struggling with ESPAsyncWebSocket.
Main goal is:

1. server receive a "long operation" request via websocket (multiple clients connected) with a json message
2. server start a very long running task (hours) in a async , non-blocking mode.
3. when the very long operation ends, the server should reply with a JSON to the requestor client (if still connected)

Step 1 is easy
Step 2 and 3 are very hard for me. When using anything like an RTOS task, a Ticker function or TaskScheduler task, even by passing the `AsyncWebSocketClient` got from the `wsOnEvent` even, every time I try to reply to the client, the esp32 panics and reboot.

Any idea ?

zekageri, 2022-05-02:
Why do you pass the AsyncWebSocketClient in the first place? Create a task at boot, pass the websocket clientID to a queue and process in the task. Or sleep the task, save the socket clinetID in a global variable and wake up the task on processing. The point is that you dont have to pass the whole client object to a task. You need only the ID and the relevant information from the packet.

You can create a struct like this:

```C
struct clientInfo{
    int clientID;
    boolean other;
};
```
**Create a queue and a task on setup:**

```C
QueueHandle_t clientQueue;

void setup(){
    clientQueue= xQueueCreate( 10, sizeof( struct clientInfo) );

    BaseType_t result = xTaskCreateUniversal(clientTask, "clientTask", 4000, NULL, 1, NULL, -1);
    if(result == pdPASS){
        Serial.printf("Task created on ANY available core with priority: %d and stack space: %d\n",1,4000);
    }else{
        Serial.println("Task failed to create.");
    }

}
```

**on request, harvest the json data into a new clientInfo struct and pass it into a queue**

```C
void socketEvent(AsyncWebSocket * server, AsyncWebSocketClient * client, AwsEventType type, void * arg, uint8_t *data, size_t len){
    if(type == WS_EVT_CONNECT){
            Serial.println("Client connected");
    }else if(type == WS_EVT_DISCONNECT){
            Serial.println("Client disconnected");
    }else if(type == WS_EVT_DATA){

       // Dont want to write the serialization and etc..
       clientInfo newClientInfo;
       newClientInfo.clientID = client->id();
       // etc...
       xQueueSend(clientQueue, &newClientInfo, portMAX_DELAY);

    }
}
```

**On the created task, wait for the queue.**

```C
void clientTask(void* parameter){
    for(;;){

      clientInfo newClientInfo;
      if (xQueueReceive(clientQueue, &newClientInfo, 0) == pdTRUE) {
          // process the data in newClientInfo
          // create a json string to send back
          String responseJson;
          ws.send(responseJson,newClientInfo.clientID);
      }else{
          vTaskDelay(100); // for example...
      }

    }
}
```

guestisp, 2022-05-02:
Sounds interesting. Doing in this way, i have to create a queue for each long-running command, right ?
If I understood properly, you are creating the queue with an endless task inside. The task is polling the queue waiting for a new job, it does the job then reply back to the client ? Isn't this blocking ?

Can I use the `clientTask` to parse the request (i'll add the request json to the struct) and do the long running operation ? Or would it block ?

zekageri, 2022-05-02:
You dont need to create a queue for every long running command. At least not necessarily. You can create multiple structs and send them all to the same queue one by one ( if they are identical data structures ) and the task will process them one by one.

Isn't this blocking? If you are on esp32, the vTaskDelay() function will let other tasks to run and it just blocks that one task where you are call the function in.

zekageri, 2022-05-02:
If you must parse every task in parallel, then you must create a task for each task when you get the message and not in the setup.
You can create a function to create a task and call that function if you got a message. But if you need it in parallel a different approach must be used

guestisp, 2022-05-02:
Blocking that single running task is good, is exactly what I need. I need to be able to run 2 or more task in parallel, but each single task, has to block the conside inside itself.

I'll try asap

zekageri, 2022-05-02:
You could create an array for all the structs for all the tasks that you want to process, on message you push a new struct to the array and create a task for that struct only and run all in parallel in it's own task.

But these are just brainstorming. If you speficy what you want to do it might help a little bit more to write a solution

zekageri, 2022-05-02:
message -> create a new struct with relevant info from message -> put struct into an array -> create a task for that struct -> task process the struct data -> delete task -> null array index

That way you will have an array of structs representing all your tasks.

But are you on esp32? Do you have external ram? What do you want to do exactly?

guestisp, 2022-05-02:
> But these are just brainstorming. If you speficy what you want to do it might help a little bit more to write a solution

easy for a c++ programmer (i'm not a c++ programmer, i use other languages): i need to control the esp32 via JSON-RPC.
json-rpc can be received with multiple channels (websocket, mqtt, HTTP POST, ...) . 90% of the received commands are very fast, but there are some commands known to be slow (hours runnings before a result). This means that via JSON-RPC I have to receive the command, "free" the connection and then, when the operation is done, reply to the original client

Something like: `startDishwasher` or `startWashingMachine`. Obviously, the command received via websocket/mqtt has to end immediatly, but the command result would arrive 3-4 hours later.

zekageri, 2022-05-02:
But i think you could process the data in parallel with just one task too. If you create an array for the structs, pushing new structs to the array and processing every array index in a task if it is not null.

Something like this:

```C
#define MAX_INFO_COUNT 20

int clientArrayTrack = 0;
typedef struct {
    int clientID;
    boolean other;
} clientInfo;

clientInfo* clientInfoArray[MAX_INFO_COUNT]

void onMessage( int clientID ){
    clientInfoArray[ clientArrayTrack ] = new clientInfo();
    
    // put the data inside
    clientInfoArray[ clientArrayTrack ] ->clientID = clientID;

    clientArrayTrack++;
    if( clientArrayTrack > MAX_INFO_COUNT ){
        // set a flag that your array is full or something or keep track of the empty indexes with an other array or something like that
    }
}
```

On the one processor task just loop trought the array and process with the infos

guestisp, 2022-05-02:
So the long-running processing is done here:
```
void clientTask(void* parameter){
    for(;;){

      clientInfo newClientInfo;
      if (xQueueReceive(clientQueue, &newClientInfo, 0) == pdTRUE) {
          // process the data in newClientInfo
          // create a json string to send back
          // Serial.printf("Info is: %s client id is: %d",newClientInfo.other ? "OTHER : "NOT OTHER",newClientInfo.clientID);
          String responseJson;
          ws.send(responseJson,newClientInfo.clientID);
      }else{
          vTaskDelay(100); // for example...
      }

    }
}
```

is not blocking because is created in a new task here:
`BaseType_t result = xTaskCreateUniversal(clientTask, "clientTask", 4000, NULL, 1, NULL, -1);`

but you are creating 1 task processing the queue, so the processor and their child-tasks are done in a different task, but in a sync way, one task per time. If the first one is 3 hours, the second one will be processed after 3 hours.

zekageri, 2022-05-02:
This is true but if you do it the array buffer way, you can loop trought your array of structs in the task as long as you need to and process every packet parallel.

```C
#define MAX_INFO_COUNT 20

int clientArrayTrack = 0;
typedef struct {
    int clientID;
    boolean other;
} clientInfo;

clientInfo* clientInfoArray[MAX_INFO_COUNT]

void onMessage( int clientID ){
    clientInfoArray[ clientArrayTrack ] = new clientInfo();
    
    // put the data inside
    clientInfoArray[ clientArrayTrack ] ->clientID = clientID;

    clientArrayTrack++;
    if( clientArrayTrack > MAX_INFO_COUNT ){
        // set a flag that your array is full or something or keep track of the empty indexes with an other array or something like that
    }
}

void clientTask(void* parameter){
  for(;;){
    for(int index = 0; i < MAX_INFO_COUNT; i++){
      if( clientInfoArray[i] != 0 ){
       // Process the info in clientInfoArray[i]
      }
    }
    vTaskDelay(5); // must let other tasks to run
  }
}
```

zekageri, 2022-05-02:
**I would create a pusher function and a freeer function.**

```C
#define MAX_INFO_COUNT 20

int clientArrayTrack = 0;
typedef struct {
    int clientID;
    boolean other;
} clientInfo;

clientInfo* clientInfoArray[MAX_INFO_COUNT];

int getFreeIndex(){
  for(int i = 0; i < MAX_INFO_COUNT; i++){
    if( clientInfoArray[i] == 0 || clientInfoArray[i] == null){
      return i;
    }
  }
  return -1;
}

void freeClientData( int index ){
  delete clientInfoArray[i];
  clientInfoArray[i] = 0;
}

void pushStruct( int clientID, String jsonString ){
  int freeIndex = getFreeIndex();
  if( freeIndex != -1 ){
      // deserialize jsonString
      clientInfoArray[ freeIndex ] = new clientInfo();
      clientInfoArray[ freeIndex ] -> clientID;
      clientInfoArray[ freeIndex ] -> jsonString["things"];
  }else{
    Serial.println("No free space to push data");
  }
}

void onMessage( int clientID, String jsonString ){
    pushStruct( clientID, jsonString );
}

void clientTask(void* parameter){
  for(;;){
    for(int index = 0; i < MAX_INFO_COUNT; i++){
      if( clientInfoArray[i] != 0 ){
       // Process the info in clientInfoArray[i]
       // if clientInfoArray[i] is done, free the index
       freeClientData(i);
      }
    }
    vTaskDelay(5); // must let other tasks to run
  }
}
```

guestisp, 2022-05-02:
Data would be identical: i'll put the json from the request and the "source", but nothing prevent me to add some field in the struct that i leave empty if not needed, in example, the `client_id` if request is made from mqtt, would be empty, but `source` would be `mqtt` and not `websocket`.

parsing the `source` i'll know how to reply.

guestisp, 2022-05-02:
(i'm also trying with the `Ticker` class, creating a new ticker on each request, but the ticker is not running, i'm still debugging)

zekageri, 2022-05-02:
I edited my examples in my last comment. For timing you can use the millis() function too

guestisp, 2022-05-02:
> I edited my examples in my last comment. For timing you can use the millis() function too

thank you .i'll give a try in a couple of minutes :-)

guestisp, 2022-05-02:
Seems working properly.

No existing library for this ? Strange...

zekageri, 2022-05-02:
It wont work with queue if you try to process in parallel. You must use this array buffer thingy or multiple tasks.

guestisp, 2022-05-02:
So i'm still at the starting point, with a loop my code was already working. The issue is how to remove the loop as I can't pass any arguments to the `Ticker::once_ms` method, and I need to pass a reference to `this` plus some additional args

zekageri, 2022-05-02:
You did not show any code nor explain any of your intentions in detail. 
I don't know what you want to do exactly.
If you want to run a task for long periods of time like this, just save the user informations you get on socket event and consume the data from where you saved it.

Where can you save the data you ask? In flash or in ram using arrays or classes.

guestisp, 2022-05-02:
I don't have any working code, i'm struggling with this from a week with absolutely no success.
I would like to store the requests in ram , your system with a queue is fine and working properly, but the single task parsing the queue will hangs when running each task. I'm trying to modify the code spawning a new task from the `clientTask` method, in this case, there is a task parsing the queue, then on new requests, a new parallel task should be started without blocking the queue loop:

```
#include "Arduino.h"

#include "constants.h"
#include "pins.h"

#include "taskqueue.h"
#include "webserver.h"

#include "freertos/task.h"
#include "freertos/queue.h"

TaskQueue::TaskQueue() { }

void TaskQueue::init() {
    clientQueue= xQueueCreate( 10, sizeof( requestInfo ) );

    BaseType_t result = xTaskCreatePinnedToCore(
        [](void * pvParameters) {
            TaskQueue *taskqueue = (TaskQueue*) pvParameters;
            taskqueue->clientTask();
        }
        ,"queueLoopTask"
        ,4000
        ,this
        ,1
        ,NULL
        ,1);

    if(result == pdPASS){
        Serial.printf("taskqueue.cpp: Task created on ANY available core with priority: %d and stack space: %d\n",1,4000);
    }else{
        Serial.println("taskqueue.cpp:  Task failed to create.");
    }
}

int TaskQueue::getFreeIndex()
{
    for (int i = 0; i < MAX_TASKS_COUNT; i++)
    {
        if (requestInfoArray[i] == 0 || requestInfoArray[i] == NULL)
        {
            return i;
        }
    }
    return -1;
}

void TaskQueue::freeClientData(int index)
{
    delete requestInfoArray[index];
    requestInfoArray[index] = 0;
}

void TaskQueue::pushStruct(int clientID, String requestJSON, String channel)
{
    int freeIndex = getFreeIndex();
    if (freeIndex != -1)
    {
        // deserialize jsonString
        requestInfoArray[freeIndex] = new requestInfo();
        requestInfoArray[freeIndex]->clientID = clientID;
        requestInfoArray[freeIndex]->requestJSON = requestJSON;
        requestInfoArray[freeIndex]->channel = channel;
    }
    else
    {
        Serial.println("No free space to push data");
    }
}

void TaskQueue::onMessage(int clientID, String requestJSON, String channel)
{
    pushStruct(clientID, requestJSON, channel);
}

void TaskQueue::clientTask()
{
    for (;;)
    {
        for (int i = 0; i < MAX_TASKS_COUNT; i++)
        {
            if (requestInfoArray[i] != 0)
            {

            taskParameters taskParams;

            taskParams.index = i;

            BaseType_t result = xTaskCreatePinnedToCore(
                    [](void * pvParameters) {
                        taskParameters *taskParams = (taskParameters*) pvParameters;
                        int i = (int) taskParams->index;
                        
                        
                        // Process the info in requestInfoArray[i]
                        Serial.print("taskqueue.cpp: INSIDE TASK, parsing command at slot #");
                        Serial.println(i);

                        Serial.print("taskqueue.cpp: client_id:");
                        Serial.print(taskqueue.requestInfoArray[i]->clientID);
                        Serial.print(",json:");
                        Serial.print(taskqueue.requestInfoArray[i]->requestJSON);
                        Serial.print(",source:");
                        Serial.println(taskqueue.requestInfoArray[i]->channel);

                        if ( taskqueue.requestInfoArray[i]->channel == "websocket" ) {
                            delay(60*1000);
                            Serial.print("taskqueue.cpp: delay response to client #");
                            Serial.println(taskqueue.requestInfoArray[i]->clientID);
                            webserver.websocket->text(taskqueue.requestInfoArray[i]->clientID, "{ FAKE JSON REPLY }");
                        }
                    
                        // if requestInfoArray[i] is done, free the index
                        taskqueue.freeClientData(i);

                        Serial.print("taskqueue.cpp: END TASK");

                    }
                    ,"clientTask"
                    ,4000
                    ,&taskParams
                    ,1
                    ,NULL
                    ,1);

                if(result == pdPASS){
                    Serial.printf("taskqueue.cpp: Task created on ANY available core with priority: %d and stack space: %d\n",1,4000);
                }else{
                    Serial.println("taskqueue.cpp:  Task failed to create.");
                }

            }
        }
        //Serial.println("taskqueue.cpp: clientTask loop");
        vTaskDelay(5); // must let other tasks to run
    }
}

```

zekageri, 2022-05-02:
You don't need queue for the array method as i said earlier.
Just push the new data to the array and process it inside your task.

```C
int clientDataIndex = 0;

void processWashingMachineData(){
    Serial.printf("Processing Washing Machine Data\n");
}

void processVacuumData(){
    Serial.printf("Processing Vacuum Data\n");
    // if this processing end, free it's index after we texted the result to the client
    freeClientData( clientDataIndex );
}

void clientTask(void* parameter){
    for(;;){
        if( clientInfoArray[clientDataIndex] != 0 ){
            if( clientInfoArray[clientDataIndex]->dataKind == WASHING_MACHINE_KIND ){
                processWashingMachineData();
            }else if( clientInfoArray[clientDataIndex]->dataKind == VACUUM_KIND ){
                processVacuumData();
            }
        }
        clientDataIndex++;
        if( clientDataIndex > MAX_INFO_COUNT ){
            clientDataIndex = 0;
        }
        vTaskDelay(1);  // Must let other tasks to run.
    }
}
```

guestisp, 2022-05-02:
```
void clientTask(void* parameter){
    for(;;){
        if( clientInfoArray[clientDataIndex] != 0 ){
            if( clientInfoArray[clientDataIndex]->dataKind == WASHING_MACHINE_KIND ){
                processWashingMachineData();
            }else if( clientInfoArray[clientDataIndex]->dataKind == VACUUM_KIND ){
                processVacuumData();
            }
        }
        clientDataIndex++;
        if( clientDataIndex > MAX_INFO_COUNT ){
            clientDataIndex = 0;
        }
        vTaskDelay(1);  // Must let other tasks to run.
    }
}
```

this is blocking. both task has to run at the same time, if both are present in the array.
is blocking because `processWashingMachineData()` or `processVacuumData();` wont return untl they finish, after 3-4 hours

Pablo2048, 2022-05-02:
@guestisp why you can not pass any argument into `Ticker::once_ms` ? What is exact error you've got? Where the crash was and what is the exact result from exception decoder? The long term tasks can be polled, or they have to be real FreeRTOS tasks (if they have to be FreeRTOS tasks, then using Ticker is useless)? Why you can not post MCVE, or the bad, but compilable code so we can work with you instead of some crippled fragments? Yes, we already know, that this code is not working - that's why you have been asking for help... All i have seen so far aims to dynamically create whole task, but without seeing complete code I'm not doing anything - zekageri waste his time doing code for you just to be informed that it is not what you want :-(

guestisp, 2022-05-02:
@Pablo2048 i don't have a MCVE code. It doesn't work.

`Ticker::once_ms` was suggested on another issue to avoid the watchdog timer in AsyncTCP lib. I'm not forced to use that, but obviously, keeping the code as short and clean as possible, is better.
What i'm trying to do is easy: calling `Ticker::once_ms` inside `wsOnEvent(AsyncWebSocket * server, AsyncWebSocketClient * client, AwsEventType type, void * arg, uint8_t *data, size_t len)` (i'm using a class, so I have to call a member function or a Lambda)

Like this:
```
triggerAction.once_ms(1000, []() {
      Serial.println("webserver.cpp: start ticker");
      delay(60*1000);
      Serial.println("webserver.cpp: end of ticker process");
    });
```

so, from there, how can I pass `this` AND the following struct ?
```
typedef struct {
            int clientID;
            String channel;
            String requestJSON;
        } requestInfo;
```

I can pass `this` but without the struct, i don't have any reference on who requested the command. Without `this` I don't have access to the websocket server to reply to the client.

During many tests, i was able to reach the websocket instance, but always resulted in the following panic:
```
webserver.cpp: reply to client #1905558911
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
```

I'm stuck at that panic. As you can see, the client id is way to long, probably is not working properly.

Then i've tried spawning the freertos process:

```
WebServer::taskParameters taskParams;
  taskParams.client_id = client_id;
  taskParams.server = server;

    xTaskCreatePinnedToCore (
      [](void * taskParams) {
        Serial.println("webserver.cpp: task wifiScanNetworks running on core ");
        Serial.println(xPortGetCoreID());

        taskParameters * data = (taskParameters *) taskParams;

        Serial.printf("webserver.cpp: task wifiScanNetworks: reply to client #%u\n", data->client_id);
        
        data->server->text(data->client_id,"test reply");

        vTaskDelete( NULL );
      },              // Task function.
      "Task1",        // name of task.
      10000,          // Stack size of task
      (void*)&taskParams, // parameter of the task
      1,              // priority of the task
      NULL,           // Task handle to keep track of created task
      1
    );             // pin task to core 0
```

```
class WebServer {
  public:
    WebServer(int port, String wsPath);
    void init();
    void begin();

  private:    
    AsyncWebServer server;
    AsyncWebSocket websocket;

    struct taskParameters {
      AsyncWebSocket * server;
      uint32_t  client_id;
    };    
      
    void initWebServer();
    void initWebSocket();
    void setCaptivePortalRedirects();

    static void processRPC(String jsonRPC, AsyncWebSocket * server, uint32_t client_id);
    static String getContentType(const String &filename);
    static bool handleStaticFile(AsyncWebServerRequest *request);

    static void wsOnEvent(AsyncWebSocket * server, AsyncWebSocketClient * client, AwsEventType type, void * arg, uint8_t *data, size_t len);

};
```

Pablo2048, 2022-05-02:
> @Pablo2048 i don't have a MCVE code. It doesn't work.

OMG we already **know that** - it's why you are asking for help...

guestisp, 2022-05-02:
> OMG we already know that - it's why you are asking for help...

I don't have each version, i'm refactoring 20-30 times a day until I found a solution. Trust me, it's much easier to provide a code sample like @zekageri did, allowing to start a "background" process and then send back the result to the websocket client (if still connected, if not, it doesn't matter)

> what Panic, where, what is exception decoder saying?
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Pablo2048, 2022-05-02:
> Trust me, it's much easier to provide a code sample like @zekageri did, allowing to start a "background" process and then send back the result to the websocket client (if still connected, if not, it doesn't matter)

easier for you, but not for us - trust me, I'm doing firmware development for >35 years.

> Guru Meditation Error: Core 1 panic'ed (LoadProhibited). Exception was unhandled.

This is not complete there is post-mortal dump after this message so use the exception decoder.

guestisp, 2022-05-02:
> This is not complete there is post-mortal dump after this message so use the exception decoder.

I don't have the previous log.
Trying to restore the old code, let's see if i'm able to trigger the same error.

zekageri, 2022-05-02:
So you dont need a long running process just trigger something after some time? If this is the case you just use the code i provided

guestisp, 2022-05-02:
> So you dont need a long running process just trigger something after some time? If this is the case you just use the code i provided

no, i have to run one or more 3 hours processes, in a non-blocking mode. They are light on cpu, only VERY VERY long.

Pablo2048, 2022-05-02:
@zekageri i think that we all get him wrong - it seems like he need to run FreeRTOS task on every "longterm" request with given arguments. After this task ends, the result is sent back to the asking websocket client...

guestisp, 2022-05-02:
> i think that we all get him wrong - it seems like he need to run FreeRTOS task on every "longterm" request with given arguments. After this task ends, the result is sent back to the asking websocket client...

correct, but the freertos task is not a requirement. It's a JSON-RPC server, where clients ask for some operation and these results msut sent back to the client (after hours)

spawning a "background" process was the easier think on my mind: i get the request, i spawn a process passing the websocketclient instance, then when the operation is done, i reply on websocketclient instance...

guestisp, 2022-05-02:
anyway, this queue works as expected. Could be done better?
Is spawning a queue process and then a new process for each operation request (only the first time, because of `processing` boolean)

```
#include "Arduino.h"

#include "constants.h"
#include "pins.h"

#include "taskqueue.h"
#include "webserver.h"

#include "freertos/task.h"
#include "freertos/queue.h"

TaskQueue::TaskQueue()
{

}

void TaskQueue::init() {
    Serial.println("taskqueue.cpp: init()");

    clientQueue= xQueueCreate( 10, sizeof( requestInfo ) );

    BaseType_t result = xTaskCreatePinnedToCore(
        [](void * pvParameters) {
            TaskQueue *taskqueue = (TaskQueue*) pvParameters;
            taskqueue->clientTask();
        }
        ,"queueLoopTask"
        ,4000
        ,this
        ,1
        ,NULL
        ,1);

    if(result == pdPASS){
        Serial.printf("taskqueue.cpp: Task created on ANY available core with priority: %d and stack space: %d\n",1,4000);
    }else{
        Serial.println("taskqueue.cpp: Task failed to create.");
    }
}

int TaskQueue::getFreeIndex()
{
    for (int i = 0; i < MAX_TASKS_COUNT; i++)
    {
        if (requestInfoArray[i] == 0 || requestInfoArray[i] == NULL)
        {
            return i;
        }
    }
    return -1;
}

int TaskQueue::countUsedSlots()
{
    int counter = 0;
    for (int i = 0; i < MAX_TASKS_COUNT; i++)
    {
        if (requestInfoArray[i] != 0 && requestInfoArray[i] != NULL)
        {
            counter++;
        }
    }
    return counter;
}

void TaskQueue::freeClientData(int index)
{
    delete requestInfoArray[index];
    requestInfoArray[index] = 0;
}

void TaskQueue::pushStruct(int clientID, String requestJSON, String channel)
{
    int freeIndex = getFreeIndex();
    if (freeIndex != -1)
    {
        // deserialize jsonString
        requestInfoArray[freeIndex] = new requestInfo();
        requestInfoArray[freeIndex]->clientID = clientID;
        requestInfoArray[freeIndex]->requestJSON = requestJSON;
        requestInfoArray[freeIndex]->channel = channel;
        requestInfoArray[freeIndex]->processing = false;
    }
    else
    {
        Serial.println("No free space to push data");
    }
}

void TaskQueue::onMessage(int clientID, String requestJSON, String channel)
{
    Serial.println("taskqueue.cpp: onMessage");
    pushStruct(clientID, requestJSON, channel);
}

void TaskQueue::clientTask()
{
    for (;;)
    {
        for (int i = 0; i < MAX_TASKS_COUNT; i++)
        {
            if (requestInfoArray[i] != 0 && requestInfoArray[i]->processing != true)
            {

            taskParameters taskParams;
            taskParams.index = i;

            BaseType_t result = xTaskCreatePinnedToCore(
                    [](void * pvParameters) {
                        taskParameters *taskParams = (taskParameters*) pvParameters;
                        int i = (int) taskParams->index;
                        
                        taskqueue.requestInfoArray[i]->processing = true;
                        
                        // Process the info in requestInfoArray[i]
                        Serial.print("taskqueue.cpp:          INSIDE TASK. slot #");
                        Serial.println(i);

                        Serial.print("taskqueue.cpp:          client_id:");
                        Serial.print(taskqueue.requestInfoArray[i]->clientID);
                        Serial.print(",json:");
                        Serial.print(taskqueue.requestInfoArray[i]->requestJSON);
                        Serial.print(",source:");
                        Serial.println(taskqueue.requestInfoArray[i]->channel);

                        if ( taskqueue.requestInfoArray[i]->channel == "websocket" ) {
                            delay(60*1000);
                            Serial.print("taskqueue.cpp:          sending reply to client_id: #");
                            Serial.println(taskqueue.requestInfoArray[i]->clientID);
                            webserver.websocket->text(taskqueue.requestInfoArray[i]->clientID, "JSON RISPOSTA");
                        }
                    
                        // if requestInfoArray[i] is done, free the index
                        taskqueue.freeClientData(i);

                        Serial.print("taskqueue.cpp:          END TASK. slot #");
                        Serial.println(i);

                        vTaskDelete( NULL );
                    }
                    ,"clientTask"
                    ,4000
                    ,&taskParams
                    ,1
                    ,NULL
                    ,1);

                if(result == pdPASS){
                    Serial.printf("taskqueue.cpp: Task created on ANY available core with priority: %d and stack space: %d\n",1,4000);
                }else{
                    Serial.println("taskqueue.cpp:  Task failed to create.");
                }

            }
        }
        //Serial.println("taskqueue.cpp: clientTask loop");
        vTaskDelay(5); // must let other tasks to run
    }
}
```

Pablo2048, 2022-05-02:
> correct, but the freertos task is not a requirement. It's a JSON-RPC server, where clients ask for some operation and these results msut sent back to the client (after hours)

well actually tthe freertos task is the only option I can see - I asked you if the long term tasks can be polled, but didn't get any answer.

Simplest method is to dynamically create structure with arguments, create the task and send pointer to this structure as task *pvParameter. And don't forget to delete the structure after the task ends... No queue, no array need imho.

guestisp, 2022-05-02:
> well actually tthe freertos task is the only option I can see - I asked you if the long term tasks can be polled, but didn't get any answer.

yes that's possible.

> Simplest method is to dynamically create structure with arguments, create the task and send pointer to this structure as task *pvParameter. And don't forget to delete the structure after the task ends... No queue, no array need imho.

like with the pointer sample you did above ? that's would be even easier. but not with Ticker class because i've seen there are some issues. I think a dedicated task would be better. Something like i've tried here with `taskParams` ?
```
WebServer::taskParameters taskParams;
  taskParams.client_id = client_id;
  taskParams.server = server;

    xTaskCreatePinnedToCore (
      [](void * taskParams) {
        Serial.println("webserver.cpp: task wifiScanNetworks running on core ");
        Serial.println(xPortGetCoreID());

        taskParameters * data = (taskParameters *) taskParams;

        Serial.printf("webserver.cpp: task wifiScanNetworks: reply to client #%u\n", data->client_id);
        
        data->server->text(data->client_id,"test reply");

        vTaskDelete( NULL );
      },              // Task function.
      "Task1",        // name of task.
      10000,          // Stack size of task
      (void*)&taskParams, // parameter of the task
      1,              // priority of the task
      NULL,           // Task handle to keep track of created task
      1
    );             // pin task to core 0
```

Pablo2048, 2022-05-02:
Yes, exactly. Just one more suggestion - remove the lambda, because there is no need to use it...

guestisp, 2022-05-02:
i'll create a class with some static methods used to handle each operation. then on the websocket event i set the pvParameters struct and spawn the task pointing to the static mrthod without lambda, i suppose

let's see if It works

Pablo2048, 2022-05-02:
Don't forget to check the result of `new` and `xTaskCreate...` and free the memory correctly - you can get very quickly out of RAM because there is no limit of tasks spawned like it was with the static array.

guestisp, 2022-05-02:
Any suggestion on how to make dynamic the function name for `xTaskCreatePinnedToCore` ? Just to avoit a lot of cut&paste of the same line, when just the first arguments change.

guestisp, 2022-05-02:
@Pablo2048 no way, every time I try to read out of the task code, it crash:

```
executor.cpp: taskProcessor running on core #1
executor.cpp: jsonRPC=
Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.
Core 1 register dump:
PC      : 0x400e2eab  PS      : 0x00060330  A0      : 0x800e1498  A1      : 0x3ffdcbb0  
A2      : 0x3ffbdd08  A3      : 0x4008c0c4  A4      : 0x400900be  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x0000007e  A9      : 0x3ff40000  
A10     : 0x0000007f  A11     : 0x00000000  A12     : 0x3ffb8218  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0xff000000  SAR     : 0x00000000  EXCCAUSE: 0x00000003  
EXCVADDR: 0x4008c0c4  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x400e2eab:0x3ffdcbb0 0x400e1495:0x3ffdcbd0 0x400e17d9:0x3ffdcbf0 0x400e1811:0x3ffdcc10 0x400d2176:0x3ffdcc30 0x400d2331:0x3ffdd0f0 0x400899de:0x3ffdd110
  #0  0x400e2eab:0x3ffdcbb0 in uartWriteBuf at /home/ale/.platformio/packages/framework-arduinoespressif32@3.10006.210326/cores/esp32/esp32-hal-uart.c:464
  #1  0x400e1495:0x3ffdcbd0 in HardwareSerial::write(unsigned char const*, unsigned int) at /home/ale/.platformio/packages/framework-arduinoespressif32@3.10006.210326/cores/esp32/HardwareSerial.cpp:190
  #2  0x400e17d9:0x3ffdcbf0 in Print::write(char const*, unsigned int) at /home/ale/.platformio/packages/framework-arduinoespressif32@3.10006.210326/cores/esp32/Print.cpp:261
      (inlined by) Print::print(String const&) at /home/ale/.platformio/packages/framework-arduinoespressif32@3.10006.210326/cores/esp32/Print.cpp:84
  #3  0x400e1811:0x3ffdcc10 in Print::println(String const&) at /home/ale/.platformio/packages/framework-arduinoespressif32@3.10006.210326/cores/esp32/Print.cpp:261
  #4  0x400d2176:0x3ffdcc30 in Executor::taskProcessor(void*) at src/executor.cpp:68
  #5  0x400d2331:0x3ffdd0f0 in Executor::execute()::{lambda(void*)#1}::_FUN(void*) at src/executor.cpp:92
      (inlined by) _FUN at src/executor.cpp:92
  #6  0x400899de:0x3ffdd110 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...

```

`taskProcessor` is this:

```
void Executor::taskProcessor(void *taskParams)
{
    int coreID = xPortGetCoreID();

    Executor *executor = (Executor *)taskParams;

    Serial.print("executor.cpp: taskProcessor running on core ");
    Serial.println(coreID);

    Serial.print("executor.cpp: jsonRPC=");
    Serial.println(executor->jsonRPC);

    StaticJsonDocument<1024> json;
    deserializeJson(json, executor->jsonRPC);
    String method = json["method"];
    
    Serial.print("executor.cpp: method:");
    Serial.println(method);

    

    if (method.equals("wifi.getNetworks"))
    {
        executor->getWifiNetworks(taskParams);
    }

    Serial.println("executor.cpp: done running taskProcessor on core ");
    Serial.println(coreID);

    vTaskDelete(NULL);
}
```

The only way i've found to call a non static (i can't use static function) method is with the following, found on the net:
```
auto task = [](void* arg) { static_cast<Executor *>(arg)->taskProcessor(arg); };
    xTaskCreatePinnedToCore(
        task, // Task function.
        "taskProcessor",  // name of task.
        10000,            // Stack size of task
        this,             // parameter of the task
        1,                // priority of the task
        NULL,             // Task handle to keep track of created task
        1);
```

**Why I can't use static methods ?** 
Because I need to share class properties across the methods, with statics this is not possible (at least in other languages I use). Without using class inheritance i have to duplicate a lot of code.

```
// executor.cpp
#include "executor.h"

#include "Arduino.h"

#define ARDUINOJSON_USE_LONG_LONG 1
#include <ArduinoJson.h>

#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>

#include "constants.h"
#include "pins.h"

/*
 * JSON-RPC coming from websocket
 */
Executor::Executor(String jsonRPC, int wsClientId)
{
    jsonRPC = jsonRPC;
    wsClientId = wsClientId;
};

/*
 * JSON-RPC coming from MQTT
 */
Executor::Executor(String jsonRPC)
{
    jsonRPC = jsonRPC;
};

void Executor::getWifiNetworks(void *taskParams)
{
    Executor *executor = (Executor *)taskParams;

    Serial.println("executor.cpp: START getWifiNetworks");

    Serial.print("executor.cpp:      ");
    Serial.println(jsonRPC);

    Serial.println("executor.cpp: END getWifiNetworks ");

    vTaskDelete(NULL);
}

void Executor::taskProcessor(void *taskParams)
{
    int coreID = xPortGetCoreID();

    Executor *executor = (Executor *)taskParams;

    Serial.println("executor.cpp: taskProcessor running on core ");
    Serial.print(coreID);

    Serial.println("executor.cpp: jsonRPC=");
    Serial.print(executor->jsonRPC);

    StaticJsonDocument<1024> json;
    deserializeJson(json, executor->jsonRPC);
    String method = json["method"];
    
    Serial.print("executor.cpp: method:");
    Serial.println(method);

    

    if (method.equals("wifi.getNetworks"))
    {
        executor->getWifiNetworks(taskParams);
    }

    Serial.println("executor.cpp: done running taskProcessor on core ");
    Serial.println(coreID);

    vTaskDelete(NULL);
}

void Executor::execute()
{
    auto task = [](void* arg) { static_cast<Executor *>(arg)->taskProcessor(arg); };
    xTaskCreatePinnedToCore(
        task, // Task function.
        "taskProcessor",  // name of task.
        10000,            // Stack size of task
        this,             // parameter of the task
        1,                // priority of the task
        NULL,             // Task handle to keep track of created task
        1);
};
```

Called in this way from the `wsOnEvent`:
```
Executor executor("{............}", wsClient->id());
executor.execute();
```

zekageri, 2022-05-02:
I wanted to let you know earlier, i think this crash is because of the String class. It can lead to very misterious ways. If you define a String variable there will be the least amount of space allocated for it, and if you put anything inside it, it will dinamically allocate space for itself. If you create a task and you create a String inside it, it may overflow the stack space. 

If this error is not because if this, you should use char arrays anyway.

guestisp, 2022-05-02:
as far as i can see, it's related to the String, even when trying to read the JSON string from the Struct used as parameters

also, even ArduinoJson Is causing issues, if the string is deserialized, it bacame empty. trying to debug without ArduinoJson but Is still a mess

guestisp, 2022-05-02:
your solution with loops and arrays works, but i'm trying to keep the code as simple as possibile. in PHP, in example, i would trigger a background operation on websocket even and when finished, the background operation directly talks to the global websocket server. 10 lines of code, 10 minutes of works. i'm not a c++ developer but i'm strugglimg into this from a Little bit less then 10 days

guestisp, 2022-05-02:
> You can dinamically create classes too. You can create a class for a type of process and just create a new class which will run in a single task.

i use tasks to keep the firmware up and running to serve users. if i run a 4 hours long operation without a dedicated task, the device hangs for 4 hours

Pablo2048, 2022-05-03:
As I said - **remove the lambda** - there is no need for using it. If you need access to class properties, then write getters and setters and put reference of the instance to the structure - it's very simple. If I have to write this, then every longterm task would be in single file to avoid spaghetti code & this lambda mess. Embedded programming is not as simple as PC programming - you don't have zillions of bytes of RAM and disk space available, GHz processors under your a*s, so you have to use your brain to find a way out.
Anyway - from exception decoder it seems like the crash come from uart dump, so at first you have to check if you give valid arguments to `Serial` in Executor class.

zekageri, 2022-05-03:
Yeah, iam with Pablo in this. You can even write to the esp with python or javascript now. There are frameworks and compilers for this. But it will be garbage. You have no ram for this stuff. Even the String class is a luxury in this world. 

As of blocking and non blocking. You can easily create non blocking code in the same task. You just dont use delay. The point of a non blocking code is to use the esp's current millis. 

I can run a hundred lamp class in a non blocking way inside a single task.  If the lamps have to auto turn off at the exact same time, they will because there is no delay inside. I even showed you an example to this with millis() func. 

You should put every different class into a single file as Pablo said.

Pablo2048, 2022-05-03:
There are another errors in your code. What do you think about this?
```cpp
jsonRPC = jsonRPC;
```
if you want o use same name of member variable (we don't have any source of header so it's just an guess) as argument name (very confusing and bad practice), then you have to use `this->jsonRPC = jsonRPC;`.

guestisp, 2022-05-03:
yes the crash is with the Serial dump, because im trying to write the content of JSON ther for unknown reason (to me) it's empty sometimes. it's the cast and the assignment thar crash, but the string can't be empty, because the method is called only After a match on the string content...

i really love to remove lambda but without that im forced to use static methods and thus i'll loose access to the class property that i need, preciously set during the class init.

as for non -blocking code: i need to use some libs, i'm not sure they are delay-free. One of these libs is for controlling stepper motors and i'm pretty sure it's using delays

Pablo2048, 2022-05-03:
Man, you are really pi**ing me off! Decoded stack dump points to executor.cpp:68, but your given source code of executor has no print statement on this line. It seems like the source code given is not the source code producing the error (another waste of my precious time :-( ). How do you think that we can help? The JSON can be empty because of stupid assignment in the constructor of Executor class. Also you don't bother of JSON deserialization result - it's very bad...

Pablo2048, 2022-05-03:
> i really love to remove lambda but without that im forced to use static methods and thus i'll loose access to the class property that i need, preciously set during the class init.

this is nonsense - the lambda don't have any access to the class properties just because your capture [] is empty. The lambda is then executed the same way like calling any other function.

guestisp, 2022-05-03:
the assignment was a typo, fixed after posting on GitHub in a new refactor

actually i don't bother of deserialization because it's a test, printing the RAW string would be ok and lead to less issue than adding a new library (that trigger Always the same LoadProhinited panic)

guestisp, 2022-05-03:
> this is nonsense - the lambda don't have any access to the class properties just because your capture [] is empty. The lambda is then executed the same way like calling any other function.

ho do you use a capture list for the lamba used as function for the task? It won't compile with a capture list

and also won't compile using a class method that is not static, that why i don't have access to the class properties i need

Pablo2048, 2022-05-03:
As I said - remove the lambda and pass pointer of the Executor instance to the task. That's all - very simple and clean.

zekageri, 2022-05-03:
With ArduinoJson if you deserialize a char array and it is constant, it will use "copy mode", but if you use a non constant string, it will use this string to store pointers to the deserialized json packet. So your string could be empty if arduinoJson wanted to. 

I don't know which version is your ArduinoJson but here is a link for this exact thing on their site: [Altered inputs](https://arduinojson.org/v6/issues/altered-input/)

guestisp, 2022-05-03:
@zekageri yes , read that Yesterday, that's why now i'm trying without ArduinoJson, to keep things simple. i'm just trying to use a plain string

guestisp, 2022-05-03:
@Pablo2048 full code with full dump, in use right now, no lambdas and no typo in the json variable assignment.
Just static methods passing a struct as argument.
It crash every time I try to access to `taskParams->requestJSON` (i'm using 20000 as process stack size, rising from 10000 didn't changed anything)

```
webserver.cpp: ws[/ws][1] text-message[67]: {"jsonrpc":"2.0","method":"wifi.getNetworks","params":{},"id":5396}
webserver.cpp: calling Executor::execute for client_id #1
webserver.cpp: JSON STRING: {"jsonrpc":"2.0","method":"wifi.getNetworks","params":{},"id":5396}
executor.cpp: running ::execute for JSON={"jsonrpc":"2.0","method":"wifi.getNetworks","params":{},"id":5396}
1
executor.cpp: requestJSON=executor.cpp: taskProcessorGuru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400014dc  PS      : 0x00060e30  A0      : 0x800e0008  A1      : 0x3ffe90a0  
A2      : 0x00000001  A3      : 0xfffffffd  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x800dfda4  A9      : 0x3ffe9070  
A10     : 0x3ffbdcc0  A11     : 0x3f400168  A12     : 0x3f400168  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000000  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000001  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff9  

ELF file SHA256: 0000000000000000

Backtrace: 0x400014dc:0x3ffe90a0 0x400e0005:0x3ffe90b0 0x400e00f5:0x3ffe90d0 0x400d1185:0x3ffe90f0 0x400899de:0x3ffe9110
  #0  0x400014dc:0x3ffe90a0 in ?? ??:0
  #1  0x400e0005:0x3ffe90b0 in Print::write(char const*) at /home/ale/.platformio/packages/framework-arduinoespressif32@3.10006.210326/cores/esp32/Print.cpp:261
  #2  0x400e00f5:0x3ffe90d0 in Print::print(char const*) at /home/ale/.platformio/packages/framework-arduinoespressif32@3.10006.210326/cores/esp32/Print.cpp:261
  #3  0x400d1185:0x3ffe90f0 in Executor::taskProcessor(void*) at /home/ale/GIT/testProject/src/executor.cpp:27
  #4  0x400899de:0x3ffe9110 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
```

```
#ifndef EXECUTOR_h
#define EXECUTOR_h

#pragma once

#define ARDUINOJSON_USE_LONG_LONG 1
#include <ArduinoJson.h>

class Executor {
    public:
        typedef struct {
            uint32_t wsClientId;
            char *requestJSON;
            String channel;
            bool processing = false;
        } taskParameters ;

        static void getWifiNetworks(void * taskParams);
        static void taskProcessor(void * taskParams);
        
        static void execute(char *json, uint32_t wsClientId);
        static void execute(char *json);
        
        Executor();

    private:
};

#endif
```

```
// executor.cpp
#include "executor.h"

#include "Arduino.h"

#define ARDUINOJSON_USE_LONG_LONG 1
#include <ArduinoJson.h>

#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>

#include "constants.h"
#include "pins.h"

void Executor::taskProcessor(void * pvParameters)
{
    int coreID = xPortGetCoreID();

    taskParameters *taskParams = (taskParameters*) pvParameters;
    uint32_t wsClientId = (uint32_t)taskParams->wsClientId;
    char *json = (char *)taskParams->requestJSON;

    Serial.print("executor.cpp: taskProcessor running on core #");
    Serial.println(coreID);

    Serial.print("executor.cpp: requestJSON=");
    Serial.print(json);
    Serial.print(", wsClientId: #");
    Serial.println(wsClientId);

    
    Serial.print("executor.cpp: done running taskProcessor on core #");
    Serial.println(coreID);

    vTaskDelete(NULL);
}

void Executor::execute(char *json, uint32_t wsClientId)
{
    Serial.println("executor.cpp: running ::execute for JSON=" + String((char*)json));

    taskParameters taskParams;
    taskParams.requestJSON = json;
    taskParams.wsClientId  = wsClientId;

    xTaskCreatePinnedToCore(
        &taskProcessor, // Task function.
        "taskProcessor",  // name of task.
        20000,            // Stack size of task
        &taskParams,             // parameter of the task
        1,                // priority of the task
        NULL,             // Task handle to keep track of created task
        1);
};
```

Called in this way inside `WebServer::wsOnEvent`:

```
void WebServer::initWebSocket() {
    Serial.println("webserver.cpp: initWebSocket");

    // attach AsyncWebSocket
    websocket->onEvent(WebServer::wsOnEvent);
    server->addHandler(websocket);
}

void WebServer::wsOnEvent(AsyncWebSocket * server, AsyncWebSocketClient * client, AwsEventType type, void * arg, uint8_t *data, size_t len){
    DynamicJsonDocument doc(1024);
    DeserializationError error;

  if(type == WS_EVT_CONNECT) {
  //........
  } else if(type == WS_EVT_DISCONNECT){
  //........
  } else if(type == WS_EVT_ERROR){
  //........
  } else if(type == WS_EVT_PONG){
  //........
  } else if(type == WS_EVT_DATA){
    //data packet
    AwsFrameInfo * info = (AwsFrameInfo*)arg;
    if(info->final && info->index == 0 && info->len == len){
      //the whole message is in a single frame and we got all of it's data
      Serial.printf("webserver.cpp: ws[%s][%u] %s-message[%llu]: ", server->url(), client->id(), (info->opcode == WS_TEXT)?"text":"binary", info->len);
      if(info->opcode == WS_TEXT){
        data[len] = 0;
        Serial.printf("%s\n", (char*)data);
      } else {
        for(size_t i=0; i < info->len; i++){
          Serial.printf("%02x ", data[i]);
        }
        Serial.printf("\n");
      }
      
      const uint32_t client_id = client->id();
      const String json = String((char*)data);

      Serial.printf("webserver.cpp: calling Executor::execute for client_id #%u\n", client_id);
      Serial.println("JSON STRING: " + json);
      Executor::execute((char*)data, client_id);

    } else {
       //........
    }
  }
}

Pablo2048, 2022-05-03:
Well, that's probably because you don't care about object lifetime. Seriously why are you still trying to make things so complicated? Why are the properties inside struct which is public? Why are you still using static methods? The Executor class has to do two things:

1. hold the arguments
2. spawn new task with himself as argument

anyway the Executor instance must remain in memory until the task finishes. And after the task finishes the Executor has to commit suicide. You can not even think that `Executor::execute(...);` would be working.

guestisp, 2022-05-03:
> Why are you still using static methods?

Because I can't use a non-static method as function for the `xTaskCreatePinnedToCore`:

```
src/executor.cpp: In member function 'void Executor::execute(char*, uint32_t)':
src/executor.cpp:57:10: error: invalid use of non-static member function
         1);

    xTaskCreatePinnedToCore(
        nonStaticMethod,  // Task function. (tried with this->nonStaticMethod too)
        "taskProcessor", // name of task.
        8196,            // Stack size of task
        &taskParams,     // parameter of the task
        1,               // priority of the task
        NULL,            // Task handle to keep track of created task
        1);
```

From here: https://www.freertos.org/FreeRTOS_Support_Forum_Archive/July_2010/freertos_Is_it_possible_create_freertos_task_in_c_3778071.html

```
If you want a member function to act as the task, you need to wrap it with wrapper that looks something like  

 void taskfun(void* parm) {
    static_cast<classname *>(parm)->memberfun();
}
```

that's what i've tried yesterday, but there is a cast to a static, so the results are the same....

Pablo2048, 2022-05-03:
Ok, so why are you trying to put all inside one class? There is no need for Executor to contain task code... Also do you know the lifetime of the data? Are you aware of the fact, that the content of `*data` of wsOnEvent disappear after return from wsOnEvent? I don't see any copy of the data, you just save POINTER to the data...

[32 more replies]

closed 2022-05-06 by guestisp