me-no-dev/ESPAsyncWebServer #1149: the mbedtls_md5_starts() was removed from the orginal md5.h

QuangKhuongIoT, 2022-05-02:
I think that issue making this error.

c:/users/delphi/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\lib299\libESP Async WebServer.a(WebAuthentication.cpp.o):(.literal._ZL6getMD5PhtPc+0x4): undefined reference to mbedtls_md5_starts' c:/users/delphi/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch3/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\lib299\libESP Async WebServer.a(WebAuthentication.cpp.o): in function getMD5(unsigned char*, unsigned short, char*)':
D:\Yazilim\Off_Road_2\ESP32\Inclinometer2/.pio/libdeps/esp32dev/ESP Async WebServer/src/WebAuthentication.cpp:73: undefined reference to `mbedtls_md5_starts'

Rotzbua, 2022-05-04:
duplicate of #1147

yochai-g, 2022-08-04:
Hi!

I got a link error:

c:/users/yla/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio/build/CentralUnit/lib71a/libESP Async WebServer.a(WebAuthentication.cpp.o): in function `getMD5(unsigned char*, unsigned short, char*)':       
C:\Users\yla\Documents\PRIVATE\Arduino\VscProjects\SmartHome\CentralUnit/.pio/libdeps/CentralUnit/ESP Async WebServer/src/WebAuthentication.cpp:73: undefined reference to `mbedtls_md5_starts'