me-no-dev/ESPAsyncWebServer #1151: Undefined reference to `SHA1Init'

arslan437, 2022-05-10:
I am unable to compile the code on platformio. 

`
c:/users/pc 13/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\m5stack-core2\lib519\libESP Async WebServer.a(AsyncWebSocket.cpp.o):(.literal._ZN22AsyncWebSocketResponseC2ERK6StringP14AsyncWebSocket+0x10): undefined reference to `SHA1Init'
c:/users/pc 13/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\m5stack-core2\lib519\libESP Async WebServer.a(AsyncWebSocket.cpp.o):(.literal._ZN22AsyncWebSocketResponseC2ERK6StringP14AsyncWebSocket+0x18): undefined reference to `SHA1Update'
c:/users/pc 13/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\m5stack-core2\lib519\libESP Async WebServer.a(AsyncWebSocket.cpp.o):(.literal._ZN22AsyncWebSocketResponseC2ERK6StringP14AsyncWebSocket+0x1c): undefined reference to `SHA1Final'
c:/users/pc 13/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\m5stack-core2\lib519\libESP Async WebServer.a(AsyncWebSocket.cpp.o): in function `AsyncWebSocketResponse::AsyncWebSocketResponse(String const&, AsyncWebSocket*)':
D:\Fiverr\185-arnaudmosse(history)\esp32-server-graph/.pio/libdeps/m5stack-core2/ESP Async WebServer/src/AsyncWebSocket.cpp:1269: undefined reference to `SHA1Init'
c:/users/pc 13/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: D:\Fiverr\185-arnaudmosse(history)\esp32-server-graph/.pio/libdeps/m5stack-core2/ESP Async WebServer/src/AsyncWebSocket.cpp:1270: undefined reference to `SHA1Update'
c:/users/pc 13/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: D:\Fiverr\185-arnaudmosse(history)\esp32-server-graph/.pio/libdeps/m5stack-core2/ESP Async WebServer/src/AsyncWebSocket.cpp:1271: undefined reference to `SHA1Final'
c:/users/pc 13/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\m5stack-core2\lib519\libESP Async WebServer.a(WebAuthentication.cpp.o):(.literal._ZL6getMD5PhtPc+0x4): undefined reference to `mbedtls_md5_starts'
c:/users/pc 13/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\m5stack-core2\lib519\libESP Async WebServer.a(WebAuthentication.cpp.o): in function `getMD5(unsigned char*, unsigned short, char*)':
D:\Fiverr\185-arnaudmosse(history)\esp32-server-graph/.pio/libdeps/m5stack-core2/ESP Async WebServer/src/WebAuthentication.cpp:73: undefined reference to `mbedtls_md5_starts'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\m5stack-core2\firmware.elf] Error 1
`

richcj10, 2022-05-10:
Same error for me.

richcj10, 2022-05-10:
Pull the latest and greatest. https://github.com/me-no-dev/ESPAsyncWebServer.git

AaronD-GH, 2022-05-13:
> Pull the latest and greatest. https://github.com/me-no-dev/ESPAsyncWebServer.git

Is that to mean that the latest and greatest has this bug, or that it's fixed in that version?

At any rate, I have the same problem.  Added `lib_deps = ESP Async WebServer` to my `platformio.ini` today, which trigged an automatic download/install, and then the build has this error.

danielh186, 2022-05-15:
There is a [patch](https://github.com/me-no-dev/ESPAsyncWebServer/pull/999) that got merged here on the github repo, but there is no new release in platformIO.
So you can just go to the library folder and deleted everything but the `.piopm` file. Then clone or download this repo and put everything in there.

arslan437, 2022-05-16:
This is not due to library. I have recently updated esp32 platform due to which this error appeared. 

There are two ways to fix the issue. 

1. Download the library from the github and paste that in lib folder of the platformio folder. 
2. downgrade the your esp32 platform in the platformio.ini file. like shown below.

`
[env:m5stack-core2]
platform = espressif32 @ ~3.5.0
board = m5stack-core2
framework = arduino
;board_build.partitions = huge_app.csv

`

AaronD-GH, 2022-05-17:
> This is not due to library. I have recently updated esp32 platform due to which this error appeared.
> 
> There are two ways to fix the issue.
> 
> 1. Download the library from the github and paste that in lib folder of the platformio folder.
> 2. downgrade the your esp32 platform in the platformio.ini file. like shown below (platform = espressif32 @ ~3.5.0).
> 
> ` [env:m5stack-core2] platform = espressif32 @ ~3.5.0 board = m5stack-core2 framework = arduino ;board_build.partitions = huge_app.csv
> 
> `

Option 2 worked for me.  Didn't try 1, as it seemed like more effort.  Thanks!

pdelorme, 2022-05-29:
Easier way to option 1 :

```
[env:esp32doit-devkit-v1]
platform = espressif32
board = esp32doit-devkit-v1
framework = arduino
lib_deps = 
	me-no-dev/AsyncTCP @ ^1.1.1
	https://github.com/me-no-dev/ESPAsyncWebServer.git
```

closed 2022-11-02 by stale[bot]

srghafoori, 2023-02-21:
> This is not due to library. I have recently updated esp32 platform due to which this error appeared.
>
> There are two ways to fix the issue.
>
> 1. Download the library from the github and paste that in [...]

Option 1 worked for me. thanks :-)