me-no-dev/ESPAsyncWebServer #1157: how to use the keepAlivePeriod for clients ?

guestisp, 2022-05-13:
I'm trying to find a way to automatically ping all connected clients by using the native ping/pong feature (as workaround i could send custom messages from the client to the server) 

I've set this in `wsOnEvent`:
```
  if (type == WS_EVT_CONNECT)
  {
    // client connected
    client->printf("Hello Client %u :)", client->id());
    client->ping();
    client->keepAlivePeriod(10);
  }
  else if (type == WS_EVT_PONG)
  {
    // pong message was received (in response to a ping request maybe)
    Serial.printf("ws[%s][%u] pong[%u]: %s\n", server->url(), client->id(), len, (len) ? (char *)data : "");
  }
```

and i'm expecting a ping/pong every 10 seconds, but nothing happens.

How should I enable the ping/pong ?

simonttp78, 2022-09-29:
I have same issue. Trying to find a way to close web socket client when client IP address is not responding to pings (client disconnected). keepAlivePeriod seems to be perfect solution, but it doesn't work. Any other way to close web socket for client that doesn't respond to pings?

microfoundry, 2024-08-15:
I had the same question and did some debugging. In the function:
```
void AsyncWebSocketClient::_onData(void* pbuf, size_t plen) {
...
      } else if (_pinfo.opcode == WS_PONG) {
        if (datalen != AWSC_PING_PAYLOAD_LEN || memcmp(AWSC_PING_PAYLOAD, data, AWSC_PING_PAYLOAD_LEN) != 0) {
          _server->_handleEvent(this, WS_EVT_PONG, NULL, data, datalen);
        }
...   
```     
The received data will only be queued as a WS_EVT_PONG **IF** the WS_PONG data **IS NOT** the default "ESPAsyncWebServer-PING" that is transmitted via the keepAlivePeriod() value. In my case, via wireshark (and additional Serial debugging added to, but omitted from the original code above), I see the PONG response is the same, hence no Event to queued to handle...