me-no-dev/ESPAsyncWebServer #1158: ERROR: Too many messages queued even if AsyncWebSocket::availableForWriteAll() returns true

alex-fu27, 2022-05-15:
I have a lot of data I need to send out via an `AsyncWebSocket` in multiple messages in a loop. So I thought I could wait for `AsyncWebSocket::availableForWriteAll()` to return true, then write my data:
```c++
AsyncWebSocket ws;
// ....
void dump() {
    for (int i /* etc ... */) {
        while (!ws.availableForWriteAll()) {
             vTaskDelay(1);
             /* or delay(100) tried both */
        }
        ws.textAll(/* some data */);
    }
}
```

And while I expected this to wait until all queues have enough space, I am getting "ERROR: Too many messages queued" and even a watchdog timeout:
```plain
E (20378) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (20378) task_wdt:  - async_tcp (CPU 0/1)
E (20378) task_wdt: Tasks currently running:
E (20378) task_wdt: CPU 0: IDLE0
E (20378) task_wdt: CPU 1: IDLE1
E (20378) task_wdt: Aborting.
abort() was called at PC 0x40147e6c on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x400885d8:0x3ffbfa40 0x40088855:0x3ffbfa60 0x40147e6c:0x3ffbfa80 0x40086efd:0x3ffbfaa0 0x40179347:0x3ffbcbb0 0x40149777:0x3ffbcbd0 0x4008b015:0x3ffbcbf0 0x40089866:0x3ffbcc10
  #0  0x400885d8:0x3ffbfa40 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #1  0x40088855:0x3ffbfa60 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #2  0x40147e6c:0x3ffbfa80 in task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c:252
  #3  0x40086efd:0x3ffbfaa0 in _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
  #4  0x40179347:0x3ffbcbb0 in esp_pm_impl_waiti at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/pm_esp32.c:492
  #5  0x40149777:0x3ffbcbd0 in esp_vApplicationIdleHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/freertos_hooks.c:108
  #6  0x4008b015:0x3ffbcbf0 in prvIdleTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
  #7  0x40089866:0x3ffbcc10 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

I am on an ESP32, platformio, ESP Async WebServer 1.2.3, AsyncTCP 1.1.1

Elbi2512, 2022-08-16:
I'v the same issue. The Flag of the availableForWriteAll is raised, but never cleared, at least not before the EPS reboots, with a remark that's it dit not received enought love in the Async handling routine. Looks like an interal issue/bug.

AcuarioCat, 2023-01-21:
Somewhat related..
The queue size in AsyncTCP is very small, 32 messages. Unless you are very tight on memory this can be increased substantially (I set mine to 256) as it does not use much memory.
If you increase the size you can also increase the defines in AsyncEventSource.h SSE_MAX_QUEUED_MESSAGES
and AsyncWebSocket.h WS_MAX_QUEUED_MESSAGES.
This will allow many more messages on the AsyncTCP queue and should prevent (or reduce..) any ERROR messages.

DeeEmm, 2023-01-21:
@AcuarioCat great info. Thanks.

How do you increase the memory allocation for queue size?

AcuarioCat, 2023-01-22:
The queue size is set in AsyncTCP.cpp, you need to change xQueueCreate from 32 to the size you want.

brunolalb, 2023-02-24:
I have the same issue.
I'm on Arduino IDE 2.0.3, ESP32 2.0.7, latest code from ESP Async WebServer Github, latest code from AsyncTCP Github

06GitHub, 2023-05-06:
> [[AcuarioCat] The queue size is set in AsyncTCP.cpp, you need to change xQueueCreate from 32 to the size you want.

Using ESPAsyncWebServer streaming https://gist.github.com/me-no-dev/d34fba51a8f059ac559bf62002e61aa3

Indeed changing queue size from 32 to 256 in AsyncTCP.cpp, solves "async_tcp watchdog error" for me  :

**Error**
```
The following tasks > did not reset the watchdog in time: E (705728) task_wdt: - async_tcp > (CPU 0/1)
```
**AsyncTCP.cpp OK**
```
static inline bool _init_async_event_queue(){
    if(!_async_queue){
        //_async_queue = xQueueCreate(32, sizeof(lwip_event_packet_t *));
        _async_queue = xQueueCreate(256, sizeof(lwip_event_packet_t *));  // prevents "async_tcp" Watchdog error 
        if(!_async_queue){
            return false;
        }
    }
    return true;
}
```

**Configuration**
ESP32-S3 8Mb PSRAM 16Mb Flash
Arduino IDE arduino-esp32 core 2.0.8
Wifi mode : Access Point

On Windows, AsyncTCP.cpp is located in C:\Users\USER\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp

Rysiek6, 2023-05-22:
When the AP disconnects, the AP's ESP32 continues to send Socket messages to the queue (this causes overflow).
The following Socket sending conditions eliminate this inconvenience.
(I can connect to the AP and disconnect, and the AP system does not throw any error messages)

	if (WiFi.status() != WL_CONNECTED && WiFi.softAPgetStationNum() < 1) {
		// do not send --> ERROR: Too many messages queued
	}else {
		notifyClients( String(a_kom) );  // send a message 
	}
	//----------

l-pastor, 2023-05-26:
I have the same problem,

IDE Arduino 2.1.0
ESP32 WROOM 32D 4MB

umartechboy, 2023-07-06:
Same here.
Sending a message every 100ms creates this error, sending it aafter 150ms solves it.

But after observing the behavior on my console, my hunch is that chucking for WL status should solve it. If(WiFi.status() != WL_CONNECTED).

mide1337, 2024-01-13:
The Problem is that I send a message every millisecond (or nanoseconds, depends on the esp32 cycle time). I made a timer error in my loop. Editing this line: 
```c++
_async_queue = xQueueCreate(32, sizeof(lwip_event_packet_t *));
``` 
cannot change the fact that I have sent to much messages lmao. I think that this is the problem of everyone in here??