me-no-dev/ESPAsyncWebServer #1162: IOS Safari causes watchdog trigger and heap coruption when handling requests

kramzarales, 2022-05-24:
I am developing an aplication for the ESP32 WROOM32E module. 
I found out that when more then just a couple requests are made to the ESP with Safari on IOS, the ESP will hard crash. 
For testing purposes i am using an iphone 6s runing IOS 13.3.1. (i dont believe this is a version/hardwere spacific problem)
The error that is generated is this:
_E (239040) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (239040) task_wdt:  - async_tcp (CPU 0/1)
E (239040) task_wdt: Tasks currently running:
E (239040) task_wdt: CPU 0: IDLE
E (239040) task_wdt: CPU 1: loopTask
E (239040) task_wdt: Aborting.

abort() was called at PC 0x400e7471 on core 0

Backtrace:0x400830b5:0x3ffbea2c |<-CORRUPTED
ELF file SHA256: 0000000000000000
Rebooting..._

used programing settings:
[image]

I have sometimes  encountered this issue on other browsers/hardwere, however it is especialy noticable on IOS devices using safari, where just refresing to get a simple webpage with as little as 3 requred assets will crash the system. In that regard the same code on chrome/firefox on PC and my android can handle many more requests (currently working perfectly with 20+ requests on loading and then requests every second).
When the esp is strugeling to handle the connections and requests, it will start showing the folowing errors:
[ 41919][W][AsyncTCP.cpp:969] _poll(): rx timeout 4
[187727][W][AsyncTCP.cpp:949] _poll(): pcb is NULL

When the number of requests is too much howerver, it will crash and give the mentioned error.

I have created a sample piece of code with which to reproduce the error in the attachment.
The way to reproduce this error: 
use SPIFFS to upload the files in the data folder to the ESP.
After that comment out some of the server.on requests. the code should now be working good, and the page should open(even tho there is nothing visible on it) The page can now be refreshed or opened on multiple devices simultaniously witout any issue.

If you leave all the server.on requests in there, like it is in the provided code, it will very soon overwhelm the esp and crash it. It might   load normaly for the first time, but if the page is refreshed/opened a couple times, it will crash.

[code.zip](https://github.com/me-no-dev/ESPAsyncWebServer/files/8760961/code.zip)

I also want to know if there is some sort of callback function i can do that, prevents the ESP from crashing if the connection process takes too long and it times out. Or perhaps a way to improve the performance of the code without decreasing the number of made requests?

MartinVerges, 2022-06-21:
I believe to encounter the same issue with a svelte UI on the ESP32 with ESP AsyncWebServer. Requesting each resource one by one works perfectly fine, but when the browser (in my case chrome) loads the included resources, the ESP crashes

```
E (44761) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (44761) task_wdt:  - async_tcp (CPU 01)
E(44761) task_wdt: Tasks currently running:
E (44761) task_wdt: CPU 0: IDLE
E (44761) task_wdt: CPU 1: loopTask
E (44761) task_wdt: Aborting.

abort() was called at PC 0x400ec629 on core 0

Backtrace:0x400835fd:0x3ffbea3c |<-CORRUPTED
  #0  0x400835fd:0x3ffbea3c in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:402
```

MartinVerges, 2022-06-21:
I just found out that #1167 solved my issue. Maybe it helps you too!