me-no-dev/ESPAsyncWebServer #1164: Compile error for ESP32 C3- based boards

jnule1a, 2022-05-30:
During compilation  for ESP32 C3-based boards (ESP32C3, Adafruit QT Py ESP32-C3, M5Stack Stamp C3) , the following error is displayed:
C:\Users\toto\Documents\Arduino\libraries\ESPAsyncWebServer-master\src\AsyncWebSocket.cpp: In member function 'IPAddress AsyncWebSocketClient::remoteIP()':
C:\Users\toto\Documents\Arduino\libraries\ESPAsyncWebServer-master\src\AsyncWebSocket.cpp:832:28: error: call of overloaded 'IPAddress(unsigned int)' is ambiguous
         return IPAddress(0U);
                            ^
In file included from C:\Users\toto\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\cores\esp32/Arduino.h:163,
                 from C:\Users\toto\Documents\Arduino\libraries\ESPAsyncWebServer-master\src\AsyncWebSocket.cpp:21:
C:\Users\toto\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\cores\esp32/IPAddress.h:51:5: note: candidate: 'IPAddress::IPAddress(const uint8_t*)'
     IPAddress(const uint8_t *address);
     ^~~~~~~~~
C:\Users\toto\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\cores\esp32/IPAddress.h:50:5: note: candidate: 'IPAddress::IPAddress(uint32_t)'
     IPAddress(uint32_t address);
     ^~~~~~~~~
C:\Users\toto\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\cores\esp32/IPAddress.h:29:7: note: candidate: 'constexpr IPAddress::IPAddress(const IPAddress&)'
 class IPAddress: public Printable
       ^~~~~~~~~
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\toto\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi

Test conditions:
- Arduino IDE 1.8.19
- ESP32 board manager 2.0.3, M5Stack board manager 2.0.3
- If board is non C3 e.g. ESP32 Dev module, compilation is ok
- sketch:
#include <Arduino.h>
#include <WiFi.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>

const char* ssid = "nhatoi";
const char* password = "wifi-1nhatoi";

AsyncWebServer server(80);

void setup(void) {
}

void loop(void) { 
}

HugoML, 2022-06-17:
I got the same error that relates to ESP32-C3 only.

jnule1a, 2022-06-17:
At last, I found the following work around: edit  libraries\ESPAsyncWebServer-master\src\AsyncWebSocket.cpp to replace “return IPAddress(0U)” by “return IPAddress(0ul)” or “return IPAddress((uin32_t) 0u)”

HugoML, 2022-06-18:
@jnule1a You rock!  I am not using WebSockets and changed the line as you said, and the compiler took it. :)
Thanks so much!

Zimbu98, 2022-09-19:
Typo in jnule1a's very helpful answer: return IPAddress((uin32_t) 0u) should be return IPAddress((uin**t**32_t) 0u)

Notice the little t added

Thanks jnule1a!

homonto, 2022-10-20:
it looks like this library is not maintained anymore ;-( I have the same error but thanks to you guys I solved it ;-)

RohanJoydhar, 2023-01-15:
i was working on Esp32-c3 module from speed studio. with esp_dash getting same error. after editing
l**ibraries\ESPAsyncWebServer-master\src\AsyncWebSocket.cpp to replace “return IPAddress(0U)” by “return IPAddress((uint32_t) 0u)”** i finnaly got my IP with error, it is a weather station project cunnently trying ant10 sensor module, my code:

`#include <WiFi.h>
#include <ESPAsyncWebServer.h>
#include <ESPDash.h>
#include <Ticker.h>
#include <Adafruit_AHTX0.h>

Adafruit_AHTX0 aht;
const char* ssid = "ESP32";  // Enter SSID here
 
Ticker periodicTicker;
AsyncWebServer server(80);
ESPDash dashboard(&server);  
 
Card card1(&dashboard, TEMPERATURE_CARD, "Temperature", "°C");
Card card2(&dashboard, HUMIDITY_CARD, "Humidity", "%");
 
void updateCards() {
  sensors_event_t humidity, temp;
  aht.getEvent(&humidity, &temp);
  float t = temp.temperature;
  float h = humidity.relative_humidity;
  card1.update(t);
  card2.update(h);
  dashboard.sendUpdates();
}
 
void setup() {
  Serial.begin(115200);
  aht.begin(); 
  WiFi.softAP(ssid);
  server.begin();
  delay(100);
  Serial.print("IP address: ");
  Serial.println(WiFi.softAPIP());
  periodicTicker.attach_ms(5000, updateCards);
}
 
void loop() {}
`
thnakyou @jnule1a @HugoML

pyalot, 2023-02-20:
pump

chezlog, 2023-03-17:
I had to use return IPAddress((uint32_t) 0K)

adojang, 2023-03-28:
> At last, I found the following work around: edit libraries\ESPAsyncWebServer-master\src\AsyncWebSocket.cpp to replace “return IPAddress(0U)” by “return IPAddress(0ul)” or “return IPAddress((uin32_t) 0u)”

This fixed it for me.

FranzKropp, 2023-04-23:
Great! The Answer from jnule1a was also the solution for me!

tahunus, 2023-07-06:
Since that line (832) is meant to return a 0.0.0.0 IP Address if no client is present (I think) besides @jnule1a solution, a IPAddress(0,0,0,0) also works.

RomanAlexandroff, 2023-09-13:
@jnule1a, your solution worked like a charm! Thanks a lot! =)

bag103, 2023-12-29:
Good afternoon .
I have modified the AsyncWebSocket.ccp library by replacing return IP Address((uin t 32_t) 0u); and continues with the error in any program that refers to <ESPAsyncWebServer.h>
any more ideas??
Thank you so much

chezlog, 2023-12-29:
You might try the suggestion from tahunus - just use
IP Address(0,0,0,0)

bag103, 2024-01-04:
Good morning.
  Thank you for your help in my case the solution was

//return IPAddress(0U); original
              return IPAddress((uint32_t)0); -------------------OK-------------------
		 //return IPAddress((uin t 32_t) 0u);
		//return IPAddress(0,0,0,0);