me-no-dev/ESPAsyncWebServer #1168: can not use https client after using  ESPAsyncWebServer

msojocs, 2022-06-07:
1. comment the line with `AsyncWebServer server(80);`, the program works without problems.
2. uncomment the line with `AsyncWebServer server(80);`, the program shows `[HTTPS] GET... failed, error: connection failed`
3. the https request may success at first request time.

Is there some method to fix it?

[image]

```
/**
   BasicHTTPSClient.ino

    Created on: 20.08.2018

*/

#define ASYNC_TCP_SSL_ENABLED 1
#include <Arduino.h>

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

#include <ESP8266HTTPClient.h>

#include <WiFiClientSecureBearSSL.h>

#include <ESPAsyncWebServer.h>
// Fingerprint for demo URL, expires on June 2, 2021, needs to be updated well before this date
const uint8_t fingerprint[20] = {0x29, 0xC7, 0x3D, 0xB3, 0x50, 0x36, 0x83, 0x0B, 0x90, 0xC1, 0x9C, 0xE0, 0xEF, 0x71, 0x72, 0xB1, 0x3F, 0xC7, 0x31, 0xE1};

ESP8266WiFiMulti WiFiMulti;
AsyncWebServer server(80);

void setup() {

  Serial.begin(115200);
  // Serial.setDebugOutput(true);

  Serial.println();
  Serial.println();
  Serial.println();

  for (uint8_t t = 4; t > 0; t--) {
    Serial.printf("[SETUP] WAIT %d...\n", t);
    Serial.flush();
    delay(1000);
  }

  WiFi.mode(WIFI_STA);
  WiFiMulti.addAP("CrazyBox_0457B8", "yycy418418a");
}

void loop() {
  // wait for WiFi connection
  if ((WiFiMulti.run() == WL_CONNECTED)) {

   std::unique_ptr<BearSSL::WiFiClientSecure>client(new BearSSL::WiFiClientSecure);
    // WiFiClientSecure client;

//    client->setFingerprint(fingerprint);
    // Or, if you happy to ignore the SSL certificate, then use the following line instead:
     client->setInsecure();

    HTTPClient https;

    Serial.print("[HTTPS] begin...\n");
    if (https.begin(*client, "https://jigsaw.w3.org/HTTP/connection.html")) {  // HTTPS

      Serial.print("[HTTPS] GET...\n");
      // start connection and send HTTP header
      int httpCode = https.GET();

      // httpCode will be negative on error
      if (httpCode > 0) {
        // HTTP header has been send and Server response header has been handled
        Serial.printf("[HTTPS] GET... code: %d\n", httpCode);

        // file found at server
        if (httpCode == HTTP_CODE_OK || httpCode == HTTP_CODE_MOVED_PERMANENTLY) {
          String payload = https.getString();
          Serial.println(payload);
        }
      } else {
        Serial.printf("[HTTPS] GET... failed, error: %s\n", https.errorToString(httpCode).c_str());
      }

      https.end();
    } else {
      Serial.printf("[HTTPS] Unable to connect\n");
    }
  }

  Serial.println("Wait 10s before next round...");
  delay(10000);
}
```

msojocs, 2022-06-08:
It seems that I did not find the right reason.

closed 2022-06-08 by msojocs

mizuapo, 2022-06-10:
Please do not close this issue, because this is a valid problem. I have the same issue, and I found another thread on this topic:

https://www.esp8266.com/viewtopic.php?p=75918

But this is not a good solution for me.

msojocs, 2022-06-11:
> Please do not close this issue, because this is a valid problem. I have the same issue, and I found another thread on this topic:
> 
> https://www.esp8266.com/viewtopic.php?p=75918
> 
> But this is not a good solution for me.

Sorry, I think your issue is different with mine.
In your issue, the http client is used in espasyncwebserver callback, 
but in my issue, the http client is used in the default loop.

UncleD4ve, 2022-09-16:
Same problem :/