me-no-dev/ESPAsyncWebServer #1172: AsyncEventSourceClient::_runQueue() causes core panic

rblilja, 2022-06-27:
Device: ESP32-WROOM-32E
Framework: Arduino + PlatformIO (6.0.2)
ESPAsyncWebServer @ 1.2.3
ESPAsyncTCP @ 1.1.1
espressif32 @ 4.4.0
Arduino @ 2.0.0

Dependency Graph:
|-- ESP Async WebServer @ 1.2.3+sha.f71e3d4
|   |-- AsyncTCP @ 1.1.1+sha.ca8ac5f
|   |-- FS @ 2.0.0
|   |-- WiFi @ 2.0.0
|   |-- ArduinoJson @ 6.19.4+sha.7c2ca77
|-- AsyncTCP @ 1.1.1+sha.ca8ac5f

Use case: AsyncEventSource sending JSON formatted data from FreeRTOS task (every 100 ms). ESP in softAP mode.

Issue:

Core panic occurs randomly just by having a single client connected to the event source. Not hitting refresh or alike. Error seems to originate from _runQueue().  

Debug output:

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400d7cfc  PS      : 0x00060930  A0      : 0x800d7d8c  A1      : 0x3ffd56b0  
A2      : 0x3ffd7130  A3      : 0x3ffc7aa4  A4      : 0x00000085  A5      : 0x3ffd71e4  
A6      : 0x0000000a  A7      : 0x36313031  A8      : 0x800d7cf2  A9      : 0x3ffd5690  
A10     : 0x82c8000c  A11     : 0x00000085  A12     : 0x00000085  A13     : 0x3ffd611e  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x82c80014  LBEG    : 0x4008462d  LEND    : 0x40084635  LCOUNT  : 0x00000027  

Backtrace:0x400d7cf9:0x3ffd56b00x400d7d89:0x3ffd56e0 0x400d7da7:0x3ffd5700 0x400d7df1:0x3ffd5720 0x400d49fb:0x3ffd5760 
  #0  0x400d7cf9:0x3ffd56b00 in AsyncEventSourceClient::_runQueue() at .pio/libdeps/featheresp32/ESP Async WebServer/src/AsyncEventSource.cpp:242 (discriminator 1)

rblilja, 2022-06-27:
Two more debug outputs, last one being a little different:

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400d9144  PS      : 0x00060730  A0      : 0x800d91d4  A1      : 0x3ffd5270  
A2      : 0x3ffd7a60  A3      : 0x3ffd7154  A4      : 0x0000009b  A5      : 0x3ffd6e02  
A6      : 0x0000000a  A7      : 0x0d7d3532  A8      : 0x800d913a  A9      : 0x3ffd5250  
A10     : 0xbaad5678  A11     : 0x3ffd75a3  A12     : 0x0000009b  A13     : 0x3ffd7448  
A14     : 0x00000000  A15     : 0x5678007d  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xbaad5680  LBEG    : 0x4008462d  LEND    : 0x40084635  LCOUNT  : 0x00000027  

Backtrace:0x400d9141:0x3ffd52700x400d91d1:0x3ffd52a0 0x400d91ef:0x3ffd52c0 0x400d9239:0x3ffd52e0 0x400d5e46:0x3ffd5320 
  #0  0x400d9141:0x3ffd52700 in AsyncEventSourceClient::_runQueue() at .pio/libdeps/featheresp32/ESP Async WebServer/src/AsyncEventSource.cpp:242 (discriminator 1)

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400d9142  PS      : 0x00060330  A0      : 0x800d91d4  A1      : 0x3ffd5220  
A2      : 0x3ffd71f4  A3      : 0xbaad5678  A4      : 0x0000009b  A5      : 0x3ffd7506  
A6      : 0x0000000a  A7      : 0x0d7d3537  A8      : 0x800d913a  A9      : 0x3ffd5200  
A10     : 0x0000009b  A11     : 0x3ffd711b  A12     : 0x0000009b  A13     : 0x3ffd7458  
A14     : 0x00000000  A15     : 0x5678007d  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xbaad5678  LBEG    : 0x4008462d  LEND    : 0x40084635  LCOUNT  : 0x00000027  

Backtrace:0x400d913f:0x3ffd52200x400d91d1:0x3ffd5250 0x400d91ef:0x3ffd5270 0x400d9239:0x3ffd5290 0x400d5e46:0x3ffd52d0 
  #0  0x400d913f:0x3ffd52200 in LinkedList<AsyncEventSourceMessage*, LinkedListNode>::Iterator::operator!=(LinkedList<AsyncEventSourceMessage*, LinkedListNode>::Iterator const&) const at .pio/libdeps/featheresp32/ESP Async WebServer/src/StringArray.h:54 (discriminator 1)
      (inlined by) AsyncEventSourceClient::_runQueue() at .pio/libdeps/featheresp32/ESP Async WebServer/src/AsyncEventSource.cpp:242 (discriminator 1)

bhupiister, 2022-10-13:
@rblilja Did you find any solution to it?

bhupiister, 2022-10-13:
I am facing this error

Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x40115e89  PS      : 0x00060730  A0      : 0x80115f22  A1      : 0x3f80d8a0
A2      : 0x3ffdeeec  A3      : 0x2c35322c  A4      : 0x184d0f5f  A5      : 0x00000001
A6      : 0x00000000  A7      : 0x3ffdec2c  A8      : 0x5f184d0f  A9      : 0x5f184d0f
A10     : 0x2c3534a6  A11     : 0x3ffe0cb4  A12     : 0x0000027a  A13     : 0x00000001
A14     : 0x3ffb6cb0  A15     : 0x3ffe4350  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c
EXCVADDR: 0x184d0f63  LBEG    : 0x40085369  LEND    : 0x40085371  LCOUNT  : 0x00000027

Backtrace:0x40115e86:0x3f80d8a0 |<-CORRUPTED
  #0  0x40115e86:0x3f80d8a0 in AsyncEventSourceClient::_runQueue() at .pio/libdeps/m5stack-core2/ESP Async WebServer/src/AsyncEventSource.cpp:242 (discriminator 2)

bhupiister, 2022-10-14:
Today i got the same error but on core 1

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40115f8d  PS      : 0x00060b30  A0      : 0x80116026  A1      : 0x3f80a960
A2      : 0x3ffdb5e0  A3      : 0x3ffdff06  A4      : 0x00000000  A5      : 0x00000001
A6      : 0x00000000  A7      : 0x3ffdff24  A8      : 0x00000000  A9      : 0x00000000
A10     : 0x000004f4  A11     : 0x3ffdbabc  A12     : 0x0000027a  A13     : 0x00000001
A14     : 0x3ffb6c68  A15     : 0x3ffd62b0  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000004  LBEG    : 0x40085369  LEND    : 0x40085371  LCOUNT  : 0x00000027

Backtrace:0x40115f8a:0x3f80a960 |<-CORRUPTED
  #0  0x40115f8a:0x3f80a960 in AsyncEventSourceClient::_runQueue() at .pio/libdeps/m5stack-core2/ESP Async WebServer/src/AsyncEventSource.cpp:242 (discriminator 2)

bhupiister, 2022-10-14:
@me-no-dev Any input on this will be highly appreciated.
Just for information. I am working on 2 cores on esp32 and both are sending events.send()

bhupiister, 2022-10-14:
@rblilja I guess this issue is mentioned here, along with the solution.
[Concurrent access to AsyncEventSourceClient](https://github.com/me-no-dev/ESPAsyncWebServer/issues/884
)
Library has locking mechanism for websockets but not for events. Ill try to implement the change and see if it improves the situation.

rblilja, 2022-10-14:
@bhupiister Cool. Perhaps I will find some time during the weekend to look into it. Even though I have started to move away from using Arduino libraries on the ESP.

bhupiister, 2022-10-14:
@rblilja esp32 has dedicated library for websockets and events ?

rblilja, 2022-10-14:
@bhupiister Sorry for being unclear. Not to my knowledge. I was speaking in general for my projects. Thus I decided not to spend too much time on this issue. But will look into your findings on the missing locking mechanism.

bhupiister, 2022-10-14:
@rblilja thanks

MrAwesomeRocks, 2022-12-23:
It looks like this was fixed in #887 but never merged.

I have this same issue when sending events at a rate of 1 per 10 ms.
Doesn't occur at 100 ms for me, but still a problem.