me-no-dev/ESPAsyncWebServer #1178: Fill buffer for a chunked response being called with buffer length of zero

LeeLeahy2, 2022-07-09:
Occasionally, the user routine that is used to fill the buffer for a chunked response is called with with a buffer length of zero.  Returning any data in this case causes a stack smash protect panic.  Returning zero (0) violates the interface definition since zero indicates that there is no more user data and that the response is complete.

LeeLeahy2, 2022-11-02:
Who is supporting the ESPAsyncWebServer?

ToMe25, 2023-05-24:
Have you tried returning `RESPONSE_TRY_AGAIN` as the content length?
In that case it should ask you again after a bit of time, hopefully with a bit more space in the buffer.
That would probably require something else reducing its ram usage in the meantime, but it might work.