me-no-dev/ESPAsyncWebServer #1179: Error 500 with SPIFFS

L3-N0X, 2022-07-10:
So, I want to run a html website on my Wemos D1 Mini which shows the temperature in a graph. For this I need a few scripts, which I downloaded and added to the data folder.
But now when I reload the site, most of the time there is a Error 500 for 3 random files. Is there a fix for this issue?

`server.on("/", HTTP_GET, [](AsyncWebServerRequest * request) {  
    request->send(SPIFFS, "/index.html", String(), false, processor);   
  });
  server.on("/styles.css", HTTP_GET, [](AsyncWebServerRequest * request) {
    request->send(SPIFFS, "/styles.css", "text/css");     
  });
  server.on("/script.js", HTTP_GET, [](AsyncWebServerRequest * request) { 
    request->send(SPIFFS, "/script.js", "text/js");     
  });
  server.on("/scripts/data.src.js", HTTP_GET, [](AsyncWebServerRequest * request) { 
    request->send(SPIFFS, "/scripts/data.src.js", "text/js");
  });
  server.on("/scripts/export-data.js", HTTP_GET, [](AsyncWebServerRequest * request) { 
    request->send(SPIFFS, "/scripts/export-data.js", "text/js");
  });
  server.on("/scripts/exporting.src.js", HTTP_GET, [](AsyncWebServerRequest * request) { 
    request->send(SPIFFS, "/scripts/exporting.src.js", "text/js");                    
  });
  server.on("/scripts/highcharts.js", HTTP_GET, [](AsyncWebServerRequest * request) { 
    request->send(SPIFFS, "/scripts/highcharts.js", "text/js"); 
  });
  server.on("/scripts/moment.js", HTTP_GET, [](AsyncWebServerRequest * request) { 
    request->send(SPIFFS, "/scripts/moment.js", "text/js");
  });
  server.on("/scripts/moment-timezone.js", HTTP_GET, [](AsyncWebServerRequest * request) { 
    request->send(SPIFFS, "/scripts/moment-timezone.js", "text/js"); 
  });`

[image]
[image]

When I type in the url of each file, they all load without any problems

Pablo2048, 2022-07-10:
It's probably because of too many concurrent attempts - try to use `serveStatic` instead of many `server.on` handlers. Also you can use gziped files instead of plain ones and if all this fail, use lazyloading or concat all javascript files to just one.

L3-N0X, 2022-07-10:
Thanks for your quick reply!
So, I tried serveStatic and gziped files:

```
server.on("/", HTTP_GET, [](AsyncWebServerRequest * request) {          // Bei Serveranfragen auf /
    request->send(SPIFFS, "/index.html", String(), false, processor);     // Index.html zurücksenden
  });
  server.on("/styles.css", HTTP_GET, [](AsyncWebServerRequest * request) { // Bei Anfrage auf /styles.css
    request->send(SPIFFS, "/styles.css", "text/css");                     // Die CSS Datei zurücksenden
  });
  server.on("/script.js", HTTP_GET, [](AsyncWebServerRequest * request) { // Bei Anfrage auf /script.js
    request->send(SPIFFS, "/script.js", "text/js");                       // Die JS Datei zurücksenden
  });
  server.on("/favicon.ico", HTTP_GET, [](AsyncWebServerRequest * request) { // Bei Anfrage auf /favicon.ico
    request->send(SPIFFS, "/favicon.ico", "text/ico");                    // Das Arduino-Icon zurücksenden
  });
  
  server.serveStatic("/scripts", SPIFFS, "/scripts/");

```
Is there a better way to use serveStatic? like for all files at once?

[image]

But I still get Errors like this:
[image]

Why are there now 404 Errors? Again, when I locate the file in the URL I can view the file.

I also tried packing all javascript files in one, but then I got Type Errors in the file.

Pablo2048, 2022-07-10:
:D the 404 errors are because you serve static files from `/scripts`, but they behave as in `/scripts//` for the web access. I'm using it like this
```cpp
  webs.rewrite("/", "/index.htm");
  webs.serveStatic("/", USEDFS, "/");
```

L3-N0X, 2022-07-10:
At first it looked like it works now, but then I reloaded again and these Errors occured:
[image]

The Code for loading the files is only this:
```
  server.rewrite("/", "/index.html");
  server.serveStatic("/", SPIFFS, "/");
```
and now loading the page lasts also much longer than before

Is this right? I also once again tried to combine the javascript files, but there this Error occured:
[image]

L3-N0X, 2022-07-10:
So, finally I´ve got it fixed!

I copied the single js files in one big file. You need the right order for the files and every file needs to be seperatet with a semicolon (In my case the files all ended already with a semicolon). Then I used [these](https://www.mischianti.org/2020/10/26/web-server-with-esp8266-and-esp32-byte-array-gzipped-pages-and-spiffs-2/)  instructions to compress my js file to a gzip. Then I used the converter on the webiste to make a filearray and copied it into a seperate file (scripts.h).
Send the array:
```
  server.on("/all.js", HTTP_GET, [] (AsyncWebServerRequest * request) {
    AsyncWebServerResponse *response = request->beginResponse_P(200, "text/js", all_js_gz, all_js_gz_len);
    response->addHeader("Content-Encoding", "gzip");
    request->send(response);
  });
```

And then it worked, without any Errors!

closed 2022-07-10 by L3-N0X