me-no-dev/ESPAsyncWebServer #1181: Crashes when using WebSockets with EXP32-C3

alwynallan, 2022-07-11:
Examples that work well with other ESP32 platforms crash the ESP32-C3 (single RISC-V core). The chip reboots whenever a client (browser) tries to connect. I get the message
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40381fa8
SPIWP:0xee
```
The bug is documented elsewhere (e.g. Vasilis comment https://randomnerdtutorials.com/esp32-websocket-server-arduino/), and resolved by eliminating `delete it` from line 131 in StringArray.h . I also had the "ambiguous" compiler error documented here with ESP32-C3.

vortigont, 2022-07-14:
Check this PR #952 for a proper fix

matthew798, 2022-09-08:
The `delete it` fix just saved me hours of frustration. What is the issue here?