me-no-dev/ESPAsyncWebServer #1182: It's not work with wifimanager

sunnyguhz, 2022-07-18:
Good day.
It's not work with wifimanager when I need.

`C:\Users\Sunnygu\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:62:18: error: redeclaration of 'HTTP_GET' HTTP_GET = 0b00000001, ^ In file included from C:\Users\Sunnygu\Documents\Arduino\libraries\CustomWiFiManagerRY/CustomWiFiManagerRY.h:17:0, from C:\Users\Sunnygu\Desktop\RY103\RY103.ino:12: C:\Users\Sunnygu\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WebServer\src/ESP8266WebServer.h:34:29: note: previous declaration 'HTTPMethod HTTP_GET' enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS }; ^ In file included from C:\Users\Sunnygu\Documents\Arduino\libraries\ESPUI\src/ESPUI.h:24:0, from C:\Users\Sunnygu\Desktop\RY103\RY103.ino:23: C:\Users\Sunnygu\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:63:18: error: redeclaration of 'HTTP_POST' HTTP_POST = 0b00000010, ^ In file included from C:\Users\Sunnygu\Documents\Arduino\libraries\CustomWiFiManagerRY/CustomWiFiManagerRY.h:17:0, from C:\Users\Sunnygu\Desktop\RY103\RY103.ino:12: C:\Users\Sunnygu\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WebServer\src/ESP8266WebServer.h:34:50: note: previous declaration 'HTTPMethod HTTP_POST' enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS }; ^ In file included from C:\Users\Sunnygu\Documents\Arduino\libraries\ESPUI\src/ESPUI.h:24:0, from C:\Users\Sunnygu\Desktop\RY103\RY103.ino:23: C:\Users\Sunnygu\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:64:18: error: redeclaration of 'HTTP_DELETE' HTTP_DELETE = 0b00000100, ^ In file included from C:\Users\Sunnygu\Documents\Arduino\libraries\CustomWiFiManagerRY/CustomWiFiManagerRY.h:17:0, from C:\Users\Sunnygu\Desktop\RY103\RY103.ino:12: C:\Users\Sunnygu\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WebServer\src/ESP8266WebServer.h:34:83: note: previous declaration 'HTTPMethod HTTP_DELETE' enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS }; ^ In file included from C:\Users\Sunnygu\Documents\Arduino\libraries\ESPUI\src/ESPUI.h:24:0, from C:\Users\Sunnygu\Desktop\RY103\RY103.ino:23: C:\Users\Sunnygu\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:65:18: error: redeclaration of 'HTTP_PUT' HTTP_PUT = 0b00001000, ^ In file included from C:\Users\Sunnygu\Documents\Arduino\libraries\CustomWiFiManagerRY/CustomWiFiManagerRY.h:17:0, from C:\Users\Sunnygu\Desktop\RY103\RY103.ino:12: C:\Users\Sunnygu\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WebServer\src/ESP8266WebServer.h:34:61: note: previous declaration 'HTTPMethod HTTP_PUT' enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS }; ^ In file included from C:\Users\Sunnygu\Documents\Arduino\libraries\ESPUI\src/ESPUI.h:24:0, from C:\Users\Sunnygu\Desktop\RY103\RY103.ino:23: C:\Users\Sunnygu\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:66:18: error: redeclaration of 'HTTP_PATCH' HTTP_PATCH = 0b00010000, ^ In file included from C:\Users\Sunnygu\Documents\Arduino\libraries\CustomWiFiManagerRY/CustomWiFiManagerRY.h:17:0, from C:\Users\Sunnygu\Desktop\RY103\RY103.ino:12: C:\Users\Sunnygu\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WebServer\src/ESP8266WebServer.h:34:71: note: previous declaration 'HTTPMethod HTTP_PATCH' enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS }; ^ In file included from C:\Users\Sunnygu\Documents\Arduino\libraries\ESPUI\src/ESPUI.h:24:0, from C:\Users\Sunnygu\Desktop\RY103\RY103.ino:23: C:\Users\Sunnygu\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:67:18: error: redeclaration of 'HTTP_HEAD' HTTP_HEAD = 0b00100000, ^ In file included from C:\Users\Sunnygu\Documents\Arduino\libraries\CustomWiFiManagerRY/CustomWiFiManagerRY.h:17:0, from C:\Users\Sunnygu\Desktop\RY103\RY103.ino:12: C:\Users\Sunnygu\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WebServer\src/ESP8266WebServer.h:34:39: note: previous declaration 'HTTPMethod HTTP_HEAD' enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS }; ^ In file included from C:\Users\Sunnygu\Documents\Arduino\libraries\ESPUI\src/ESPUI.h:24:0, from C:\Users\Sunnygu\Desktop\RY103\RY103.ino:23: C:\Users\Sunnygu\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:68:18: error: redeclaration of 'HTTP_OPTIONS' HTTP_OPTIONS = 0b01000000, ^ In file included from C:\Users\Sunnygu\Documents\Arduino\libraries\CustomWiFiManagerRY/CustomWiFiManagerRY.h:17:0, from C:\Users\Sunnygu\Desktop\RY103\RY103.ino:12: C:\Users\Sunnygu\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WebServer\src/ESP8266WebServer.h:34:96: note: previous declaration 'HTTPMethod HTTP_OPTIONS' enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS }; ^ In file included from C:\Users\Sunnygu\Documents\Arduino\libraries\ESPUI\src/ESPUI.h:24:0, from C:\Users\Sunnygu\Desktop\RY103\RY103.ino:23: C:\Users\Sunnygu\Documents\Arduino\libraries\ESPAsyncWebServer-master\src/ESPAsyncWebServer.h:69:18: error: redeclaration of 'HTTP_ANY' HTTP_ANY = 0b01111111, ^ In file included from C:\Users\Sunnygu\Documents\Arduino\libraries\CustomWiFiManagerRY/CustomWiFiManagerRY.h:17:0, from C:\Users\Sunnygu\Desktop\RY103\RY103.ino:12: C:\Users\Sunnygu\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WebServer\src/ESP8266WebServer.h:34:19: note: previous declaration 'HTTPMethod HTTP_ANY' enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_HEAD, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS }; ^ C:\Users\Sunnygu\Desktop\RY103\RY103.ino: In function 'void callback(char*, byte*, unsigned int)': C:\Users\Sunnygu\Desktop\RY103\RY103.ino:610:20: warning: 'BUILTIN_LED' is deprecated (declared at C:\Users\Sunnygu\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4\variants\nodemcu/../generic/common.h:79): use LED_BUILTIN [-Wdeprecated-declarations] digitalWrite(BUILTIN_LED, HIGH); // Turn the LED off by making the voltage HIGH`

sfranzyshen, 2022-07-18:
https://github.com/alanswx/ESPAsyncWiFiManager

sunnyguhz, 2022-07-25:
> https://github.com/alanswx/ESPAsyncWiFiManager

Think you for your help.It is works.Thinks.

closed 2022-07-25 by sunnyguhz