me-no-dev/ESPAsyncWebServer #1183: Can't enable SSL for ESP32: error: 'AcSSlFileHandler' has not been declared

strocode, 2022-07-18:
Hi,

I'm trying to enable SSL for ESP32 - perhaps this isn't supported? I've scraped together a few hints from various ports. 

I'm using platformio and versions described below.

The error I get on compile is:
```
In file included from src/main.cpp:17:
.pio/libdeps/nodemcu-32s/ESP Async WebServer/src/ESPAsyncWebServer.h:412:27: error: 'AcSSlFileHandler' has not been declared
     void onSslFileRequest(AcSSlFileHandler cb, void* arg);
                           ^~~~~~~~~~~~~~~~
Compiling .pio/build/nodemcu-32s/libea0/WiFi/WiFiUdp.cpp.o
src/main.cpp: In function 'void initSecureWebserver()':
src/main.cpp:85:10: error: invalid user-defined conversion from 'initSecureWebserver()::<lambda(void*, const char*, uint8_t**)>' to 'int' [-fpermissive]
   }, NULL);
          ^
src/main.cpp:69:82: note: candidate is: 'initSecureWebserver()::<lambda(void*, const char*, uint8_t**)>::operator int (*)(void*, const char*, uint8_t**)() const' <near match>
   server.onSslFileRequest([](void * arg, const char *filename, uint8_t **buf) -> int {
                                                                                  ^~~
src/main.cpp:69:82: note:   no known conversion from 'int (*)(void*, const char*, uint8_t**)' {aka 'int (*)(void*, const char*, unsigned char**)'} to 'int'
In file included from src/main.cpp:17:
.pio/libdeps/nodemcu-32s/ESP Async WebServer/src/ESPAsyncWebServer.h:412:10: note:   initializing argument 1 of 'void AsyncWebServer::onSslFileRequest(int, void*)'
     void onSslFileRequest(AcSSlFileHandler cb, void* arg);
```

Code is:
```
// Tried to Add SSL but it didn't work
// .pio/libdeps/nodemcu-32s/ESP Async WebServer/src/ESPAsyncWebServer.h:412:27: error: 'AcSSlFileHandler' has not been declared
#define ASYNC_TCP_SSL_ENABLED 1 

#include <Arduino.h>

#define WIFI_SSID "WIFI_SSID_1"
#define WIFI_PASSWORD "PASSWORD_FOR_WIFI_SSID_1"
#include <WiFiMulti.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <ESPmDNS.h>
#include "SPIFFS.h"

WiFiMulti wifiMulti;
AsyncWebServer server(443);

void initWifi()
{
    // Setup wifi
  WiFi.mode(WIFI_STA);
  wifiMulti.addAP(WIFI_SSID, WIFI_PASSWORD);
  while (wifiMulti.run() != WL_CONNECTED) {
    Serial.print(".");
    delay(100);
  }
  Serial.print(' ');
  Serial.println(WiFi.localIP());
}

// Tried to add SSL but it didn't work
// Need to add this; https://github.com/me-no-dev/ESPAsyncWebServer/issues/75
void initSecureWebserver()
{
    // Web Server Root URL
  server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(SPIFFS, "/index.html", "text/html");
  });

  server.serveStatic("/", SPIFFS, "/");

  server.onSslFileRequest([](void * arg, const char *filename, uint8_t **buf) -> int {
    Serial.printf("SSL File: %s\n", filename);
    File file = SPIFFS.open(filename, "r");
    if(file){
      size_t size = file.size();
      uint8_t * nbuf = (uint8_t*)malloc(size);
      if(nbuf){
        size = file.read(nbuf, size);
        file.close();
        *buf = nbuf;
        return size;
      }
      file.close();
    }
    *buf = 0;
    return 0;
  }, NULL);

  // Start server
  server.beginSecure("/Cert.pem", "/Key.pem", NULL);
}

// Initialize SPIFFS
void initSPIFFS() {
  if (!SPIFFS.begin()) {
    Serial.println("An error has occurred while mounting SPIFFS");
  }
  Serial.println("SPIFFS mounted successfully");
}

void setup() {
  Serial.begin(115200);
  initWifi();
  initSPIFFS();
  initSecureWebserver();
}

void loop() {
  Serial.println("Wait 10s");
  delay(10000);
}
```

platformio.ini is:
```
[env:nodemcu-32s]
platform = espressif32
board = nodemcu-32s
framework = arduino
monitor_speed = 115200
lib_deps = 
	https://github.com/me-no-dev/ESPAsyncWebServer.git # need latest version as the ESP Async WebServer has bugs
```

Dependencies are:
```
> Executing task in folder AsyncServerSSLTEst: platformio pkg list --environment nodemcu-32s <

Resolving nodemcu-32s dependencies...
Platform espressif32 @ 5.0.0 (required: espressif32)
├── framework-arduinoespressif32 @ 3.20003.220626 (required: platformio/framework-arduinoespressif32 @ ~3.20003.0)
├── tool-esptoolpy @ 1.30300.0 (required: platformio/tool-esptoolpy @ ~1.30300.0)
├── tool-mkfatfs @ 2.0.1 (required: platformio/tool-mkfatfs @ ~2.0.0)
├── tool-mklittlefs @ 1.203.210628 (required: platformio/tool-mklittlefs @ ~1.203.0)
├── tool-mkspiffs @ 2.230.0 (required: platformio/tool-mkspiffs @ ~2.230.0)
└── toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3 (required: espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3)

Libraries
└── ESP Async WebServer @ 1.2.3+sha.f71e3d4 (required: git+https://github.com/me-no-dev/ESPAsyncWebServer.git)
│   └── AsyncTCP @ 1.1.1 (required: me-no-dev/AsyncTCP @ ^1.1.1)
```

ZanzyTHEbar, 2022-11-08:
Did you ever fix this? I have the same issue `AcSSlFileHandler has not been declared` - that's the only error i get on compile. Everything else works for me. 

I have defined `#define ASYNC_TCP_SSL_ENABLED 1` at the top level before my includes
and generated my key and cert files.

strocode, 2022-11-13:
No, never managed to fix it.

ZanzyTHEbar, 2022-11-13:
damn - i wonder why the devs have not added support for it.

dmarc1234, 2022-11-17:
It would be really good to get this supported as we have been using ESPAsyncWebServer for a number of years and the consensus of opinion is that webservers without security are a thing of the past.
@me-no-dev any chance this could be done in the near future ?

playmiel, 2023-02-03:
use https://github.com/yubox-node-org/AsyncTCPSock.git and https://github.com/yubox-node-org/ESPAsyncWebServer
remove:` {
      "owner": "me-no-dev",
      "name": "ESPAsyncTCP",
      "version": "^1.2.2",
      "platforms": "espressif8266"
    },
    {
      "owner": "me-no-dev",      
      "name": "AsyncTCP",
      "version": "^1.1.1",
      "platforms": "espressif32"
    },` in library.json of ESPAsyncWebServer

dmarc1234, 2023-02-03:
Hello, great to see that this is being looked at, thank you.

However I am a little confused, beginSecure() will now compile, as its included within AsyncTCP.h, but there is still no underlying code within AsyncTCP.cpp to start the process ie. there is no beginSecure() routine within AsyncTCP.cpp

What am I missing ?

playmiel, 2023-02-03:
actually I thought the ssl part was added with AsyncTCPSock but no you can use this example https://github.com/Bmooij/AsyncTCP-https-server-example
otherwise or if you can merge this example with the current AsyncTCPSock

dmarc1234, 2023-02-06:
Unfortunately not and the example provided does not include it either.

Digging a little deeper it looks like the ESP8266 version, ESPAsyncTCP, does include beginSecure() but the ESP32 version, AsyncTCP, does not so this functionality has never been implemented in this version of the library.

Not sure what to try next, any ideas ?

chrisdiphoorn, 2023-02-09:
> 

Maybe This library - AsyncTCP_SSL

playmiel, 2023-02-09:
this library is for esp32 and includes beginsecure :https://github.com/Bmooij/AsyncTCP.git

playmiel, 2023-02-09:
follow the configuration of the example https://github.com/Bmooij/AsyncTCP-https-server-example with 
```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
build_flags =
    -DCORE_DEBUG_LEVEL=5
    -DASYNC_TCP_SSL_ENABLED
board_build.embed_txtfiles =
    example.crt
    example.key
lib_deps =
    https://github.com/Bmooij/AsyncTCP.git#mbed-tls
    ESP Async WebServer@1.2.3
``` 
in platformio.ini
avec les indications que j'ai donner ici
> utilisez https://github.com/yubox-node-org/AsyncTCPSock.git et https://github.com/yubox-node-org/ESPAsyncWebServer supprimer : ` { "owner": "me-no-dev", "name": "ESPAsyncTCP", "version": "^1.2.2", "platforms": "espressif8266" }, { "owner": "me-no-dev", "name": "AsyncTCP", "version": "^1.1.1", "platforms": "espressif32" },`dans library.json de ESPAsyncWebServer

playmiel, 2023-02-09:
I tested and the code works but it is slow and unstable

justbendev, 2023-02-10:
### Hello :wave: 

You (@playmiel) are saying that [yubox-node-org/AsyncTCPSock](https://github.com/yubox-node-org/AsyncTCPSock.git) have an SSL implementation, that part i get it and see it in the code.

> I tested and the code works but it is slow and unstable

Are you saying that the ESPAsyncWebServer fork of yubox is working with SSL ?

Because i have read the code for the last 15 mins and i'am pretty sure SSL is not implemented at all. 

It will compile fine but its just an empty function so nothing will happen if you start your server with BeginSecure even with yubox-node-org/ESPAsyncWebServer fork.

Because this is the definition of the function beginSecure in yubox-node-org/AsyncTCPSock 
```cpp
#if ASYNC_TCP_SSL_ENABLED
    // Dummy, so it compiles with ESP Async WebServer library enabled.
    void onSslFileRequest(AcSSlFileHandler cb, void* arg) {};
    void beginSecure(const char *cert, const char *private_key_file, const char *password) {};
#endif
```

So basically [Bmooij/AsyncTCP-https-server-example](https://github.com/Bmooij/AsyncTCP-https-server-example/blob/master/src/main.cpp) will do nothing since it's executing an empty function.

If i missed something please let me know :+1: 

As @dmarc1234 said 
> It would be really good to get this supported as we have been using ESPAsyncWebServer for a number of years and the consensus of opinion is that webservers without security

In fact, it is mandatory for all **Secure** IOT Devices.

---

#### Have a great day.

chrisdiphoorn, 2023-02-11:
I have had no luck getting ssl working on esp32... Gone back to http for the moment.

playmiel, 2023-02-11:
use this :https://github.com/Bmooij/AsyncTCP.git there is ssl even if it is slow then follow this example :https://github.com/Bmooij/AsyncTCP-https-server-example with what I said above as indication

justbendev, 2023-02-11:
## Alternative Library
@chrisdiphoorn This is currently the best ESP32 HTTPS Server library [HTTPS_Server_Generic](https://github.com/khoih-prog/HTTPS_Server_Generic) it's based on [esp32_https_server](https://github.com/fhessel/esp32_https_server) but without the bugs (**Do not use esp32_https_server**)

playmiel, 2023-02-13:
hello if possible I will make a complete example to show you that it really works , It's true that the explanations are bad, sorry

chrisdiphoorn, 2023-02-14:
ive had a look at the code - looks fine and i will need to change mycode to get this working....
but still need to investigate the missing processor function.... where i currently pass values back into the HTML code as the page is rendered in the DOM.

server.on("/config.html", HTTP_GET, [](AsyncWebServerRequest *request){
      request->send(SPIFFS, "/config.html", String(), false, processor);
    });

String processor(const String& var){

      if (var == "DEVID"){
        return String(DEVflash.id);
      }
      if (var == "DEV") {
        return String(DEVflash.dev_name);
      }
      if (var == "DEV_USER" ){
        return String(DEVflash.dev_username);
      }
... Replaces "$DEVID$" witht the DEVflash.id value.

Is this the same as using this.. (&handleSwitch)?
 ResourceNode * nodeSwitch = new ResourceNode("/led/*/*", "POST", &handleSwitch);

A little more complicated than what im currently using....

Regards

Chris Diphoorn

dmarc1234, 2023-02-21:
So I think I am getting a little closer, the branch https://github.com/Bmooij/AsyncTCP/tree/mbed-tls does include the function beginSecure() within AsyncTCP.cpp and there are two other files tcp_mbedtls.c and tcp_mbedtls.h that are included for TLS support.
The only problem is that when trying to compile you get two errors as follows:

tcp_mbedtls.cpp:760: undefined reference to _tcp_write4ssl(tcp_pcb*, char const*, unsigned int, unsigned char, void*)
tcp_mbedtls.cpp:760: undefined reference to _tcp_output4ssl(tcp_pcb*, void*)

I think these are due to the jump pad definitions in AsyncTCP.cpp as follows:

```
// Jump pads for _tcp_*4ssl function below to get access to _closed_slot.
// I'm sure there has to be a better way to do this...

esp_err_t AsyncClient::_tcp_output4ssl(tcp_pcb * pcb) {
    return _tcp_output(pcb, _closed_slot);
}

esp_err_t AsyncClient::_tcp_write4ssl(tcp_pcb * pcb, const char* data, size_t size, uint8_t apiflags) {
    return _tcp_write(pcb, _closed_slot, data, size, apiflags);
}
```
Does anyone have any ideas ?

playmiel, 2023-02-21:
> use https://github.com/yubox-node-org/AsyncTCPSock.git and https://github.com/yubox-node-org/ESPAsyncWebServer remove:` { "owner": "me-no-dev", "name": "ESPAsyncTCP", "version": "^1.2.2", [...]

In fact there is a conflict with the dependencies of asyncwebserver as I say it is necessary to remove it from the library.json file of asyncwebserver otherwise it changes the version of asynctcp that you have by asynctcp 1.1

playmiel, 2023-02-21:
after that delete the two asynctcp then reinstall the version of Bmooij

dmarc1234, 2023-02-21:
That library, as discussed above does not work, it does not have beginSecure() declared and therefore does nothing.

playmiel, 2023-02-21:
oh sorry I misspoke, I meant to remove { "owner": "me-no-dev", "name": "ESPAsyncTCP", "version": "^1.2. 2", "platforms": "espressif8266" }, { "owner": "me-no-dev", "name": "AsyncTCP", "version": "^1.1.1", "platforms": "espressif32" }, in library.json of ESPAsyncWebServer (from me-no-dev) and after that delete the two asynctcp then reinstall the version of Bmooij

Adrianotiger, 2023-02-27:
The undeclared variable is declared in this library:
[ESPAsyncTCP.h](https://github.com/me-no-dev/ESPAsyncTCP/blob/15476867dcbab906c0f1d47a7f63cdde223abeab/src/ESPAsyncTCP.h)

So, I think it is enough to include `ESPAsyncTCP.h`.

I am just trying to make it work too (with sockets). I would like to use the default libraries.

dmarc1234, 2023-03-21:
Getting closer, can now get https://github.com/Bmooij/AsyncTCP/tree/mbed-tls to compile when added the standard ESPAsyncWebServer and have the HTTPs server starting:

```
T network_event- Loading the server cert
T network_event- Loading the server key
T network_event- Seeding the random number generator...
T network_event- Setting up the SSL data...
T network_event- tcp_ssl_new_server completed succesfully
```
 and delivering simple XML and HTML:

```
T async_tcp- tcp_ssl_read(3ffe2b70, 3fff9b30)
T async_tcp- start handshake: 0
T async_tcp- tcp_ssl_recv: ctx: 0x3FFF0760, buf: 0x3FFF0E98, len: 5
T async_tcp- tcp_ssl_recv: len: 5, recv_len: 5, pbuf_offset: 0, tcp_pbuf len: 517.
T async_tcp- tcp_ssl_recv: ctx: 0x3FFF0760, buf: 0x3FFF0E9D, len: 512
T async_tcp- tcp_ssl_recv: len: 512, recv_len: 512, pbuf_offset: 5, tcp_pbuf len: 517.
T async_tcp- tcp_ssl_send: ctx: 0x3FFF0760, buf: 0x3FFF4FF8, len: 96
T async_tcp- tcp_ssl_send: tcp_write(3ffe2b70, 3fff4ff8, 96, 3fff0644)
T async_tcp- tcp_ssl_send: tcp_output: 96 / 96
T async_tcp- tcp_ssl_send: ctx: 0x3FFF0760, buf: 0x3FFF4FF8, len: 1025
T async_tcp- tcp_ssl_send: tcp_write(3ffe2b70, 3fff4ff8, 1025, 3fff0644)
T async_tcp- tcp_ssl_send: tcp_output: 1025 / 1025
T async_tcp- tcp_ssl_send: ctx: 0x3FFF0760, buf: 0x3FFF4FF8, len: 406
T async_tcp- tcp_ssl_send: tcp_write(3ffe2b70, 3fff4ff8, 406, 3fff0644)
T async_tcp- tcp_ssl_send: tcp_output: 406 / 406
T async_tcp- tcp_ssl_send: ctx: 0x3FFF0760, buf: 0x3FFF4FF8, len: 9
T async_tcp- tcp_ssl_send: tcp_write(3ffe2b70, 3fff4ff8, 9, 3fff0644)
T async_tcp- tcp_ssl_send: tcp_output: 9 / 9
T async_tcp- tcp_ssl_recv: ctx: 0x3FFF0760, buf: 0x3FFF0E98, len: 5
T async_tcp- tcp_ssl_recv: len: 5, recv_len: 0, pbuf_offset: 517, tcp_pbuf len: 517.
T async_tcp- tcp_ssl_read: return total_bytes: 0
T async_tcp- tcp_ssl_read(3ffe2b70, 3fffaf44)
T async_tcp- start handshake: 8
T async_tcp- tcp_ssl_recv: ctx: 0x3FFF0760, buf: 0x3FFF0E98, len: 5
T async_tcp- tcp_ssl_recv: len: 5, recv_len: 5, pbuf_offset: 0, tcp_pbuf len: 194.
T async_tcp- tcp_ssl_recv: ctx: 0x3FFF0760, buf: 0x3FFF0E9D, len: 138
T async_tcp- tcp_ssl_recv: len: 138, recv_len: 138, pbuf_offset: 5, tcp_pbuf len: 194.
T async_tcp- tcp_ssl_recv: ctx: 0x3FFF0760, buf: 0x3FFF0E98, len: 5
T async_tcp- tcp_ssl_recv: len: 5, recv_len: 5, pbuf_offset: 143, tcp_pbuf len: 194.
T async_tcp- tcp_ssl_recv: ctx: 0x3FFF0760, buf: 0x3FFF0E9D, len: 1
T async_tcp- tcp_ssl_recv: len: 1, recv_len: 1, pbuf_offset: 148, tcp_pbuf len: 194.
T async_tcp- tcp_ssl_recv: ctx: 0x3FFF0760, buf: 0x3FFF0E98, len: 5
T async_tcp- tcp_ssl_recv: len: 5, recv_len: 5, pbuf_offset: 149, tcp_pbuf len: 194.
T async_tcp- tcp_ssl_recv: ctx: 0x3FFF0760, buf: 0x3FFF0E9D, len: 40
T async_tcp- tcp_ssl_recv: len: 40, recv_len: 40, pbuf_offset: 154, tcp_pbuf len: 194.
T async_tcp- tcp_ssl_send: ctx: 0x3FFF0760, buf: 0x3FFF4FF8, len: 6
T async_tcp- tcp_ssl_send: tcp_write(3ffe2b70, 3fff4ff8, 6, 3fff0644)
T async_tcp- tcp_ssl_send: tcp_output: 6 / 6
T async_tcp- tcp_ssl_send: ctx: 0x3FFF0760, buf: 0x3FFF4FF8, len: 45
T async_tcp- tcp_ssl_send: tcp_write(3ffe2b70, 3fff4ff8, 45, 3fff0644)
T async_tcp- tcp_ssl_send: tcp_output: 45 / 45
T async_tcp- Protocol is TLSv1.2 Ciphersuite is TLS-ECDHE-RSA-WITH-AES-256-GCM-SHA384
T async_tcp- Verifying peer X.509 certificate...T async_tcp- handshake error: 0
T async_tcp- 
T async_tcp- MbedTLS message code: 0
T async_tcp- tcp_ssl_read: return total_bytes: 0
T async_tcp- tcp_ssl_read(3ffe2b70, 3fff92c0)
T async_tcp- tcp_ssl_recv: ctx: 0x3FFF0760, buf: 0x3FFF0E98, len: 5
T async_tcp- tcp_ssl_recv: len: 5, recv_len: 5, pbuf_offset: 0, tcp_pbuf len: 750.
T async_tcp- tcp_ssl_recv: ctx: 0x3FFF0760, buf: 0x3FFF0E9D, len: 745
T async_tcp- tcp_ssl_recv: len: 745, recv_len: 745, pbuf_offset: 5, tcp_pbuf len: 750.
T async_tcp- tcp_ssl_read: read_bytes: 721, total_bytes: 0, tot_len: 750, pbuf_offset: 750
T async_tcp- tcp_ssl_write(3ffe2b70, 3fffa704, len=1492)
T async_tcp- tcp_ssl_send: ctx: 0x3FFF0760, buf: 0x3FFF4FF8, len: 1521
T async_tcp- tcp_ssl_send: tcp_write(3ffe2b70, 3fff4ff8, 1521, 3fff0644)
T async_tcp- tcp_ssl_send: tcp_output: 1521 / 1521
T async_tcp- tcp_ssl_recv: ctx: 0x3FFF0760, buf: 0x3FFF0E98, len: 5
T async_tcp- tcp_ssl_recv: len: 5, recv_len: 0, pbuf_offset: 750, tcp_pbuf len: 750.
[... 25 lines cut ...]
E (79943) task_wdt: Tasks currently running:
E (79943) task_wdt: CPU 0: IDLE0
E (79943) task_wdt: CPU 1: IDLE1
E (79943) task_wdt: Aborting.
abort() was called at PC 0x400facbc on core 0

Backtrace: 0x4008e3f8:0x3ffbe170 0x4008e629:0x3ffbe190 0x400facbc:0x3ffbe1b0 0x40085815:0x3ffbe1d0 0x401b37a7:0x3ffbc120 0x400fc7f6:0x3ffbc140 0x4008c2a5:0x3ffbc160 0x4008aab1:0x3ffbc180
```
Backtrace is a little non-descript:

```
Decoding stack results
0x4008e3f8: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x4008e629: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x400facbc: task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c line 174
0x401b37a7: esp_pm_impl_waiti at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/pm_esp32.c line 492
0x400fc7f6: esp_vApplicationIdleHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/freertos_hooks.c line 63
[... cut ...] unner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 3382
0x4008aab1: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```
But looks like a memory corruption issue within SSL that's causing the WDT timeout.

Possible buffer issue so has anyone any idea where the allocations are done.

Very nearly there and page delivery speed is not too bad so definitely usable.

playmiel, 2023-04-11:
I tested things to try to correct these problems but it does not give anything, even increasing the memory for the processing of the asyntcp task it does not work, in my opinion it should completely review the processing mbedtls

kbhuinfo, 2023-06-20:
> I tested things to try to correct these problems but it does not give anything, even increasing the memory for the processing of the asyntcp task it does not work, in my opinion it should completely review the processing mbedtls

Any other directions or suggestions?

dgtized, 2023-08-10:
@justbendev The Bmooji AsyncTCP fork *does* have an implementation of [beginSecure](https://github.com/Bmooij/AsyncTCP/blob/mbed-tls/src/AsyncTCP.cpp#L1629-L1671) on the `mbed-tls` branch. 
I believe https://github.com/me-no-dev/AsyncTCP/pull/90 shows all the changes on that fork. I haven't experimented to see how it works (though it looks like @dmarc1234 may have had some luck), but @playmiel is correct that there is at least a partial implementation on the branch. 

For my project, I needed a secure context to access the browser WebMIDI api to ship data to an ESP32. I was able to standup a simple esp-idf/https web server in parallel with my existing ESPAsyncWebServer. I cribbed an implementation from https://github.com/espressif/esp-idf/blob/master/examples/protocols/https_server/simple/main/main.c, and https://github.com/espressif/esp-idf/blob/master/examples/protocols/http_server/file_serving/main/file_server.c, and then fixed the CORS errors so that code hosted on the https server could interface with the XML-RPC & websocket API I had on the async http server. It's not the nicest solution but it did let me avoid digging into TLS implementations in AsyncTCP, while only requiring duplication of a handful of endpoints on the esp-idf https web server.

SavageTee, 2024-03-27:
Any updates on this

jahm86, 2024-03-28:
> Any updates on this

Yes, the solution exposed by me up, worked for me.

SavageTee, 2024-03-28:
> > Any updates on this
> 
> Yes, the solution exposed by me up, worked for me.
> 
> Do you need help?

Is this fix for websocket or https webserver

SavageTee, 2024-03-28:
> > Any updates on this
> 
> Yes, the solution exposed by me up, worked for me.
> 
> Do you need help?

c:/users/imtee/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\imTee\AppData\Local\Temp\arduino\sketches\A8587B274BF647287BBACD5EAD456938\sketch\sketch_mar26a.ino.cpp.o:(.literal._Z5setupv+0x48): undefined reference to `AsyncWebServer::beginSecure(char const*, char const*, char const*)'
c:/users/imtee/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\imTee\AppData\Local\Temp\arduino\sketches\A8587B274BF647287BBACD5EAD456938\sketch\sketch_mar26a.ino.cpp.o: in function `setup()':
C:\Users\imTee\Desktop\TeeCodeSmart\sketch_mar26a/sketch_mar26a.ino:71: undefined reference to `AsyncWebServer::beginSecure(char const*, char const*, char const*)'
collect2.exe: error: ld returned 1 exit status
Multiple libraries were found for "WiFi.h"
  Used: C:\Users\imTee\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\WiFi
  Not used: C:\Users\imTee\Documents\Arduino\libraries\WiFi
exit status 1

Compilation error: exit status 1

jahm86, 2024-03-28:
> > > Any updates on this
> > 
> > 
> > Yes, the solution exposed by me up, worked for me.
> > Do you need help?
> 
> Is this fix for websocket or https webserver

I applied this on a MQTT web server, in which i enabled SSL.

SavageTee, 2024-03-28:
> > I got a solution that **just works**! I tested a MQTT client in ESP32 with ssl enabled and it's working fine!
> > 
> > Platformio installs the library in the directory "_[projectPath]/.pio/libdeps/[envName]/ESP Async WebServer_". Any file explained will be relative to this path.
> > 
> > First, i added to the file _./library.json_, inside "_dependencies_" the option:
> > 
> > ```json
> > {
> >   "owner": "khoih-prog",
> >   "name": "AsyncTCP_SSL",
> >   "version": ">=1.3.1",
> >   "platforms": "espressif32"
> > }
> > ```
> > 
> > Then, i modified the file _./src/ESPAsyncWebServer.h_. I replaced the line 33:
> > 
> > ```c
> > #include <AsyncTCP.h>
> > ```
> > 
> > With this macro:
> > 
> > ```c
> > #if ASYNC_TCP_SSL_ENABLED
> >     #include <AsyncTCP_SSL.h>
> >     #define AcSSlFileHandler AcSSlFileHandlerSSL
> >     #define WSAsyncClient AsyncSSLClient
> >     #define WSAsyncServer AsyncSSLServer
> > #else
> >     #include <AsyncTCP.h>
> >     #define WSAsyncClient AsyncClient
> >     #define WSAsyncServer AsyncServer
> > #endif
> > ```
> > 
> > Finally, i replaced every "_AsyncClient_" and "_AsyncServer_" in all files inside _./src_ by "_WSAsyncClient_" and "_WSAsyncServer_", respectively.
> > 
> > For example, in the line 140 of _./src/ESPAsyncWebServer.h_ i changed:
> > 
> > ```c
> >     AsyncClient* _client;
> > ```
> > 
> > with:
> > 
> > ```c
> >     WSAsyncClient* _client;
> > ```
> > 
> > This makes the compiler to choose between AsyncClient/AsyncServer or AsyncSSLClient/AsyncSSLServer based on the ASYNC_TCP_SSL_ENABLED flag.
> > 
> > Obviously, the last step does not apply for the macro inserted in line 33 of _./src/ESPAsyncWebServer.h_.
> > 
> > **Note**: "_WSAsyncClient_" and "_WSAsyncServer_" means _Web Server AsyncClient_ and _Web Server AsyncServer_ (nothing special, just invented). But "_WSAsyncClient_" and "_WSAsyncServer_" can be replaced by any word that be different to "_AsyncClient_", "_AsyncServer_", "_AsyncSSLClient_", "_AsyncSSLServer_" and any word inside the source files.
> 
> Did you try this?

Yes i did , trying this solution gives the following error

c:/users/imtee/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\imTee\AppData\Local\Temp\arduino\sketches\A8587B274BF647287BBACD5EAD456938\sketch\sketch_mar26a.ino.cpp.o:(.literal._Z5setupv+0x48): undefined reference to AsyncWebServer::beginSecure(char const*, char const*, char const*)' c:/users/imtee/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\imTee\AppData\Local\Temp\arduino\sketches\A8587B274BF647287BBACD5EAD456938\sketch\sketch_mar26a.ino.cpp.o: in function setup()':
C:\Users\imTee\Desktop\TeeCodeSmart\sketch_mar26a/sketch_mar26a.ino:71: undefined reference to `AsyncWebServer::beginSecure(char const*, char const*, char const*)'
collect2.exe: error: ld returned 1 exit status
Multiple libraries were found for "WiFi.h"
Used: C:\Users\imTee\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\WiFi
Not used: C:\Users\imTee\Documents\Arduino\libraries\WiFi
exit status 1

Compilation error: exit status 1