me-no-dev/ESPAsyncWebServer #1184: Erreur de compilation pour la carte ESP32

pascal060, 2022-07-18:
I publish an issue here 👍 [https://forum.arduino.cc/t/why-i-have-issue-using-arduino-portable-version-with-esp32-boards-only/1013099](url)

atesin, 2022-10-03:
in english please

pascal060, 2022-10-06:
Hi there,

I am trying portable arduino ide, following this tutorial :
[https://docs.arduino.cc/software/ide-v1/tutorials/PortableIDE 1](https://docs.arduino.cc/software/ide-v1/tutorials/PortableIDE)

[image]

if I compile empty sketch for arduino AVR boards, it works, but if I do that for esp32 boards, I have this issue : c:\users\pasca\downloads\arduino-1.8.19-windows\arduino-1.8.19\portable\packages\esp32\tools\xtensa-esp32-elf-gcc\gcc8_4_0-esp-2021r2-patch3\xtensa-esp32-elf\include\c++\8.4.0\cstdlib:41:10: fatal error: bits/c++config.h: No such file or directory

I specify that I had add esp32 boards from : https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json

I tried with arduino sam board and it is ok.

The empty sketch is :

void setup() {
// put your setup code here, to run once:
}

void loop() { 
// put your main code here, to run repeatedly:
}

Thanks for help