me-no-dev/ESPAsyncWebServer #1188: Uncaught TypeError: Cannot read properties of null (reading 'getElementsByTagName')

NA7KR, 2022-07-27:
I get this error Uncaught TypeError: Cannot read properties of null (reading 'getElementsByTagName') the data been sent looks OK, 
It displays correct data on the page.
Just lots of errors in debug window.

my sent data is 
``<?xml version = '1.0'?>
<Data>
<bForward>4095</bForward>
<vForward>3.2</vForward>
<bReflected>4095</bReflected>
<vReflected>3.2</vReflected>
<Relay1>0</Relay1>
<Relay2>0</Relay2>
<Relay3>0</Relay3>
<Relay4>0</Relay4>
</Data>
``
code is ``
function response()
			{
				var message = 0;
				var barwidth;
				var currentsensor;
				var xmlResponse;
				var xmldoc;
				var dt = new Date();
				var color = "#e8e8e8";
				xmlResponse=xmlHttp.responseXML;
				document.getElementById("time").innerHTML = dt.toLocaleTimeString();
				document.getElementById("date").innerHTML = dt.toLocaleDateString();

				// *******************************************************************
				// bForward
				// *******************************************************************
				xmldoc = xmlResponse.getElementsByTagName("bForward");
				message = xmldoc[0].firstChild.nodeValue;
				if (message > 2048)
				{
					color = "#aa0000";
				}
				else 
				{
					color = "#0000aa";
				}
				document.getElementById("bfor").innerHTML=message;
				width = message / 40.95;
				document.getElementById("bfor").style.width=(width+"%");
				document.getElementById("bfor").style.backgroundColor=color;
				document.getElementById("bfor").style.borderRadius="5px";

				// *******************************************************************
				// vForward
				// *******************************************************************
				xmldoc = xmlResponse.getElementsByTagName("vForward"); //volts for A0
				message = xmldoc[0].firstChild.nodeValue;
				document.getElementById("vfor").innerHTML=message;
				document.getElementById("vfor").style.width=(barwidth+"%");
				document.getElementById("vfor").style.backgroundColor=color;

zekageri, 2022-07-27:
`xmlResponse.getElementsByTagName("bForward");` has no tags. Did you mean document?

Pablo2048, 2022-07-28:
@NA7KR  did You try to debug the xmlResponse content with `console.log(xmlResponse);` ? If the response is correct, then this has nothing to do with ESPAsyncWebServer librar and is merely the bug in Your Javascript code.