me-no-dev/ESPAsyncWebServer #1201: NSURLSession Websocket crashes ESP; Major incompatibility with Safari and Chrome (MacOS + iOS) browsers

teknoel, 2022-08-23:
I originally discovered this issue about a year ago. An ESP will instantly crash as a result of using Websockets and this library. Opening, closing and reopening a websocket will crash the ESP every time. Today, I discovered that Google Chrome on iOS is also triggering crashes. I thought it was just Apple Safari. The bug is related to NSURLSesson WebSocket protocol now favoured by almost all web browser applications. 

This bug completely kills my product development. I have been working for two full years on commercial project. Now, my work has been made useless because of this bug.  me-no-dev is now me-no-found. The owner has abandoned this library several years ago. 

I WOULD PAY SOMEONE TO FIX THIS LIBRARY.

Please read below for my original issue posted from over a year ago. The origin post was automatically closed because of lack of action. 

***

After some instinctive snooping, I have found a major incompatibility with Apple Safari on MacOS and iOS. I'm not sure what exactly is going on, but definitely put my finger on the issue being NSURLSession Websocket enabled in Safari "Experimental Feature" settings. This will crash the ESP when closing and reopening the websocket with the same client. I have confirmed that disabling Apple Safari NSURLSession Websocket stops the ESP from crashing.

Apple started enabling NSURLSession Websocket by default in iOS15 and MacOS 11+ (Big Sur or Monterey ?) Hence, the problem started showing up on Mac and iPhones etc. Every time you update Safari, iOS, or MacOS, the NSURLSession Websocket setting is enabled again.

Older system software has the experimental feature disabled by default: iOS 14.2 and Mac OS 10.14.x (Mojave)

I'm not a internet transport protocol geek, so I can't help other than tell you this general incompatibility issue. I was googling about what NSURL is doing and some developers were suggesting that: a) apple is transmitting a bad websocket data closure frame, or. b) it is a new standard for websocket protocol?¿

This bug is a real show stopper as it kills my ESP product dev.

Sincerely,

Noel Rubin http:://teknoelogy.com

teknoel, 2022-08-23:
@me-no-dev I would pay you to fix this issue...

jonathanvanpeteghem, 2022-08-24:
I was also making a product that used the async webserver. And I think many more of us. But I had to abandon it, because the lib is not stable.

@me-no-dev did a tremendous job, because I have to be honnest no clue about what does what in the lib. I think there are not many specialists like him in this case. 
I tried several libs from forks, those fix then one issue, but then I encounter other issues. 

I really hope that the people from Espressif see the need for this library and give @me-no-dev  the time to fix the issues, or to rewrite the lib where necessary. 

So @me-no-dev my comment is not meant negative, just to let you know the need from the ESP community.

zekageri, 2022-08-24:
Iam also in this group and my heart breaks when i think about this amazing masterpiece left in the mud

teknoel, 2022-08-24:
I just did a test relating to this NSURLSession WebSocket incompatibility. This test was done today on several ESP8266's, ESP-12S modules. The test only opened a websocket. There was no actual data transfers events or messages whatsoever. This library still crashes. This is not good! Can anyone tell me if the Arduino official non-synchronous library has this issue with NSURLSession Websockets used in all modern browsers? https://github.com/Links2004/arduinoWebSockets

arjenhiemstra, 2022-09-17:
https://github.com/me-no-dev/ESPAsyncWebServer/issues/1097

The linked repo on my github is more stable (although I switched to Mongoose for the websocket part myself, since then 0 issues)