me-no-dev/ESPAsyncWebServer #1202: Captive Portal Questions

tissi-2, 2022-08-25:
Dear community,

I would like to achieve the following: 

1. a user connects to an open wifi with his mobile device
2. a pop up .html file should be loaded from the ESPAsyncWebserver
3. the captive portal pops up on the display and shows the pop up .html
4. if the user accepts the terms of condition and pushes the accept button, the user should connect/stay connected to the wifi that has no internet connection
5. a welcome.html should be opened in the standard browser of the mobile device

I got 1 and 3 working with the [captive_portal.ino](https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/examples/CaptivePortal/CaptivePortal.ino) from this page but I am struggeling with the syntax to get the rest working. I am wondering if it is technically possible to achieve my goal with only 1 nodeMCU. Any help is appreciated!

zekageri, 2022-08-25:
It is possible.

barbiani, 2022-08-25:
What are the correct names of these html files to ve served?

My captive portal trial did not open the browser once the wifi was
connected.

zekageri, 2022-08-26:
> What are the correct names of these html files to ve served? My captive portal trial did not open the browser once the wifi was connected.

I don't know what do you mean.

tissi-2, 2022-08-27:
I don't think that this is possible. Just to make it clear: nodeMCU to a mobile device is the only connection. All files have to be handled by the nodeMCU. There is no internet connection available. How is it possible to accept the connection, close the captive portal popup and open up a browser in 1 step? There are a lot of discussions going on to get rid of apple's CNA, and afaik there isn't any working workaround available. Maybe you have a good solution, @zekageri ?

zekageri, 2022-08-28:
Captive portal is a browser. You can redirect the user from one URL to another while still using captive portal. 

If you are talking about the fact that you can not open an other application from client js, well thats right. 

If you want to get the network creds in captive portal and disconnect from ap and after that, connect to a wifi and after that, open a specific page, it is not possible. 

But you can redirect the user to an other URL while the ap connection and the captive is open

tissi-2, 2022-08-28:
Thanks for your answer! Yes, you are right. I was able to achieve all of that. Due to security reasons it is not possible to force a browser like safari, chrome or firefox to open when a link in a captive portal is pressed (Step 5).

zekageri, 2022-08-28:
Server side redirect? One page site?

tissi-2, 2022-08-28:
Do you mean something like this?

```
    server.on("/link_to_website", HTTP_GET, [](AsyncWebServerRequest *request){
    request->redirect("http://esp8266.com");
    });
```
With this code I stay in the captive portal.

zekageri, 2022-08-28:
But do you link it to an external website? You must link it to an URL specified on your server.

tissi-2, 2022-08-28:
No, that was just an example. I link it to /end.html. But redirecting brings me back to my index.html. For the time being I am not sure how to handle this. I think it is the handler that handles every request that is not defined with "server.on". 

`server.addHandler(new CaptiveRequestHandler()).setFilter(ON_AP_FILTER);`

```
class CaptiveRequestHandler : public AsyncWebHandler {
public:
  CaptiveRequestHandler() {}
  virtual ~CaptiveRequestHandler() {}

  bool canHandle(AsyncWebServerRequest *request){
    //request->addInterestingHeader("ANY");
    return true;
  }
  void handleRequest(AsyncWebServerRequest *request) {
    request->send(SPIFFS, "/index.html", "text/html");
  }
};
```
This is my redirect:

```
  server.on("/end", HTTP_GET, [](AsyncWebServerRequest *request){
    request->redirect("/end.html");
  });
```
I have pretty much accomplished what I was looking for. If I'll get the redirects right or will be able to close the captive portal window with a button click, I would be very happy.

zekageri, 2022-08-29:
Put your endpoints into

`CaptiveRequestHandler() {}`

zekageri, 2022-09-06:
Indeed it has to loop trought. If you got a request the server will search it in the saved endpoints one by one. It compares strings. That's why you should register the most requested routes first

CDFER, 2022-11-15:
I'm trying to solve some of these problems too, I've got a mostly working demo here: https://github.com/CDFER/Captive-Portal-ESP32

- I found a fix for windows 11 (redirect to logout.net not the ip of your webserver)
- includes the ampdu_rx_disable android workaround
- uses a public IP to fix Samsung issues
- never have the word success on your page as it can brake ios and probably macOS
- Caching header in HTML
- 404 responses to reduce useless calls and free up CPU time