me-no-dev/ESPAsyncWebServer #1205: esp32 crash when wifi lost

Vorms, 2022-09-05:
Hello,
When the disconnect event arrive, I try to close the socket:
void removeClient(){ 
  char buf[200];
if (isClientToBeRemoved){
    if (clientsMap.find(itForRemove->second.id) != clientsMap.end()){
      if (itForRemove->second.client->status() == WS_CONNECTED){
        itForRemove->second.client->close();  
      } else {
#ifdef DEBUG
        Serial.printf("Le websocket: %ld est déja déconnecté!\r\n", itForRemove->second.id);
#endif        
      }

The crash debugger:
Decoding stack results
0x4016e38f: AsyncClient::space() at F:\users\Vorms\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp line 742
0x400e6b01: AsyncClient::canSend() at F:\users\Vorms\Documents\Arduino\libraries\AsyncTCP-master\src\AsyncTCP.cpp line 1154
0x400de4f4: AsyncWebSocketClient::_queueControl(AsyncWebSocketControl*) at F:\users\Vorms\Documents\Arduino\libraries\ESPAsyncWebServer-master\src\AsyncWebSocket.cpp line 564
0x400de591: AsyncWebSocketClient::close(unsigned short, char const*) at F:\users\Vorms\Documents\Arduino\libraries\ESPAsyncWebServer-master\src\AsyncWebSocket.cpp line 590
0x400d49b9: removeClient() at F:\users\Vorms\Documents\Arduino\Ne202_auto_connect_new/Ne202_auto_connect_new.ino line 739
0x400d8174: loop() at F:\users\Vorms\Documents\Arduino\Ne202_auto_connect_new/Ne202_auto_connect_new.ino line 598
0x400e9961: loopTask(void*) at F:\users\Vorms\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.6\cores\esp32\main.cpp line 23
0x4008ec4e: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

Please keep me in touch !!!

Best regards
Thierry Vorms

zekageri, 2022-09-06:
If the socket disconnected you can not close it because it was destroyed earlier. That is the crash

Vorms, 2022-09-06:
Hoh thanks a lot for your response!
Your library is very good, but, may be you can publish that in case of
disconnect th ws is destroy! For me it was very difficult to find !

Best regards
Thierry Vorms