me-no-dev/ESPAsyncWebServer #1210: SPIFFSEditor not displaying file contents on ESP32 with LittleFS

steff393, 2022-09-17:
Hi, 
thanks for this great library. I used it a lot on ESP8266. But now with ESP32 and LittleFS I found that the SPIFFSEditor doesn't display the file contents (only the file names on the left frame).
I got the error message:
[E][vfs_api.cpp:29] open(): myFile.txt does not start with /

The following change (i.e. adding the "/" in front of the filename) fixes this behavior:
https://github.com/me-no-dev/ESPAsyncWebServer/blob/f71e3d427b5be9791a8a2c93cf8079792c3a9a26/src/SPIFFSEditor.cpp#L397
`request->_tempFile = _fs.open(String("/") + request->arg("edit"), "r");`

But I'm not sure if this is really the solution or only a quick workaround. What is your opinion?

blazoncek, 2022-09-19:
I have noticed the same behaviour, though the fork we use (from @Aircoookie) works correctly on ESP32 but fails on ESP32-S2 and ESP32-S3.

slimline33, 2022-09-27:
The same problem here on my esp32 wroom.

[image]

the workaround does not help :(

steff393, 2022-10-03:
After I had to introduce that "/" into multiple different places, I searched for a more generic approach: 
It now works also, when I introduce a
```
#ifdef ESP32
        output += "/";
#endif
```

here in between line 465 and 466:
https://github.com/me-no-dev/ESPAsyncWebServer/blob/f71e3d427b5be9791a8a2c93cf8079792c3a9a26/src/SPIFFSEditor.cpp#L464-L468

This makes the filenames in the left frame of the SPIFFSEditor being displayed as "/index.html" instead of "index.html" and also the subsequent actions like Save, Delete, etc. use then the correct filename including "/".

So from my point of view this could be a real solution.

However I don't yet understand, why it's working in Aircookies fork. So there might be also another solution.

blazoncek, 2022-10-03:
@steff393 You may want to use '/' instead of "/".

Aircoookie, 2022-10-04:
Thank you for finding what seems like a good solution!
Note that at least for my fork/environment, filenames correctly include `/` as is, so your proposed change makes it `//`, which appears to work fine, but feels hacky.

My hunch is that the LittleFS library by lorol returns filenames with leading `/` while the version built into the (recent) ESP32 arduino core does not.
Instead of a define, I just added a check for leading slash to my fork (v2.0.7), seems to work well.

silk-indus, 2022-10-09:
Hi to all.
This has still one problem. The files are correctly edited and viewed in sub-directories. Unfortunately I was unable to view/edit file in subdir. Anybody has the same problem? 

Moreover, the Platformio littlefs uploader composes bad bin file (in core version 5.2). I am solving this by composing littlefs bin separately using of mklittlefs.exe and then upload manually.

steff393, 2022-10-10:
I digged a little deeper, to find out, why `entry.name()` is returning the filename without '/', while on the other hand `_fs.open()` insists on the '/'. Because in the end, both functions come from the same source module, so I'd expect they should behave consistent: 
[VFSFileImpl::name()](https://github.com/espressif/arduino-esp32/blob/a5f03a86512b270be1080d7e7cb82e4fd71c9a1a/libraries/FS/src/vfs_api.cpp#L446)
[VFSFileImpl::open()](https://github.com/espressif/arduino-esp32/blob/a5f03a86512b270be1080d7e7cb82e4fd71c9a1a/libraries/FS/src/vfs_api.cpp#L21)

`name()` calls then the function [pathToFileName(path())](https://github.com/espressif/arduino-esp32/blob/1a7962ece8a4c6ffa1d64c5a86ed8ee58dde10ba/cores/esp32/esp32-hal-misc.c#L271), which is returning only characters after the last '/' in the file path (i.e. it removes the slash).

But there is also another function `path()` in vfs_api.cpp, which returns the full path and filename including the slash(es).

Therefore I think the best solution would be:
```
#ifdef ESP32
        output += String(entry.path());
#else
        output += String(entry.name());
#endif
```
On my system it works well and I'm going to raise a PR accordingly.
Unfortunately `path()` is not existing in ESP8266, so the `#ifdef` seems necessary.

closed 2022-10-30 by steff393