me-no-dev/ESPAsyncWebServer #1223: AsyncWebServer download SPIFFS file truncated

mugginsjm, 2022-10-19:
Hi can someone please help.
I have been using SPIFFS to log activity on my app and have a file of approx 200 lines. On my web page when I call:
 request->send(SPIFFS, "/fridge.txt", "text/plain");
I get a truncated file (approx 50 lines) and then this at bottom.

HTTP/1.1 302 Found
Content-Length: 0
Connection: close
Location: /
Accept-Ranges: none

Similarly if I attempt a download using 
request->send(SPIFFS, "/fridge.txt", String(), true);
the process freezes.
I'm sure this is something obvious but with numerous hours of searching I cannot resolve.
thanks
John

mugginsjm, 2022-10-19:
Looks like I need to use "chunking" but the examples I have found are very complex (to me) and don't seem to lend themselves to simple Spiffs text files. Is there a simpler method?

Pablo2048, 2022-10-20:
You didn't give any MCVE to reproduce so it's hard to guess (and waste of time). Basically said serving of SPIFFS (is obsolete, use LittleFS instead) file is working. I'm transferring >100kB this way. File response already use chunked encoding internally. You can also try to let the server serve file internally using `serveStatic`.

mugginsjm, 2022-10-20:
Hi
Thanks for response. I had originally been using LITTLEFS for ESP32 and just tried SPIFFS as a last resort. I will put together an abridged version to demonstrate the problem.
thanks

mugginsjm, 2022-11-04:
Hi back again.
I thought I had a reliable download from LittleFS but now find with larger log files of 600 lines plus, approx 120k, that download stalls and I get "download-network issue" Any advice anyone... thanks
John