me-no-dev/ESPAsyncWebServer #1225: redirect() with ESP.restart() or ESP.reset() does not function

artofit, 2022-10-27:
v1.9.1 [ESPAsyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer)
nodeMCUv2

Hello,

if redirect() methods works, I see it fails to work with ESP.restart() or ESP.reset()

I want to reboot if requested in the url to reboot,
the html page has an auto refresh
as the redirect (fails to be excuted?) with the restart instruction, the html refresh makes the esp endlessly reboot, till the moment one changes the url.

AsyncWebServer  gasyncSrv(80);

```
gasyncSrv.on("/modeReboot", HTTP_GET, [](AsyncWebServerRequest *request){
        // you must leave this url otherwise infinite reboot
        // UI loops on it
        Serial.println("reboot change url");
        request->redirect("/");
        ESP.restart();  // make ESP8266 crash,
      // reboots but as url is not redirected to root or any other(stays on /modeReboot), the html refresh triggers restart
     // commenting ESP.restart(); then the redirect() method works splendid
    });
```

Thanks

stefann63, 2022-12-02:
Handling async events must be kept short and without system calls.
I had same issue and got around by creating a global variable...
uint16_t BackgroungCommand =0;
Set this variable in your async response. 
Check it in loop() and execute your desired command and clear it.

jasonk, 2022-12-10:
Things like this are the reason that in general you should not use a GET request to perform actions, especially actions that would be problematic if the happened by accident. This is what POST is for, and using POST would avoid this problem because when it reconnects the browser would ask if you wanted to resubmit or not.