me-no-dev/ESPAsyncWebServer #1228: garbage included at compilation of const char index_html[] PROGMEM = R"rawliteral

kalex3000, 2022-11-01:
hi, I use 
const char index_html[] PROGMEM = R"rawliteral(
<body>
<script>
// large HTTP and Javascript functions
</script>
</body>
</html>
)rawliteral";
the compiler introduce a lot of garbage in it, anyone that can help me to fix this issue, I'm using Arduino IDE??

dedobbin, 2022-11-27:
Ran into the same issue. Seems to have to do with the way lambdas callbacks (for server.on) capture the PROGMEM variables

kalex3000, 2022-11-28:
When 
"const char index_html[] PROGMEM = R"rawliteral( " 
is more than 10Kbyte has that issue. I fixed splitting the main index_html to keep small amount of memory. and also writing some part of the index_html in the String processor(const String& var){
thanks. that issue was even if the Var was global!!.