me-no-dev/ESPAsyncWebServer #1235: Response doesnt send anything from SPIFF

abratchik, 2022-11-19:
Hello,

I'm trying to serve a file from SPIFFS and it doesn't work, please kindly help. I'm using the ESP32 CAM board, with SPIFFS. The library I'm using is downloaded and installed from this repo, so it should be the latest. Here is my code, called from the setup():

```
    server.on("/portal", HTTP_GET, [](AsyncWebServerRequest *request){
        Serial.println("GET portal received");
        if(SPIFFS.exists("/portal.html"))
            Serial.println("portal.html found");
        else
            Serial.println("portal.html NOT found");

        request->send(SPIFFS, "/portal.html", "text/html");
    });
```

I'm getting the serial printout as expected:
```
GET portal received
portal.html found
```

But the browser dowsnt show the content, only an endless spinner or a blank page. I tried to trace the send call, and added some debug in the AsyncFileResponse to understand what is happening:

```
      ...
  _content = fs.open(_path, "r");
  _contentLength = _content.size();

  Serial.print("File size:");
  Serial.println(_contentLength);
      ...
```

The Serial.printout shows that the file is non-zero lenght:

```
File size:1278
```

But still the content of the file is not returned to the browser.

However, if I form the response from the PROGMEM or a memory buffer, everything is working just fine. What am I missing? Looks like returning content from SPIFFS is the issue but I cannot understand why? I can open files from there and read the content without any problem.

Belleson, 2022-11-19:
server.on has a watchdog timer and is sensitive to what events happen in the response. Try it without the Serial.println statements.

abratchik, 2022-11-19:
Nope, this doesn't help. I just removed all Serial.println and tried again - doesn't work.  But request.sent_P from PROGMEM array works just fine.

abratchik, 2022-11-20:
Interesting that switching from SPIFFS to SD solved the problem. Probably the issue is in some special way how the SPIFFS should be mounted. It works in other code though. For example, I can read and write files to SPIFFS but it refuses to work with the AsyncWebServer

abratchik, 2023-01-07:
Closing this issue as SPIFF is obsolete anyways. Switching to SD/LittleFS solved the problem

closed 2023-01-07 by abratchik