me-no-dev/ESPAsyncWebServer #1243: Still begin maintained...?

Zhu-jiatong, 2022-11-30:
Is this repo still actively maintained @me-no-dev? Or are there any maintained fork(s) out there.
Will this become part of the [arduino-esp32](https://github.com/espressif/arduino-esp32)? 
Thank you.

HamzaHajeir, 2022-12-15:
Intereseted in having an answer, it seems that he's buzy creating other bugs.

HamzaHajeir, 2022-12-15:
One can checkout this [library](https://github.com/philbowles/H4AsyncWebServer), but I'm sorry to say its author has passed away 6 months ago. 

We're trying to get his paperworks and publish the unpublished updates, then perhaps maintain the project.

MacDada, 2023-03-13:
https://github.com/me-no-dev/ESPAsyncWebServer/issues/1282#issuecomment-1465847065

DRSDavidSoft, 2024-09-06:
Continued in #1282