me-no-dev/ESPAsyncWebServer #1246: AsyncTCP crashes after a while

stefann63, 2022-12-03:
PLATFORM: Espressif 32 (5.2.0) > NodeMCU-32S
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20005.220925 (2.0.5) 
 - tool-esptoolpy @ 1.40201.0 (4.2.1) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 39 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ArduinoJson @ 6.19.4
|-- AsyncTCP @ 1.1.1
|-- ESP Async WebServer @ 1.2.3
|   |-- AsyncTCP @ 1.1.1
|   |-- FS @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- ESP32 Async UDP @ 2.0.0
|-- WiFi @ 2.0.0
|-- Ticker @ 2.0.0
|-- Wire @ 2.0.0
|-- EEPROM @ 2.0.0
|-- Update @ 2.0.0
|-- ParserCMD
|-- FS @ 2.0.0
|-- SPIFFS @ 2.0.0
|   |-- FS @ 2.0.0
|-- DNSServer @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- SPI @ 2.0.0
....
RAM:   [==        ]  18.9% (used 62088 bytes from 327680 bytes)
Flash: [=======   ]  68.6% (used 899469 bytes from 1310720 bytes)

Some crash reports collected...

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x400e28b7  PS      : 0x00060e30  A0      : 0x800e293f  A1      : 0x3ffb4740  
A2      : 0x00000000  A3      : 0x3ffdd89c  A4      : 0x00000000  A5      : 0x3ffcae5c  
A6      : 0x3ffcae74  A7      : 0x00060023  A8      : 0x801174d4  A9      : 0x3ffb4730  
A10     : 0x00000000  A11     : 0x00000006  A12     : 0x00000014  A13     : 0x0000ffff  
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x00000019  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000001c  LBEG    : 0x400891e0  LEND    : 0x400891f6  LCOUNT  : 0xffffffff  

Backtrace:0x400e28b4:0x3ffb47400x400e293c:0x3ffb4770 0x4010a5f9:0x3ffb4790 0x4010f5e2:0x3ffb4800 0x401145aa:0x3ffb4830 0x401047dd:0x3ffb4850 

0x400e28b7: AsyncServer::_accept(tcp_pcb*, signed char) at C:\Users\stefan\Dropbox\eclipse-ws\SAmaster/.pio/libdeps/nodemcu-32s/AsyncTCP/src/AsyncTCP.cpp:1296
0x400891e0: memcpy at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:163
0x400891f6: memcpy at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memcpy.S:197
0x400e28b4: AsyncServer::_accept(tcp_pcb*, signed char) at C:\Users\stefan\Dropbox\eclipse-ws\SAmaster/.pio/libdeps/nodemcu-32s/AsyncTCP/src/AsyncTCP.cpp:1294
0x400e293c: AsyncServer::_s_accept(void*, tcp_pcb*, signed char) at C:\Users\stefan\Dropbox\eclipse-ws\SAmaster/.pio/libdeps/nodemcu-32s/AsyncTCP/src/AsyncTCP.cpp:1333
0x4010a5f9: tcp_process at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/tcp_in.c:945
 (inlined by) tcp_input at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/tcp_in.c:438
0x4010f5e2: ip4_input at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/ipv4/ip4.c:800
0x401145aa: ethernet_input at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/netif/ethernet.c:186
0x401047dd: tcpip_thread_handle_msg at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/tcpip.c:180
 (inlined by) tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/tcpip.c:154
********************************

CORRUPT HEAP: Bad tail at 0x3ffdfb6c. Expected 0xbaad5678 got 0xbaad5677
assert failed: multi_heap_free multi_heap_poisoning.c:253 (head != NULL)
Backtrace:0x40083651:0x3ffd9ef00x4008b745:0x3ffd9f10 0x40090b4d:0x3ffd9f30 0x40090793:0x3ffda060 0x40083ab5:0x3ffda080 0x40090b7d:0x3ffda0a0 0x400f8823:0x3ffda0c0 0x400e2c99:0x3ffda0f0 
ELF file SHA256: 0000000000000000

0x40083651: panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:402
0x4008b745: esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:128
0x40090b4d: __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
0x40090793: multi_heap_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/multi_heap_poisoning.c:253
 (inlined by) multi_heap_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/multi_heap_poisoning.c:245
0x40083ab5: heap_caps_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/heap_caps.c:340
0x40090b7d: free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/heap.c:39
0x400f8823: esp_task_wdt_delete at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/task_wdt.c:351
0x400e2c99: _async_service_task(void*) at C:\Users\stefan\Dropbox\eclipse-ws\SAmaster/.pio/libdeps/nodemcu-32s/AsyncTCP/src/AsyncTCP.cpp:196
*******************************

0x40083651: panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:402
0x4008b745: esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:128
0x40090b4d: __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
0x40090793: multi_heap_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/multi_heap_poisoning.c:253
 (inlined by) multi_heap_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/multi_heap_poisoning.c:245
0x40083ab5: heap_caps_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/heap_caps.c:340
0x40090b7d: free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/heap.c:39
0x40155f99: operator delete(void*) at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/del_op.cc:49
0x400e61ed: LinkedList<AsyncWebSocketMessage*, LinkedListNode>::remove(AsyncWebSocketMessage* const&) at C:\Users\stefan\Dropbox\eclipse-ws\SAmaster/.pio/libdeps/nodemcu-32s/ESP Async WebServer/src/StringArray.h:131
 (inlined by) AsyncWebSocketClient::_runQueue() at C:\Users\stefan\Dropbox\eclipse-ws\SAmaster/.pio/libdeps/nodemcu-32s/ESP Async WebServer/src/AsyncWebSocket.cpp:528
0x400e68ea: AsyncWebSocketClient::_onAck(unsigned int, unsigned int) at C:\Users\stefan\Dropbox\eclipse-ws\SAmaster/.pio/libdeps/nodemcu-32s/ESP Async WebServer/src/AsyncWebSocket.cpp:515
0x400e68f9: std::_Function_handler<void (void*, AsyncClient*, unsigned int, unsigned int), AsyncWebSocketClient::AsyncWebSocketClient(AsyncWebServerRequest*, AsyncWebSocket*)::{lambda(void*, AsyncClient*, unsigned int, unsigned int)#4}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, unsigned int&&, AsyncClient*&&) at C:\Users\stefan\Dropbox\eclipse-ws\SAmaster/.pio/libdeps/nodemcu-32s/ESP Async WebServer/src/AsyncWebSocket.cpp:480
 (inlined by) _M_invoke at c:\users\stefan\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:297
0x400e261b: std::function<void (void*, AsyncClient*, unsigned int, unsigned int)>::operator()(void*, AsyncClient*, unsigned int, unsigned int) const at c:\users\stefan\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:687
 (inlined by) AsyncClient::_sent(tcp_pcb*, unsigned short) at C:\Users\stefan\Dropbox\eclipse-ws\SAmaster/.pio/libdeps/nodemcu-32s/AsyncTCP/src/AsyncTCP.cpp:898
 (inlined by) AsyncClient::_sent(tcp_pcb*, unsigned short) at C:\Users\stefan\Dropbox\eclipse-ws\SAmaster/.pio/libdeps/nodemcu-32s/AsyncTCP/src/AsyncTCP.cpp:893
0x400e27a8: AsyncClient::_s_sent(void*, tcp_pcb*, unsigned short) at C:\Users\stefan\Dropbox\eclipse-ws\SAmaster/.pio/libdeps/nodemcu-32s/AsyncTCP/src/AsyncTCP.cpp:1203
0x400e2d55: _async_service_task(void*) at C:\Users\stefan\Dropbox\eclipse-ws\SAmaster/.pio/libdeps/nodemcu-32s/AsyncTCP/src/AsyncTCP.cpp:165
 (inlined by) _async_service_task at C:\Users\stefan\Dropbox\eclipse-ws\SAmaster/.pio/libdeps/nodemcu-32s/AsyncTCP/src/AsyncTCP.cpp:194

Application is using a one TCP connection, one UDP port and a WEB server with one WEB socket.
WEB socket is using a JSON a global DynamicJsonDocument jBuf(2048); that is actually 200-500 bytes long.
The .onEvent(...) doesn't have any String, or other memory allocations, only fills variables from the received JSON.
Sometime it will udp.writeTo(...) 2-5 packets and/or send a packet over TCP connection.

Crashes are rare, sometime it will work for days.

stefann63, 2022-12-03:
Probably related...
Sometimes i will get this message printed over and over the serial port.
ERROR: Too many messages queued
It's printed from AsyncEventSource.cpp or AsyncWebSocket.cpp
Message disappears after closing the WEB browser.

stefann63, 2022-12-03:
Added ws_station.cleanupClients(); to be called from loop(); every second i got new type of crashes after about 30 minutes.  
Running 6 WEB clients from 2 PC's and sending about 300bytes every 250mS
	  serializeJson(jBuf,js);
	  ws_station.textAll(js);

**********
0x40083651: panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:402
0x4008b745: esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:128
0x40090b4d: __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
0x40090793: multi_heap_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/multi_heap_poisoning.c:253
 (inlined by) multi_heap_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/multi_heap_poisoning.c:245
0x40083ab5: heap_caps_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/heap_caps.c:340
0x40090b7d: free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/heap.c:39
0x40155d05: operator delete(void*) at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/gcc/libstdc++-v3/libsupc++/del_op.cc:49
0x400e6185: LinkedList<AsyncWebSocketMessage*, LinkedListNode>::remove(AsyncWebSocketMessage* const&) at C:\Users\stefan\Dropbox\eclipse-ws\SAmaster/.pio/libdeps/nodemcu-32s/ESP Async WebServer/src/StringArray.h:131
 (inlined by) AsyncWebSocketClient::_runQueue() at C:\Users\stefan\Dropbox\eclipse-ws\SAmaster/.pio/libdeps/nodemcu-32s/ESP Async WebServer/src/AsyncWebSocket.cpp:528
0x400e6882: AsyncWebSocketClient::_onAck(unsigned int, unsigned int) at C:\Users\stefan\Dropbox\eclipse-ws\SAmaster/.pio/libdeps/nodemcu-32s/ESP Async WebServer/src/AsyncWebSocket.cpp:515
0x400e6891: std::_Function_handler<void (void*, AsyncClient*, unsigned int, unsigned int), AsyncWebSocketClient::AsyncWebSocketClient(AsyncWebServerRequest*, AsyncWebSocket*)::{lambda(void*, AsyncClient*, unsigned int, unsigned int)#4}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, unsigned int&&, AsyncClient*&&) at C:\Users\stefan\Dropbox\eclipse-ws\SAmaster/.pio/libdeps/nodemcu-32s/ESP Async WebServer/src/AsyncWebSocket.cpp:480
 (inlined by) _M_invoke at c:\users\stefan\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:297
0x400e25b3: std::function<void (void*, AsyncClient*, unsigned int, unsigned int)>::operator()(void*, AsyncClient*, unsigned int, unsigned int) const at c:\users\stefan\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:687
 (inlined by) AsyncClient::_sent(tcp_pcb*, unsigned short) at C:\Users\stefan\Dropbox\eclipse-ws\SAmaster/.pio/libdeps/nodemcu-32s/AsyncTCP/src/AsyncTCP.cpp:898
 (inlined by) AsyncClient::_sent(tcp_pcb*, unsigned short) at C:\Users\stefan\Dropbox\eclipse-ws\SAmaster/.pio/libdeps/nodemcu-32s/AsyncTCP/src/AsyncTCP.cpp:893
0x400e2740: AsyncClient::_s_sent(void*, tcp_pcb*, unsigned short) at C:\Users\stefan\Dropbox\eclipse-ws\SAmaster/.pio/libdeps/nodemcu-32s/AsyncTCP/src/AsyncTCP.cpp:1203
0x400e2ced: _async_service_task(void*) at C:\Users\stefan\Dropbox\eclipse-ws\SAmaster/.pio/libdeps/nodemcu-32s/AsyncTCP/src/AsyncTCP.cpp:165
 (inlined by) _async_service_task at C:\Users\stefan\Dropbox\eclipse-ws\SAmaster/.pio/libdeps/nodemcu-32s/AsyncTCP/src/AsyncTCP.cpp:194

*************
0x40083651: panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:402
0x4008b745: esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:128
0x40090b4d: __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
0x40090793: multi_heap_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/multi_heap_poisoning.c:253
 (inlined by) multi_heap_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/multi_heap_poisoning.c:245
0x40083ab5: heap_caps_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/heap_caps.c:340
0x40090b7d: free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/heap.c:39
0x40155bb5: ram_phy_get_vdd33 at /home/cff/gittree/chip7.1_phy/chip_7.1/board_code/app_test/pp/phy/phy_chip_v7_cal.c:2346
0x400e6059: AsyncWebSocket::AsyncWebSocket(String const&) at c:\users\stefan\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:676 (discriminator 2)
 (inlined by) AsyncWebSocket::AsyncWebSocket(String const&) at C:\Users\stefan\Dropbox\eclipse-ws\SAmaster/.pio/libdeps/nodemcu-32s/ESP Async WebServer/src/AsyncWebSocket.cpp:855 (discriminator 2)
0x400e6756: LinkedList<AsyncWebSocketMessage*, LinkedListNode>::free() at C:\Users\stefan\Dropbox\eclipse-ws\SAmaster/.pio/libdeps/nodemcu-32s/ESP Async WebServer/src/StringArray.h:162
 (inlined by) AsyncWebSocketClient::~AsyncWebSocketClient() at C:\Users\stefan\Dropbox\eclipse-ws\SAmaster/.pio/libdeps/nodemcu-32s/ESP Async WebServer/src/AsyncWebSocket.cpp:491
0x400e6765: LinkedList<AsyncWebSocketControl*, LinkedListNode>::free() at C:\Users\stefan\Dropbox\eclipse-ws\SAmaster/.pio/libdeps/nodemcu-32s/ESP Async WebServer/src/StringArray.h:164
 (inlined by) AsyncWebSocketClient::~AsyncWebSocketClient() at C:\Users\stefan\Dropbox\eclipse-ws\SAmaster/.pio/libdeps/nodemcu-32s/ESP Async WebServer/src/AsyncWebSocket.cpp:492
[... cut ...]
0x400e2614: std::function<void (void*, AsyncClient*, void*, unsigned int)>::operator()(void*, AsyncClient*, void*, unsigned int) const at c:\users\stefan\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:687
 (inlined by) AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) at C:\Users\stefan\Dropbox\eclipse-ws\SAmaster/.pio/libdeps/nodemcu-32s/AsyncTCP/src/AsyncTCP.cpp:915
0x400e2bc1: AsyncClient::_fin(tcp_pcb*, signed char) at C:\Users\stefan\Dropbox\eclipse-ws\SAmaster/.pio/libdeps/nodemcu-32s/AsyncTCP/src/AsyncTCP.cpp:887

beckmx, 2022-12-04:
you need to share some code, the logs doesnt say much without the code itself

stefann63, 2022-12-05:
@beckmx i don't see in the trace info where in my code the problem originated. 
There are references to AsyncWebSocket.cpp and following is the code that sends data to the WEB client.
In current configuration data send to each client is ~300bytes every 250ms. Testing with 2-4 clients on two host's and it restarts in 1-3h.
If there is no WEB client, code works for days without restart.
There is also UDP traffic on one port, 
Very similar code is working without problem on ESP8266.

This code is called from loop(); as background service
----
char js[2048];
void webSocketService(){
static unsigned long cleanup_timer = millis() + 1000;
  if ( millis() > cleanup_timer) {
	  ws_station.cleanupClients();
	  cleanup_timer = millis() + 1000;
  }

  if ( millis() > sanet_timer){						// SANet slot
	  sanet_timer = millis() + SANet.utime_web;
	  if (ws_station.count() == 0) return;

	  DynamicJsonDocument jBufS(2048);

	  jBufS["state"] = SANet.WEBstate;						// 		F("update");
	  jBufS["khz"] = Frequency;
	  if(SANet.meter_n > 0){
		  jBufS["meter"] = NVMEM.Node[SANet.meter_n-1].Name;
		  jBufS["pwr"] = SANet.pwr;				// FWDPower;
		  sprintf_P(js, PSTR("%5.2f"),SANet.swr);
		  jBufS["swr"] = js;		//String (SANet.swr);				// SWR;
	  }
	  if (SANet.pa_node[0] != 0) {
		  int n = SANet.pa_node[0] - 1;
		  if (node[n].pamp != nullptr){
			  JsonObject pa1 = jBufS.createNestedObject("pa1");	//L=22+36+15+16+32 = 121
			  pa1["name"] = NVMEM.Node[n].Name;
			  pa1["pwr"] = node[n].pamp->pwr;
			  sprintf_P(js, PSTR("%5.2f"),node[n].pamp->swr);
			  pa1["swr"] = js; 		//String (node[n].pamp->swr);
			  pa1["degc"] = node[n].pamp->degc;
			  pa1["state"] = node[n].pamp->state;
			  if (node[n].pamp->fault[0] != 0) pa1["fault"] = node[n].pamp->fault;
		  }
	  }
	  if (SANet.rotator_n > 0){
		  JsonObject rot = jBufS.createNestedObject("rotator");
		  rot["name"] = NVMEM.Node[SANet.rotator_n-1].Name;	// 50
		  rot["state"] = SANet.rotator.state;
		  rot["az"] = SANet.rotator.az;
		  if (SANet.rotator.alim1 >=0) rot["alim1"] = SANet.rotator.alim1;
		  if (SANet.rotator.alim2 >=0) rot["alim2"] = SANet.rotator.alim2;
		  if (SANet.rotator.atdeg >=0) rot["atdeg"] = SANet.rotator.atdeg;
		  if (SANet.rotator.dxdeg >=0) rot["dxdeg"] = SANet.rotator.dxdeg;
		  if (SANet.rotator.dxqrb >=0) rot["dxqrb"] = SANet.rotator.dxqrb;
		  if (strlen(SANet.rotator.fault)) rot["fault"] = SANet.rotator.fault;
	  }
	  if (updateSAWeb_f) {									// update/init all WEB
		  updateSAWeb_f = false;
		  JsonArray ant = jBufS.createNestedArray("ant");
		  for (int i =0;  i < AntList.tx_found; i++){						// 6+ 32*32 + 16 ~ 1056 max
			  ant.add( NVMEM.Ant[NVMEM.Mem[AntList.tx_idx[i]].Ant].Name);
			  if (AntTX == NVMEM.Mem[AntList.tx_idx[i]].Ant + 1){
				  jBufS["antidx"] = i;
			  }
		  }

		  JsonArray ant3 = jBufS.createNestedArray("ant3");	// RX only
		  for (int i =0;  i < AntList.rx_found; i++){
			  ant3.add( NVMEM.Ant[NVMEM.Mem[AntList.rx_idx[i]].Ant].Name);
			  if (AntRX == NVMEM.Mem[AntList.rx_idx[i]].Ant + 1){
				  jBufS["antidx3"] = i;
			  }
		  }
		  jBufS["msg"] = SANet.WEBmsg;
		  jBufS["band"] = HAM_Band;
		  jBufS["shackpwr"] = (int)ShackPower;
	  }
	  serializeJson(jBufS,js);
	  ws_station.textAll(js);
  }
}

gregorykolasis, 2022-12-09:
I have the same behavior like yours 
ESP-Core-Version:2.0.4 , also Tested with 2.0.0

Latest Versions for ESPAsyncWebServer/AsyncTCP

gregorykolasis, 2022-12-14:
Any update on this problem ?

BlueAndi, 2023-01-02:
Can you try with: AsyncTcpSock instead of AsyncTcp?
https://github.com/yubox-node-org/AsyncTCPSock

stefann63, 2023-01-12:
Lowering the traffic about two times make it way more stable. Now it works for days without restart.

gregorykolasis, 2023-01-13:
> Can you try with: AsyncTcpSock instead of AsyncTcp? https://github.com/yubox-node-org/AsyncTCPSock

This seemed to drop me some errors from time time 
[776424][E][AsyncTCP.cpp:797] _flushWriteQueue(): socket 52 lwip_write() failed errno=104

But it won't crash!
So it somehow improved the system!

BlueAndi, 2023-01-14:
104 means connection reset from peer (remote device), which is handled by the AsyncTcpSock. No worry about.

mathieucarbou, 2024-01-07:
> Can you try with: AsyncTcpSock instead of AsyncTcp? https://github.com/yubox-node-org/AsyncTCPSock

Tried swapping AsyncTCP with https://github.com/yubox-node-org/AsyncTCPSock but it did not work:

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400ff868  PS      : 0x00060130  A0      : 0x80109e98  A1      : 0x3ffd4cf0  
A2      : 0x00000058  A3      : 0x3ffd4d74  A4      : 0x00000010  A5      : 0x3ffd4cdc  
A6      : 0x3ffcb790  A7      : 0x3ffd4cdc  A8      : 0x800e0e26  A9      : 0x00000008  
A10     : 0x3ffd4d0c  A11     : 0x401d8844  A12     : 0x3ffd4d0c  A13     : 0x0000000c  
A14     : 0x3ffd4ca0  A15     : 0x00000008  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000058  LBEG    : 0x4008a450  LEND    : 0x4008a45a  LCOUNT  : 0x00000000  

Backtrace: 0x400ff865:0x3ffd4cf0 0x40109e95:0x3ffd4d40 0x400e0e23:0x3ffd4db0 0x400f4e74:0x3ffd4e50 0x4012895d:0x3ffd4e70

  #0  0x400ff865:0x3ffd4cf0 in std::enable_if<std::__and_<std::__not_<std::__is_tuple_like<std::_Any_data> >, std::is_move_constructible<std::_Any_data>, std::is_move_assignable<std::_Any_data> >::value, void>::type std::swap<std::_Any_data>(std::_Any_data&, std::_Any_data&) at /Users/mat/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/bits/move.h:193
      (inlined by) std::function<void (AsyncWebServerRequest*)>::swap(std::function<void (AsyncWebServerRequest*)>&) at /Users/mat/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:548
      (inlined by) std::function<void (AsyncWebServerRequest*)>::operator=(std::function<void (AsyncWebServerRequest*)> const&) at /Users/mat/.platformio/packages/toolchain-xtensa-esp32@8.4.0+2021r2-patch5/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:463
      (inlined by) AsyncCallbackWebHandler::onRequest(std::function<void (AsyncWebServerRequest*)>) at .pio/libdeps/pro-esp32-debug/ESPAsyncWebServer-esphome/src/WebHandlerImpl.h:82
      (inlined by) AsyncWebServer::onNotFound(std::function<void (AsyncWebServerRequest*)>) at .pio/libdeps/pro-esp32-debug/ESPAsyncWebServer-esphome/src/WebServer.cpp:172
  #1  0x40109e95:0x3ffd4d40 in Mycila::HTTPdClass::begin() at lib/MycilaHTTPd/MycilaHTTPd.cpp:25
  #2  0x400e0e23:0x3ffd4db0 in YaSolR::Controller::loop() at src/YaSolRController.cpp:409
  #3  0x400f4e74:0x3ffd4e50 in loop() at src/main.cpp:144
  #4  0x4012895d:0x3ffd4e70 in loopTask(void*) at /Users/mat/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50

ELF file SHA256: a8f210ee0c4ff47a

E (3827) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
```