me-no-dev/ESPAsyncWebServer #1247: WebSockets doesn't work in esp-idf but works in the arduino-ide.

TopGgg, 2022-12-03:
Hi!
I recently moved to esp-idf from the arduino-ide and added your library as a component, and the web server is working great!
but when I try to use web sockets it crashes with the following error:
```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400deed4  PS      : 0x00060430  A0      : 0x800e09a1  A1      : 0x3ffd07e0  
0x400deed4: String::buffer() const at /media/adarw/Data/Documents/Programing/cpp/robot/esp/espCodeIdf/espcode/espCode/espCode/components/arduino/cores/esp32/WString.h:342
 (inlined by) String::c_str() const at /media/adarw/Data/Documents/Programing/cpp/robot/esp/espCodeIdf/espcode/espCode/espCode/components/arduino/cores/esp32/WString.h:250
 (inlined by) AsyncWebServerRequest::_removeNotInterestingHeaders() at /media/adarw/Data/Documents/Programing/cpp/robot/esp/espCodeIdf/espcode/espCode/espCode/components/ESPAsyncWebServer/src/WebRequest.cpp:184
 (inlined by) AsyncWebServerRequest::_removeNotInterestingHeaders() at /media/adarw/Data/Documents/Programing/cpp/robot/esp/espCodeIdf/espcode/espCode/espCode/components/ESPAsyncWebServer/src/WebRequest.cpp:181

A2      : 0x3ffd0cfc  A3      : 0x3ffd1778  A4      : 0x3ffb793c  A5      : 0x3ffd0da8  
A6      : 0x400e2710  A7      : 0xff000000  A8      : 0x800def39  A9      : 0x3ffd07c0  
0x400e2710: AsyncWebRewrite::match(AsyncWebServerRequest*) at /media/adarw/Data/Documents/Programing/cpp/robot/esp/espCodeIdf/espcode/espCode/espCode/components/ESPAsyncWebServer/src/ESPAsyncWebServer.h:323

A10     : 0x3ffd1778  A11     : 0xb80ae660  A12     : 0x00000001  A13     : 0x3ffcc874  
A14     : 0x0000000d  A15     : 0x00000000  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xb80ae66f  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe  

Backtrace: 0x400deed1:0x3ffd07e0 0x400e099e:0x3ffd0810 0x400e0c01:0x3ffd0850 0x400e0e11:0x3ffd08a0 0x400e3b2e:0x3ffd08c0 0x400e3dde:0x3ffd0910
0x400deed1: StringArray::containsIgnoreCase(String const&) at /media/adarw/Data/Documents/Programing/cpp/robot/esp/espCodeIdf/espcode/espCode/espCode/components/ESPAsyncWebServer/src/StringArray.h:181
 (inlined by) AsyncWebServerRequest::_removeNotInterestingHeaders() at /media/adarw/Data/Documents/Programing/cpp/robot/esp/espCodeIdf/espcode/espCode/espCode/components/ESPAsyncWebServer/src/WebRequest.cpp:182

0x400e099e: AsyncWebServerRequest::_parseLine() at /media/adarw/Data/Documents/Programing/cpp/robot/esp/espCodeIdf/espcode/espCode/espCode/components/ESPAsyncWebServer/src/WebRequest.cpp:571
 (inlined by) AsyncWebServerRequest::_parseLine() at /media/adarw/Data/Documents/Programing/cpp/robot/esp/espCodeIdf/espcode/espCode/espCode/components/ESPAsyncWebServer/src/WebRequest.cpp:554

0x400e0c01: AsyncWebServerRequest::_onData(void*, unsigned int) at /media/adarw/Data/Documents/Programing/cpp/robot/esp/espCodeIdf/espcode/espCode/espCode/components/ESPAsyncWebServer/src/WebRequest.cpp:123

0x400e0e11: std::_Function_handler<void (void*, AsyncClient*, void*, unsigned int), AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*)::{lambda(void*, AsyncClient*, void*, unsigned int)#8}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, std::_Any_data const&, unsigned int&&) at /media/adarw/Data/Documents/Programing/cpp/robot/esp/espCodeIdf/espcode/espCode/espCode/components/ESPAsyncWebServer/src/WebRequest.cpp:76
 (inlined by) _M_invoke at /home/adarw/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:297

0x400e3b2e: std::function<void (void*, AsyncClient*, void*, unsigned int)>::operator()(void*, AsyncClient*, void*, unsigned int) const at /home/adarw/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.4.0/bits/std_function.h:687
 (inlined by) AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) at /media/adarw/Data/Documents/Programing/cpp/robot/esp/espCodeIdf/espcode/espCode/espCode/components/AsyncTCP/src/AsyncTCP.cpp:934

0x400e3dde: AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, signed char) at /media/adarw/Data/Documents/Programing/cpp/robot/esp/espCodeIdf/espcode/espCode/espCode/components/AsyncTCP/src/AsyncTCP.cpp:1210
 (inlined by) _handle_async_event at /media/adarw/Data/Documents/Programing/cpp/robot/esp/espCodeIdf/espcode/espCode/espCode/components/AsyncTCP/src/AsyncTCP.cpp:162
 (inlined by) _async_service_task at /media/adarw/Data/Documents/Programing/cpp/robot/esp/espCodeIdf/espcode/espCode/espCode/components/AsyncTCP/src/AsyncTCP.cpp:197

ELF file SHA256: 041b001a8d748bff

Rebooting...
```
When I use the arduino-ide it works great, but when I try to use it in esp-idf it doesn't work.

Link to a github repo with my code: https://github.com/TopGgg/RobotCode

I'd love if someone could kindly point me whats wrong..
Thanks,
Adar

cosmoiler, 2023-01-03:
It looks like I have a similar problem in asp-idf. Any requests to the web server fail.

I use Arduno and ESPAsyncWebServer components in esp-idf v4.4.3.

When I use the arduino-ide (vscode + platformio) it works great, but when I try to use it in esp-idf it doesn't work.

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x401163a7  PS      : 0x00060230  A0      : 0x801164c5  A1      : 0x3ffd64c0  
0x401163a7: String::isSSO() const at C:/Users/Cosmoiler/Documents/Prog/Firmware/ESP32/workspace/cosmoiler-idf/components/arduino/cores/esp32/WString.h:324
 (inlined by) String::buffer() const at C:/Users/Cosmoiler/Documents/Prog/Firmware/ESP32/workspace/cosmoiler-idf/components/arduino/cores/esp32/WString.h:342
 (inlined by) String::c_str() const at C:/Users/Cosmoiler/Documents/Prog/Firmware/ESP32/workspace/cosmoiler-idf/components/arduino/cores/esp32/WString.h:250
 (inlined by) AsyncWebServerRequest::_removeNotInterestingHeaders() at C:/Users/Cosmoiler/Documents/Prog/Firmware/ESP32/workspace/cosmoiler-idf/components/ESPAsyncWebServer/src/WebRequest.cpp:184

A2      : 0x3ffc430c  A3      : 0x3ffc053c  A4      : 0x3ffc053c  A5      : 0x3ffc053c
A6      : 0x3ffc431c  A7      : 0x00000000  A8      : 0x8011641c  A9      : 0x3ffd64a0
A10     : 0x3ffc3750  A11     : 0xf3761096  A12     : 0x3ffc3750  A13     : 0x00000000
A14     : 0x0000000d  A15     : 0x00000000  SAR     : 0x0000001c  EXCCAUSE: 0x0000001c
EXCVADDR: 0xf37610a5  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe  

Backtrace: 0x401163a4:0x3ffd64c0 0x401164c2:0x3ffd6510 0x401165c5:0x3ffd6550 0x401167b9:0x3ffd65b0 0x4011aab1:0x3ffd65d0 0x4011ab01:0x3ffd6600 0x4011ab45:0x3ffd6620 0x4011acfe:0x3ffd6640 0x4011adcf:0x3ffd6660 0x4008e06d:0x3ffd6690
0x401163a4: LinkedListNode<AsyncWebHeader*>::value() at C:/Users/Cosmoiler/Documents/Prog/Firmware/ESP32/workspace/cosmoiler-idf/components/ESPAsyncWebServer/src/StringArray.h:35
 (inlined by) LinkedList<AsyncWebHeader*, LinkedListNode>::Iterator::operator*() const at C:/Users/Cosmoiler/Documents/Prog/Firmware/ESP32/workspace/cosmoiler-idf/components/ESPAsyncWebServer/src/StringArray.h:55
 (inlined by) AsyncWebServerRequest::_removeNotInterestingHeaders() at C:/Users/Cosmoiler/Documents/Prog/Firmware/ESP32/workspace/cosmoiler-idf/components/ESPAsyncWebServer/src/WebRequest.cpp:183

0x401164c2: AsyncWebServerRequest::_parseLine() at C:/Users/Cosmoiler/Documents/Prog/Firmware/ESP32/workspace/cosmoiler-idf/components/ESPAsyncWebServer/src/WebRequest.cpp:571

0x401165c5: AsyncWebServerRequest::_onData(void*, unsigned int) at C:/Users/Cosmoiler/Documents/Prog/Firmware/ESP32/workspace/cosmoiler-idf/components/ESPAsyncWebServer/src/WebRequest.cpp:123

0x401167b9: std::_Function_handler<void (void*, AsyncClient*, void*, unsigned int), AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*)::{lambda(void*, AsyncClient*, void*, unsigned int)#8}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, std::_Any_data const&, unsigned int&&) at C:/Users/Cosmoiler/Documents/Prog/Firmware/ESP32/workspace/cosmoiler-idf/components/ESPAsyncWebServer/src/WebRequest.cpp:76
 (inlined by) _M_invoke at c:\esp\tools\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:297

0x4011aab1: std::function<void (void*, AsyncClient*, void*, unsigned int)>::operator()(void*, AsyncClient*, void*, unsigned int) const at c:\esp\tools\tools\xtensa-esp32-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32-elf\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:687

0x4011ab01: AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) at C:/Users/Cosmoiler/Documents/Prog/Firmware/ESP32/workspace/cosmoiler-idf/managed_components/AsyncTCP/src/AsyncTCP.cpp:934

0x4011ab45: AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, signed char) at C:/Users/Cosmoiler/Documents/Prog/Firmware/ESP32/workspace/cosmoiler-idf/managed_components/AsyncTCP/src/AsyncTCP.cpp:1210

0x4011acfe: _handle_async_event(lwip_event_packet_t*) at C:/Users/Cosmoiler/Documents/Prog/Firmware/ESP32/workspace/cosmoiler-idf/managed_components/AsyncTCP/src/AsyncTCP.cpp:162

0x4011adcf: _async_service_task(void*) at C:/Users/Cosmoiler/Documents/Prog/Firmware/ESP32/workspace/cosmoiler-idf/managed_components/AsyncTCP/src/AsyncTCP.cpp:197

0x4008e06d: vPortTaskWrapper at C:/Esp/esp-idf/components/freertos/port/xtensa/port.c:131

ELF file SHA256: 4161853d3a2c994d

CPU halted.
```

cosmoiler, 2023-01-05:
Problem when calling the function `serve Static("/", SPIFFS, "/").setDefaultFile("index.html ");` and working with websockets.

barbiani, 2023-01-05:
How about this?

https://github.com/DanielKnoop/ESPAsyncWebServer/commit/a38a3ca980399ed81feb11d303dd880529c780e3

 And

https://github.com/me-no-dev/ESPAsyncWebServer/pull/952

cosmoiler, 2023-01-06:
Thanks for the direction of the search.

Here is this fork [Aircookie/ESPAsyncWebServer](https://github.com/Aircoookie/ESPAsyncWebServer ) works almost well (except for 'serveStatic("/", SPIFFS, "/")`)