me-no-dev/ESPAsyncWebServer #1251: Exception (3) crashes in umm_malloc_core

mamama1, 2022-12-19:
Hi

what follows is an exception trace I can reproduce by running a specific action in the web interface of my firmware.
I am afraid I won't be able to post a minimal sketch to reproduce on your side due to the massive complexity of the project. I could however share the complete source code if that helps, but it ain't very easy to set up the build environment.

I'm using the latest version of ESPAsyncWebServer and ESPAsyncTCP from platformio registry and latest Arduino Core 3.0.2.

I'm hoping for clues or general advice whether I'm likely to do something wrong or whether a bug in ESPAsyncWebServer is possible or likely or whether maybe this could even be a bug in the Arduino Core. As far as I'm aware, I'm not doing any slow/blocking stuff while I'm in sys context and I do a delay frequently in the main loop. Overall the whole code should be very non-blocking.
If some work is incoming through eg the webinterface, we queue it into a task list which then is worked on by a task processor, for example.

Any guidance would be highly appreciated. The only clue I have is, that this crash doesn't seem to occur if I in my task processor do not delete the last task it was working on, creating a memory leak, obviously. As far as I can tell, nothing is referencing to the deleted task later on, so I have no idea why deleting it could cause a crash. I delete the task like this, at the end of the function:

```
// Free the memory that was occupied by the data of this task (calls the destructor)
	delete currentTask;
	
	// Remove this task from the list
	this->Tasks.remove(currentTask);
	currentTask = NULL;
```

The line
0x40205bcc in WebSocketTaskProcessor::ProcessTasks() at src/WebSocketTaskProcessor.cpp:577
from the stack trace btw points to the snippet above, to the line "currentTask = NULL;". But the exceptions vary and sometimes I get a 3 and sometimes a 9. Sometimes the trace is way shorter also.

currentTask is 
`WebSocketTask *currentTask = this->Tasks.front();`

WebSocketTask is a class:
```
class WebSocketTask
{
	public:
	
		uint32_t ClientID;
		uint8_t	 *Data;
		size_t 	 DataLen;
		
		uint8_t	 UpdatedNodeID;

		WebSocketTask(uint32_t clientID, uint8_t *data, size_t len);
		WebSocketTask(uint8_t updatedNodeID);

		bool IsNodeUpdateTask(void);

		~WebSocketTask(void);

};
```

and Tasks is a list:
`std::list<WebSocketTask *> Tasks;`

```

Thanks for any guidance!

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (3):
epc1=0x40100ff2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40073dd0 depc=0x00000000

LoadStoreError: Processor internal physical address or data error during load or store
  epc1=0x40100ff2 in umm_malloc_core at C:\Users\user\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc/umm_malloc.cpp:635

LoadStoreError: Processor internal physical address or data error during load or store
  epc1=0x40100ff2 in umm_malloc_core at C:\Users\user\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc/umm_malloc.cpp:635

>>>stack>>>

[... 403 lines cut ...]
0x40101151 in umm_free_core at C:\Users\user\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc/umm_malloc.cpp:549
0x40101151 in umm_free_core at C:\Users\user\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc/umm_malloc.cpp:549
0x40101151 in umm_free_core at C:\Users\user\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc/umm_malloc.cpp:549
0x4010133e in free at C:\Users\user\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc/umm_malloc.cpp:595
0x4025723c in operator delete(void*) at /workdir/repo/gcc-gnu/libstdc++-v3/libsupc++/del_op.cc:50
0x4023ceac in operator delete(void*, unsigned int) at /workdir/repo/gcc-gnu/libstdc++-v3/libsupc++/del_ops.cc:34
0x40205004 in std::__cxx11::list<WebSocketTask*, std::allocator<WebSocketTask*> >::remove(WebSocketTask* const&) at c:\users\user\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.3.0\ext/new_allocator.h:133
0x4025723c in operator delete(void*) at /workdir/repo/gcc-gnu/libstdc++-v3/libsupc++/del_op.cc:50
0x40205bcc in WebSocketTaskProcessor::ProcessTasks() at src/WebSocketTaskProcessor.cpp:577
0x40100af5 in millis at C:\Users\user\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_wiring.cpp:188
0x40100af5 in millis at C:\Users\user\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_wiring.cpp:188
0x40106631 in ets_timer_arm_new at ??:?
0x402266f4 in esp_get_cycle_count at C:\Users\user\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_features.h:65
 (inlined by) EspClass::getCycleCount() at C:\Users\user\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Esp.h:219
 (inlined by) esp_yield_within_cont at C:\Users\user\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp:115
 (inlined by) __esp_yield at C:\Users\user\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp:121
0x402277c6 in delay at C:\Users\user\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_wiring.cpp:54
0x40206e11 in loop at src/main.cpp:354
0x40229c68 in std::_Function_handler<void (bool), void (*)(bool)>::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\user\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.3.0\bits/std_function.h:271
0x40229c54 in std::_Function_handler<void (bool), void (*)(bool)>::_M_invoke(std::_Any_data const&, bool&&) at c:\users\user\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.3.0\bits/std_function.h:289
0x401005d0 in ets_post at C:\Users\user\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp:181
0x402267ea in loop_wrapper() at C:\Users\user\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp:201

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

mamama1, 2022-12-19:
This is another crash, again in umm_malloc_cpp:635 :

```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (3):
epc1=0x40100ff2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4001ef20 depc=0x00000000

LoadStoreError: Processor internal physical address or data error during load or store
  epc1=0x40100ff2 in umm_malloc_core at C:\Users\neumluc.NNET\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc/umm_malloc.cpp:635

LoadStoreError: Processor internal physical address or data error during load or store
  epc1=0x40100ff2 in umm_malloc_core at C:\Users\neumluc.NNET\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc/umm_malloc.cpp:635

>>>stack>>>

ctx: cont
sp: 3ffffc50 end: 3fffffc0 offset: 0190
3ffffde0:  20363534 30303028 29303030 00000065  
3ffffdf0:  3fff1bfc 00000020 00000066 00000065
3ffffe00:  00000000 00000020 00000008 40101375
3ffffe10:  39203163 4020c4d2 00000008 40226367
3ffffe20:  00000000 3fff1dfc 3fff5d84 4020c4d2
3ffffe30:  00000000 3ffffe40 0000000c 40205004
3ffffe40:  12d5469b 3ffffeb8 3fff1be8 3fff26b0
3ffffe50:  3fff1dfc 3fff1bfc 3fff1dfc 4020d57f
3ffffe60:  3fff26b4 3fff1bfc 3fff1be8 40205b60
3ffffe70:  00000063 62ff1655 b4721e53 00000090
3ffffe80:  00000000 4bc6a7f0 3fa0040d 67ae147a
3ffffe90:  00000000 00000000 4bc6a7f0 00000000
3ffffea0:  00000000 00000000 40100af5 00000000
3ffffeb0:  3ffffecc 3fff1be8 3ffffecc 3fff1be8
3ffffec0:  40229c10 40204e10 3f9f0212 00000063
3ffffed0:  3fff5e2c 4bc6a7f0 3fa014fb bd2f1a9f
3ffffee0:  00000000 00000000 4bc6a7f0 00000000
3ffffef0:  4010645a 3fff0d28 40100af5 00000000  
3fffff00:  001049bf 3fff2494 00000000 00000001
3fffff10:  12d545b2 3fff32fc 00000004 3fff31e8
3fffff20:  3fff4d34 4020a208 00000004 3fff31e8
3fffff30:  3fffdad0 00000000 3fff2494 3fff31e8
3fffff40:  3fffdad0 00000000 3ffe8910 40206c72
3fffff50:  007a1200 cc320b8c 00000000 00000001
3fffff60:  00000000 00000000 40229c68 40229c54
3fffff70:  007a1200 cc320a1b 00005d00 00000914
3fffff80:  00000000 00000000 00000001 401005d0
3fffff90:  3fffdad0 00000000 3ffe8910 3fff31e8
3fffffa0:  3fffdad0 00000000 3ffe8910 402267ea
3fffffb0:  feefeffe feefeffe feefeffe 40101751
<<<stack<<<

0x40101375 in malloc at C:\Users\neumluc.NNET\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc/umm_malloc.cpp:821
0x4020c4d2 in LinkedListNode<AsyncWebSocketMessageBuffer*>::LinkedListNode(AsyncWebSocketMessageBuffer*) at .pio\libdeps\ESP8266\ESP Async WebServer\src/StringArray.h:32
 (inlined by) LinkedList<AsyncWebSocketMessageBuffer*, LinkedListNode>::add(AsyncWebSocketMessageBuffer* const&) at .pio\libdeps\ESP8266\ESP Async WebServer\src/StringArray.h:67
 (inlined by) AsyncWebSocket::makeBuffer(unsigned char*, unsigned int) at .pio\libdeps\ESP8266\ESP Async WebServer\src/AsyncWebSocket.cpp:1222
0x40226367 in operator new(unsigned int) at C:\Users\neumluc.NNET\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/abi.cpp:41
0x4020c4d2 in LinkedListNode<AsyncWebSocketMessageBuffer*>::LinkedListNode(AsyncWebSocketMessageBuffer*) at .pio\libdeps\ESP8266\ESP Async WebServer\src/StringArray.h:32
 (inlined by) LinkedList<AsyncWebSocketMessageBuffer*, LinkedListNode>::add(AsyncWebSocketMessageBuffer* const&) at .pio\libdeps\ESP8266\ESP Async WebServer\src/StringArray.h:67
 (inlined by) AsyncWebSocket::makeBuffer(unsigned char*, unsigned int) at .pio\libdeps\ESP8266\ESP Async WebServer\src/AsyncWebSocket.cpp:1222
0x40205004 in std::__cxx11::list<WebSocketTask*, std::allocator<WebSocketTask*> >::remove(WebSocketTask* const&) at c:\users\neumluc.nnet\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.3.0\ext/new_allocator.h:133
0x4020d57f in AsyncWebSocket::textAll(char const*, unsigned int) at .pio\libdeps\ESP8266\ESP Async WebServer\src/AsyncWebSocket.cpp:971
0x40205b60 in WebSocketTaskProcessor::ProcessTasks() at src/WebSocketTaskProcessor.cpp:550
0x40100af5 in millis at C:\Users\neumluc.NNET\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_wiring.cpp:188
0x40229c10 in std::_Function_handler<void (char*), WebSocketTaskProcessor::ProcessTasks()::{lambda(char*)#1}>::_M_manager(std::_Any_data&, std::_Function_handler<void (char*), WebSocketTaskProcessor::ProcessTasks()::{lambda(char*)#1}> const&, std::_Manager_operation) at c:\users\neumluc.nnet\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.3.0\bits/std_function.h:271
0x40204e10 in std::_Function_handler<void (char*), WebSocketTaskProcessor::ProcessTasks()::{lambda(char*)#1}>::_M_invoke(std::_Any_data const&, char*&&) at c:\users\neumluc.nnet\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.3.0\bits/std_function.h:289
0x40101375 in malloc at C:\Users\neumluc.NNET\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc/umm_malloc.cpp:821
0x4020c4d2 in LinkedListNode<AsyncWebSocketMessageBuffer*>::LinkedListNode(AsyncWebSocketMessageBuffer*) at .pio\libdeps\ESP8266\ESP Async WebServer\src/StringArray.h:32
 (inlined by) LinkedList<AsyncWebSocketMessageBuffer*, LinkedListNode>::add(AsyncWebSocketMessageBuffer* const&) at .pio\libdeps\ESP8266\ESP Async WebServer\src/StringArray.h:67
 (inlined by) AsyncWebSocket::makeBuffer(unsigned char*, unsigned int) at .pio\libdeps\ESP8266\ESP Async WebServer\src/AsyncWebSocket.cpp:1222
[... cut ...] std_function.h:289
0x401005d0 in ets_post at C:\Users\neumluc.NNET\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp:181
0x402267ea in loop_wrapper() at C:\Users\neumluc.NNET\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp:201
0x40101751 in cont_wrapper at C:\Users\neumluc.NNET\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/cont.S:81

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

gibo77, 2022-12-19:
Looks like hardware issue. Check your power supply. 

ctx: cont
sp: 3ffffc50 end: 3fffffc0 offset: 0190
3ffffde0:  20363534 30303028 29303030 00000065  
3ffffdf0:  3fff1bfc 00000020 00000066 00000065
3ffffe00:  00000000 00000020 00000008 40101375
3ffffe10:  39203163 4020c4d2 00000008 40226367
3ffffe20:  00000000 3fff1dfc 3fff5d84 4020c4d2
3ffffe30:  00000000 3ffffe40 0000000c 40205004
3ffffe40:  12d5469b 3ffffeb8 3fff1be8 3fff26b0
3ffffe50:  3fff1dfc 3fff1bfc 3fff1dfc 4020d57f
3ffffe60:  3fff26b4 3fff1bfc 3fff1be8 40205b60
3ffffe70:  00000063 62ff1655 b4721e53 00000090
3ffffe80:  00000000 4bc6a7f0 3fa0040d 67ae147a
3ffffe90:  00000000 00000000 4bc6a7f0 00000000
3ffffea0:  00000000 00000000 40100af5 00000000
3ffffeb0:  3ffffecc 3fff1be8 3ffffecc 3fff1be8
3ffffec0:  40229c10 40204e10 3f9f0212 00000063
3ffffed0:  3fff5e2c 4bc6a7f0 3fa014fb bd2f1a9f
3ffffee0:  00000000 00000000 4bc6a7f0 00000000
3ffffef0:  4010645a 3fff0d28 40100af5 00000000  
3fffff00:  001049bf 3fff2494 00000000 00000001
3fffff10:  12d545b2 3fff32fc 00000004 3fff31e8
3fffff20:  3fff4d34 4020a208 00000004 3fff31e8
3fffff30:  3fffdad0 00000000 3fff2494 3fff31e8
3fffff40:  3fffdad0 00000000 3ffe8910 40206c72
3fffff50:  007a1200 cc320b8c 00000000 00000001
3fffff60:  00000000 00000000 40229c68 40229c54
3fffff70:  007a1200 cc320a1b 00005d00 00000914
3fffff80:  00000000 00000000 00000001 401005d0
3fffff90:  3fffdad0 00000000 3ffe8910 3fff31e8
3fffffa0:  3fffdad0 00000000 3ffe8910 402267ea
3fffffb0:  feefeffe feefeffe feefeffe 40101751
<<<stack<<<

0x40101375 in malloc at C:\Users\neumluc.NNET\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc/umm_malloc.cpp:821
0x4020c4d2 in LinkedListNode<AsyncWebSocketMessageBuffer*>::LinkedListNode(AsyncWebSocketMessageBuffer*) at .pio\libdeps\ESP8266\ESP Async WebServer\src/StringArray.h:32
 (inlined by) LinkedList<AsyncWebSocketMessageBuffer*, LinkedListNode>::add(AsyncWebSocketMessageBuffer* const&) at .pio\libdeps\ESP8266\ESP Async WebServer\src/StringArray.h:67
 (inlined by) AsyncWebSocket::makeBuffer(unsigned char*, unsigned int) at .pio\libdeps\ESP8266\ESP Async WebServer\src/AsyncWebSocket.cpp:1222
0x40226367 in operator new(unsigned int) at C:\Users\neumluc.NNET\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/abi.cpp:41
0x4020c4d2 in LinkedListNode<AsyncWebSocketMessageBuffer*>::LinkedListNode(AsyncWebSocketMessageBuffer*) at .pio\libdeps\ESP8266\ESP Async WebServer\src/StringArray.h:32
 (inlined by) LinkedList<AsyncWebSocketMessageBuffer*, LinkedListNode>::add(AsyncWebSocketMessageBuffer* const&) at .pio\libdeps\ESP8266\ESP Async WebServer\src/StringArray.h:67
 (inlined by) AsyncWebSocket::makeBuffer(unsigned char*, unsigned int) at .pio\libdeps\ESP8266\ESP Async WebServer\src/AsyncWebSocket.cpp:1222
0x40205004 in std::__cxx11::list<WebSocketTask*, std::allocator<WebSocketTask*> >::remove(WebSocketTask* const&) at c:\users\neumluc.nnet\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.3.0\ext/new_allocator.h:133
0x4020d57f in AsyncWebSocket::textAll(char const*, unsigned int) at .pio\libdeps\ESP8266\ESP Async WebServer\src/AsyncWebSocket.cpp:971
0x40205b60 in WebSocketTaskProcessor::ProcessTasks() at src/WebSocketTaskProcessor.cpp:550
0x40100af5 in millis at C:\Users\neumluc.NNET\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_wiring.cpp:188
0x40229c10 in std::_Function_handler<void (char*), WebSocketTaskProcessor::ProcessTasks()::{lambda(char*)#1}>::_M_manager(std::_Any_data&, std::_Function_handler<void (char*), WebSocketTaskProcessor::ProcessTasks()::{lambda(char*)#1}> const&, std::_Manager_operation) at c:\users\neumluc.nnet\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.3.0\bits/std_function.h:271
0x40204e10 in std::_Function_handler<void (char*), WebSocketTaskProcessor::ProcessTasks()::{lambda(char*)#1}>::_M_invoke(std::_Any_data const&, char*&&) at c:\users\neumluc.nnet\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.3.0\bits/std_function.h:289
0x40101375 in malloc at C:\Users\neumluc.NNET\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc/umm_malloc.cpp:821
0x4020c4d2 in LinkedListNode<AsyncWebSocketMessageBuffer*>::LinkedListNode(AsyncWebSocketMessageBuffer*) at .pio\libdeps\ESP8266\ESP Async WebServer\src/StringArray.h:32
 (inlined by) LinkedList<AsyncWebSocketMessageBuffer*, LinkedListNode>::add(AsyncWebSocketMessageBuffer* const&) at .pio\libdeps\ESP8266\ESP Async WebServer\src/StringArray.h:67
 (inlined by) AsyncWebSocket::makeBuffer(unsigned char*, unsigned int) at .pio\libdeps\ESP8266\ESP Async WebServer\src/AsyncWebSocket.cpp:1222
0x40226367 in operator new(unsigned int) at C:\Users\neumluc.NNET\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/abi.cpp:41
0x4020c4d2 in LinkedListNode<AsyncWebSocketMessageBuffer*>::LinkedListNode(AsyncWebSocketMessageBuffer*) at .pio\libdeps\ESP8266\ESP Async WebServer\src/StringArray.h:32
 (inlined by) LinkedList<AsyncWebSocketMessageBuffer*, LinkedListNode>::add(AsyncWebSocketMessageBuffer* const&) at .pio\libdeps\ESP8266\ESP Async WebServer\src/StringArray.h:67
 (inlined by) AsyncWebSocket::makeBuffer(unsigned char*, unsigned int) at .pio\libdeps\ESP8266\ESP Async WebServer\src/AsyncWebSocket.cpp:1222
[... cut ...] its/std_function.h:289
0x401005d0 in ets_post at C:\Users\neumluc.NNET\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp:181
0x402267ea in loop_wrapper() at C:\Users\neumluc.NNET\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp:201
0x40101751 in cont_wrapper at C:\Users\neumluc.NNET\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/cont.S:81

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

mamama1, 2022-12-19:
> Looks like hardware issue. Check your power supply.

I'll attach a scope to it, but I doubt it. ESP8266 has a 1000uF electrolytic cap (nichicon) very near and is connected via a short, high quality USB cable to a powered USB hub...

mamama1, 2022-12-19:
Now a very short crash:
```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (3):
epc1=0x40107226 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4002e704 depc=0x00000000

LoadStoreError: Processor internal physical address or data error during load or store
  epc1=0x40107226 in umm_disconnect_from_free_list at C:\Users\neumluc.NNET\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc/umm_malloc.cpp:320

LoadStoreError: Processor internal physical address or data error during load or store
  epc1=0x40107226 in umm_disconnect_from_free_list at C:\Users\neumluc.NNET\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc/umm_malloc.cpp:320

>>>stack>>>

ctx: cont
sp: 3ffffc80 end: 3fffffc0 offset: 0190
3ffffe10:  3ffe93c8 00000008 3fff6c2c 4020bc6a  
3ffffe20:  000003f8 3ffe8914 00001fc4 40101151
3ffffe30:  3fff1c1c 3fff66d4 3fff6c2c 00000000
3ffffe40:  3fff1e1c 00000020 3fff6bdc 4010133e
3ffffe50:  3fff1e1c 3fff692c 3fff692c 40204dcc
3ffffe60:  00000001 3fff1c1c 3fff1c08 40205bb9
3ffffe70:  33f9fe44 62ff1655 b4721e53 00000090
3ffffe80:  01060200 00000000 00000000 00000000
3ffffe90:  00000000 00000000 00000000 00000000
3ffffea0:  00000000 00000000 40100000 00000000
3ffffeb0:  00018b99 3fff1c08 3ffffecc 3fff1c08
3ffffec0:  01c9698d e4a04425 29300213 00000000
3ffffed0:  3fff692c 33f9fe44 62ff1655 b4721e53
3ffffee0:  00000090 01060200 00000000 00000000
3ffffef0:  00000000 00000000 00000000 00000000
3fffff00:  00000000 00000000 00000000 02010000
3fffff10:  01c90006 3fff331c 00000004 3fff3208  
3fffff20:  3fffdad0 00000000 00000004 3fff3208
3fffff30:  3fffdad0 00000000 3fff24b4 3fff3208
3fffff40:  3fffdad0 00000000 3ffe8910 40206c72
3fffff50:  007a1200 e7233f7b 00000000 00000001
3fffff60:  00000000 00000000 40229cc0 40229cac
3fffff70:  007a1200 e7233e0a 00003e00 00000970
3fffff80:  00000000 00000000 00000001 401005d0
3fffff90:  3fffdad0 00000000 3ffe8910 3fff3208
3fffffa0:  3fffdad0 00000000 3ffe8910 40226842
3fffffb0:  feefeffe feefeffe feefeffe 40101751
<<<stack<<<

0x4020bc6a in AsyncWebSocketBasicMessage::~AsyncWebSocketBasicMessage() at .pio\libdeps\ESP8266\ESP Async WebServer\src/AsyncWebSocket.cpp:322
0x40101151 in umm_free_core at C:\Users\neumluc.NNET\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc/umm_malloc.cpp:549
0x4010133e in free at C:\Users\neumluc.NNET\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc/umm_malloc.cpp:595
0x40204dcc in WebSocketTask::~WebSocketTask() at src/WebSocketTask.cpp:58
0x40205bb9 in WebSocketTaskProcessor::ProcessTasks() at src/WebSocketTaskProcessor.cpp:573 (discriminator 1)
0x40100000 in _stext at ??:?
0x4020bc6a in AsyncWebSocketBasicMessage::~AsyncWebSocketBasicMessage() at .pio\libdeps\ESP8266\ESP Async WebServer\src/AsyncWebSocket.cpp:322
0x40101151 in umm_free_core at C:\Users\neumluc.NNET\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc/umm_malloc.cpp:549
0x4010133e in free at C:\Users\neumluc.NNET\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc/umm_malloc.cpp:595
0x40204dcc in WebSocketTask::~WebSocketTask() at src/WebSocketTask.cpp:58
0x40205bb9 in WebSocketTaskProcessor::ProcessTasks() at src/WebSocketTaskProcessor.cpp:573 (discriminator 1)
0x40100000 in _stext at ??:?
0x40206c72 in loop at src/main.cpp:229
0x40229cc0 in std::_Function_handler<void (bool), void (*)(bool)>::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at c:\users\neumluc.nnet\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.3.0\bits/std_function.h:271
0x40229cac in std::_Function_handler<void (bool), void (*)(bool)>::_M_invoke(std::_Any_data const&, bool&&) at c:\users\neumluc.nnet\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.3.0\bits/std_function.h:289
0x401005d0 in ets_post at C:\Users\neumluc.NNET\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp:181
0x40226842 in loop_wrapper() at C:\Users\neumluc.NNET\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp:201
0x40101751 in cont_wrapper at C:\Users\neumluc.NNET\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/cont.S:81

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

mamama1, 2022-12-19:
And another one, remember - all goes away if I do not delete my WebSocketTask *currentTask as mentioned in the first post.
I have one WTF moment right after the other

```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (9):
epc1=0x4020c4e4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x22657465 depc=0x00000000

LoadStoreAlignmentCause: Load or store to an unaligned address
  epc1=0x4020c4e4 in LinkedList<AsyncWebSocketMessageBuffer*, LinkedListNode>::add(AsyncWebSocketMessageBuffer* const&) at .pio\libdeps\ESP8266\ESP Async WebServer\src/StringArray.h:72
     (inlined by) AsyncWebSocket::makeBuffer(unsigned char*, unsigned int) at .pio\libdeps\ESP8266\ESP Async WebServer\src/AsyncWebSocket.cpp:1225

LoadStoreAlignmentCause: Load or store to an unaligned address
  epc1=0x4020c4e4 in LinkedList<AsyncWebSocketMessageBuffer*, LinkedListNode>::add(AsyncWebSocketMessageBuffer* const&) at .pio\libdeps\ESP8266\ESP Async WebServer\src/StringArray.h:72
     (inlined by) AsyncWebSocket::makeBuffer(unsigned char*, unsigned int) at .pio\libdeps\ESP8266\ESP Async WebServer\src/AsyncWebSocket.cpp:1225

>>>stack>>>

ctx: cont
sp: 3ffffc90 end: 3fffffc0 offset: 0190
3ffffe20:  00000000 3fff1e1c 3fff5b2c 4020c4d2  
3ffffe30:  00000000 3ffffe40 0000000c 40205004
3ffffe40:  02eb1cdd 3ffffeb8 3fff1c08 3fff26d0
3ffffe50:  3fff1e1c 3fff1c1c 3fff1e1c 4020d5d7
3ffffe60:  3fff26d4 3fff1c1c 3fff1c08 40205b60
3ffffe70:  00000063 62ff1655 b4721e53 00000090
3ffffe80:  00000000 4bc6a7f0 09dbf3c3 bb22d0e5
3ffffe90:  00000000 00000000 4bc6a7f0 00000000
3ffffea0:  00000000 00000000 40100af5 00000000
3ffffeb0:  3ffffecc 3fff1c08 3ffffecc 3fff1c08
3ffffec0:  40229c68 40204e10 09cf0212 00000063
3ffffed0:  3fff61cc 4bc6a7f0 09dc0483 04dd2f1a  
3ffffee0:  00000000 00000000 4bc6a7f0 00000000
3ffffef0:  4010645a 3fff0d48 40100af5 00000000
3fffff00:  00028626 03147e1b 3fff331c 00000000
3fffff10:  02eb1bf5 3fff331c 00000004 3fff3208
3fffff20:  3fffdad0 00000000 00000004 3fff3208
3fffff30:  3fffdad0 00000000 3fff24b4 3fff3208
3fffff40:  3fffdad0 00000000 3ffe8910 40206c72
3fffff50:  007a1200 0117d174 00000000 00000001
3fffff60:  00000000 00000000 40229cc0 40229cac
3fffff70:  007a1200 0117d003 00005d00 000009a0
3fffff80:  00000000 00000000 00000001 401005d0
3fffff90:  3fffdad0 00000000 3ffe8910 3fff3208
3fffffa0:  3fffdad0 00000000 3ffe8910 40226842
3fffffb0:  feefeffe feefeffe feefeffe 40101751
<<<stack<<<

0x4020c4d2 in LinkedListNode<AsyncWebSocketMessageBuffer*>::LinkedListNode(AsyncWebSocketMessageBuffer*) at .pio\libdeps\ESP8266\ESP Async WebServer\src/StringArray.h:32
 (inlined by) LinkedList<AsyncWebSocketMessageBuffer*, LinkedListNode>::add(AsyncWebSocketMessageBuffer* const&) at .pio\libdeps\ESP8266\ESP Async WebServer\src/StringArray.h:67
 (inlined by) AsyncWebSocket::makeBuffer(unsigned char*, unsigned int) at .pio\libdeps\ESP8266\ESP Async WebServer\src/AsyncWebSocket.cpp:1225
0x40205004 in std::__cxx11::list<WebSocketTask*, std::allocator<WebSocketTask*> >::remove(WebSocketTask* const&) at c:\users\neumluc.nnet\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.3.0\ext/new_allocator.h:133
0x4020d5d7 in AsyncWebSocket::textAll(char const*, unsigned int) at .pio\libdeps\ESP8266\ESP Async WebServer\src/AsyncWebSocket.cpp:974
0x40205b60 in WebSocketTaskProcessor::ProcessTasks() at src/WebSocketTaskProcessor.cpp:550
0x40100af5 in millis at C:\Users\neumluc.NNET\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_wiring.cpp:188
0x40229c68 in std::_Function_handler<void (char*), WebSocketTaskProcessor::ProcessTasks()::{lambda(char*)#1}>::_M_manager(std::_Any_data&, std::_Function_handler<void (char*), WebSocketTaskProcessor::ProcessTasks()::{lambda(char*)#1}> const&, std::_Manager_operation) at c:\users\neumluc.nnet\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.3.0\bits/std_function.h:271
0x40204e10 in std::_Function_handler<void (char*), WebSocketTaskProcessor::ProcessTasks()::{lambda(char*)#1}>::_M_invoke(std::_Any_data const&, char*&&) at c:\users\neumluc.nnet\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.3.0\bits/std_function.h:289
0x4020c4d2 in LinkedListNode<AsyncWebSocketMessageBuffer*>::LinkedListNode(AsyncWebSocketMessageBuffer*) at .pio\libdeps\ESP8266\ESP Async WebServer\src/StringArray.h:32
 (inlined by) LinkedList<AsyncWebSocketMessageBuffer*, LinkedListNode>::add(AsyncWebSocketMessageBuffer* const&) at .pio\libdeps\ESP8266\ESP Async WebServer\src/StringArray.h:67
 (inlined by) AsyncWebSocket::makeBuffer(unsigned char*, unsigned int) at .pio\libdeps\ESP8266\ESP Async WebServer\src/AsyncWebSocket.cpp:1225
0x40205004 in std::__cxx11::list<WebSocketTask*, std::allocator<WebSocketTask*> >::remove(WebSocketTask* const&) at c:\users\neumluc.nnet\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.3.0\ext/new_allocator.h:133
0x4020d5d7 in AsyncWebSocket::textAll(char const*, unsigned int) at .pio\libdeps\ESP8266\ESP Async WebServer\src/AsyncWebSocket.cpp:974
0x40205b60 in WebSocketTaskProcessor::ProcessTasks() at src/WebSocketTaskProcessor.cpp:550
0x40100af5 in millis at C:\Users\neumluc.NNET\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_wiring.cpp:188
0x40229c68 in std::_Function_handler<void (char*), WebSocketTaskProcessor::ProcessTasks()::{lambda(char*)#1}>::_M_manager(std::_Any_data&, std::_Function_handler<void (char*), WebSocketTaskProcessor::ProcessTasks(): [... cut ...] std_function.h:289
0x401005d0 in ets_post at C:\Users\neumluc.NNET\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp:181
0x40226842 in loop_wrapper() at C:\Users\neumluc.NNET\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp:201
0x40101751 in cont_wrapper at C:\Users\neumluc.NNET\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/cont.S:81

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

mamama1, 2022-12-19:
yet another variant:

```
User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Panic umm_malloc.cpp:229 umm_get_ptr_context

>>>stack>>>

ctx: cont
sp: 3ffffdd0 end: 3fffffc0 offset: 0000
3ffffdd0:  3ffffe50 3fff6114 3fff60fc 3fff26d0
3ffffde0:  000000fe 00000000 00000000 00000000
3ffffdf0:  00000000 00000000 00000000 3fff3250
3ffffe00:  0000006b 000000a1 3fff6114 3fff26d0
3ffffe10:  3fff1e1c 00000001 00000001 40226d7a  
3ffffe20:  00000000 00000000 00000000 40226de7
3ffffe30:  3fff1e1c 3fff5b14 00000000 4010122c
3ffffe40:  00000000 00000020 00000001 40101339
3ffffe50:  3fff1e1c 3fff5b2c 3fff5b2c 40204dcc  
3ffffe60:  3fff26d4 3fff1c1c 3fff1c08 40205bb9
3ffffe70:  00000063 62ff1655 b4721e53 00000090
3ffffe80:  00000000 4bc6a7f0 00910de7 476c8b43  
3ffffe90:  00000000 00000000 4bc6a7f0 00000000
3ffffea0:  00000000 00000000 40100af5 00000000
3ffffeb0:  3ffffecc 3fff1c08 3ffffecc 3fff1c08
3ffffec0:  40229c68 40204e10 008f0212 00000063  
3ffffed0:  3fff5b2c 4bc6a7f0 00911e7f 86e978d4
3ffffee0:  00000000 00000000 4bc6a7f0 00000000
3ffffef0:  4010645a 3fff0d48 40100af5 00000000  
3fffff00:  00002526 002d1518 3fff331c 00000000
3fffff10:  002af48b 3fff331c 00000004 3fff3208
3fffff20:  3fffdad0 00000000 00000004 3fff3208
3fffff30:  3fffdad0 00000000 3fff24b4 3fff3208  
3fffff40:  3fffdad0 00000000 3ffe8910 40206c72
3fffff50:  007a1200 318ec4a8 00000000 00000000
3fffff60:  00000000 00000000 40229cc0 40229cac  
3fffff70:  007a1200 318ec337 00005500 00000994
3fffff80:  00000000 00000000 00000001 401005d0
3fffff90:  3fffdad0 00000000 3ffe8910 3fff3208  
3fffffa0:  3fffdad0 00000000 3ffe8910 40226842
3fffffb0:  feefeffe feefeffe feefeffe 40101751  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

mamama1, 2022-12-19:
Oh wow, we got a 28

```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x4020d6f9 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
  epc1=0x4020d6f9 in AsyncWebSocketClient::_onAck(unsigned int, unsigned int) at .pio\libdeps\ESP8266\ESP Async WebServer\src/AsyncWebSocket.cpp:522

LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
  epc1=0x4020d6f9 in AsyncWebSocketClient::_onAck(unsigned int, unsigned int) at .pio\libdeps\ESP8266\ESP Async WebServer\src/AsyncWebSocket.cpp:522

>>>stack>>>

ctx: sys
sp: 3fffec50 end: 3fffffb0 offset: 0190
3fffede0:  3fffee70 3fff6a74 3fff624c 4020d62d  
3fffedf0:  00000032 00000000 4bc6a7f0 00000000
3fffee00:  00007fff 032ae5bb 40100af5 00000067
3fffee10:  3fffee70 3fff6a74 3fff624c 4020d731
3fffee20:  00f0078e 00004288 3fff6a3c 40100181
3fffee30:  3fff6a74 3fff624c 00000067 00000032
3fffee40:  3fffee70 3fff6a74 3fff624c 40208f0e
3fffee50:  3fff3afc 3fff5dc4 3fff3b00 3fff3af0
3fffee60:  3fff5dc4 3fff3af4 3fff3af5 40208f50
3fffee70:  3fff5c20 3fff5c14 3fff5aec 3fff3af0
3fffee80:  3fff5dc4 3fff3af4 3fff3af5 4026a910
3fffee90:  00000000 3ffeb9b4 00000000 3fffdab0
3fffeea0:  00000010 00000000 3fff0d88 00000000
3fffeeb0:  3fff0000 00000020 00000048 3fff35e4
3fffeec0:  3fff3924 3fff5b0a 3fff5aec 402680d5  
3fffeed0:  00000014 3fff35e4 0000132c 40101151
3fffeee0:  00000000 3fff4d54 3ffef5e0 3fff54cc
3fffeef0:  3fffdc80 00000020 3fff5f44 3fff54cc
3fffef00:  3fff35e4 00000008 3fff5aec 40260a8d
3fffef10:  3fffdc80 3fff5454 3fff5f44 4026088c
3fffef20:  4028ed29 3fff5454 3fff5f44 4028ed3b
3fffef30:  3fff5afc 3fff5aec 00000000 3fffdcb0
3fffef40:  4028601b 00000000 3fff5f44 4028a57f
3fffef50:  40000f49 3fffdab0 3fffdab0 40000f49
3fffef60:  40000e19 000a49e1 00000000 00000005
3fffef70:  3fffefc0 aa55aa55 000000ed 40105d55
3fffef80:  40105d5b 00000000 00000005 4010131c
3fffef90:  4010000d 74333131 5f656d69 401000ab
3fffefa0:  4027af5c 3fffef3c 4027af0d 3fffff18
3fffefb0:  3fffffc0 00000000 00000000 feefeffe
3fffefc0:  feefeffe feefeffe feefeffe feefeffe
3fffefd0:  feefeffe feefeffe feefeffe feefeffe
3fffefe0:  feefeffe feefeffe feefeffe feefeffe
3fffeff0:  feefeffe feefeffe feefeffe feefeffe
3ffff000:  feefeffe feefeffe feefeffe feefeffe
3ffff010:  feefeffe feefeffe feefeffe feefeffe
3ffff020:  feefeffe feefeffe feefeffe feefeffe
3ffff030:  feefeffe feefeffe feefeffe feefeffe
3ffff040:  feefeffe feefeffe feefeffe feefeffe
3ffff050:  feefeffe feefeffe feefeffe feefeffe
3ffff060:  feefeffe feefeffe feefeffe feefeffe
3ffff070:  feefeffe feefeffe feefeffe feefeffe
3ffff080:  feefeffe feefeffe feefeffe feefeffe
3ffff090:  feefeffe feefeffe feefeffe feefeffe
3ffff0a0:  feefeffe feefeffe feefeffe feefeffe
3ffff0b0:  feefeffe feefeffe feefeffe feefeffe
3ffff0c0:  feefeffe feefeffe feefeffe feefeffe
3ffff0d0:  feefeffe feefeffe feefeffe feefeffe
3ffff0e0:  feefeffe feefeffe feefeffe feefeffe
3ffff0f0:  feefeffe feefeffe feefeffe feefeffe
3ffff100:  feefeffe feefeffe feefeffe feefeffe
3ffff110:  feefeffe feefeffe feefeffe feefeffe
[... 350 lines cut ...]
0x402374a0 in __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:179
0x40237564 in __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:232
0x40237564 in __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:232
0x40237564 in __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:232
0x40231a31 in _printf_i at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c:196 (discriminator 1)
0x40237564 in __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:232
0x402374a0 in __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:179
0x402374a0 in __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:179
0x402970f3 in espconn_regist_recvcb at ??:?
0x40231b60 in _printf_i at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c:246
0x40237564 in __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:232
0x402970f3 in espconn_regist_recvcb at ??:?
0x402970f3 in espconn_regist_recvcb at ??:?
0x402970f5 in espconn_regist_recvcb at ??:?
0x40237765 in _svfprintf_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:528
0x4029015c in espconn_udp_server at ??:?
0x40231b60 in _printf_i at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c:246
0x40237564 in __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:232
0x4029015c in espconn_udp_server at ??:?
0x4029015c in espconn_udp_server at ??:?
0x4029015e in espconn_udp_server at ??:?
0x40237765 in _svfprintf_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:528

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

mamama1, 2022-12-19:
Now I got the exception from my 2nd post again. So it is not absolutely random but the same bunch of exceptions seems to be repeating. Sometimes I can go bonkers in the webinterface without anything happening and sometimes it crashes almost every click. I guess that happens when you do much stuff asynchronously... Not easy to debug.

nickion, 2023-05-01:
Getting similar backtraces with what is likely related if not the same issue with the 8266. I'm using web sockets and serving pages, operating as AP_STA, with various networking operations such as network scans triggered from loop as required, I found adding a call to yield() in various points in loop helped, but there can still be instability. Receiving pongs from a couple of clients after a pingAll() tends to lead to a crash soon after as well. The ottowinter esphome port seems preferable as it's smaller and uses around 10K less ram, but the ping/pong issues remains unfortunately.