me-no-dev/ESPAsyncWebServer #1253: How to download large (chunked) files from flash?

av4625, 2022-12-21:
I was wondering if you could help me understand how I can download large files (10-100kb) from an SD card?
I am trying to use chunked responses but it doesn't work currently.
```C++
server_.on("/download", HTTP_GET, [this](AsyncWebServerRequest *request)
{
    File file{SD.open("/path/to/file.json", FILE_READ)};

    AsyncWebServerResponse *response = request->beginChunkedResponse(
        "application/json",
        [file](
            uint8_t* buffer,
            const size_t max_len,
            const size_t index) mutable -> size_t
        {
            Serial.print("Max length: ");
            Serial.println(max_len);

            // Tried sending less than max length
            const size_t length{file.read(buffer, (max_len >> 1))};
            Serial.print("Length: ");
            Serial.println(length);

            if (length == 0)
            {
                file.close();
            }

           return length;
        });

    // Force download
    response->addHeader("Content-Disposition", "attachment; filename=\"file.json\"");
    response->addHeader("Transfer-Encoding", "chunked");
    request->send(response);
});
```

When I print `max_len` it starts at ~5500, it then just keeps getting smaller by the amount I `read` from the `SD` card. It does this until it gets so small and then the ESP32 and browser just hang. In the network tab of the browser it just spins waiting on this request.

I have tried the following which uses `AsyncFileResponse` also:
```C++
server_.on("/download", HTTP_GET, [this](AsyncWebServerRequest *request)
{
    request->send(SD, "/path/to/file.json",  "application/json", true);
});
```

This has the same affect. Both of the above methods work if I send a file that is small (less than 5kb). Am I doing something wrong I see people and docs saying that they have tested with large files. I want the file to download from the browser which is why I set the `Content-Disposition` header in the first example and set `download` to `true` in the second. (I have tried setting `download` to `false` and that didn't make any difference.

av4625, 2022-12-22:
I have done some more research on this and found if I move my server calls out of the event loop handlers but keep the event loops running in the background, I still have the same issue.

But If I move my server calls out of the event loop handlers and also don't start the event loops, I am now able to download large files again.

This is very strange, I have lots of handlers on the server and these have all worked for a long time. Some of them get data from the SD card and I am able to download small files from the SD card (chunked) without issue while the event loops have been running in the background.

I tried putting the event loops at a lower priority (2) than the task that is created by AsyncTCP.

I should add that the event loops would be very close to "idle" while the server is running, an event would be sent just after the server starts and then again if a button is pressed which would stop the server and another event would be sent when the server stops.

I'm not sure why the presence of event loops would effect this?

av4625, 2022-12-22:
Thought I would add to this incase someone has a similar issue.

Looks like it was running out of RAM even though I had `149584` free heap 😱 

What I did was reduce the chunk size to `512`. This works consistently and only takes 2-3 seconds to download 300kb (when the ESP32 is a AP).

A chunk size of `1024` works most of the time but after a few downloads it hangs again, is something leaking somewhere?

I think that the `maxLen` param of the callback should take the available RAM into account, because for me and a few other issues I have seen it is allowing a chunk which just hangs the ESP32 (its quite far out as well, considering I have had to reduce the chunk by over 10 times), guessing how big of a chunk to use so you don't run out of what looks like a lot of RAM shouldn't be something the library user should need to do. It is also likely to change if I further develop the project and use a little more RAM. I will have to continually check if the chunk size needs reduced.

Code:
```C++
server_.on("/download", HTTP_GET, [this](AsyncWebServerRequest *request)
{
    File file{SD.open("/path/to/file.json", FILE_READ)};

    AsyncWebServerResponse *response = request->beginChunkedResponse(
        "application/json",
        [file](
            uint8_t* buffer,
            const size_t max_len,
            const size_t index) mutable -> size_t
        {
            size_t length;

            // Restrict chunk size so we don't run out of RAM
            static const size_t max_chunk{512};
            if (max_chunk < max_len)
            {
                length = file.read(buffer, max_chunk);
            }
            else
            {
                length = file.read(buffer, max_len);
            }

            if (length == 0)
            {
                file.close();
            }

            return length;
        });

    // Force download
    response->addHeader("Content-Disposition", "attachment; filename=\"file.json\"");
    request->send(response);
});
```

jordanfuerst, 2023-01-12:
I have been experiencing the exact same behavior. My current work-around has been to define maxLen to about 2k. still crashes occasionally, but I can live with it.

```
// Serve up our csv file that the graph page asks for
  server.on("/data", HTTP_GET, [](AsyncWebServerRequest *request)
            {
    AsyncWebServerResponse *response = request->beginChunkedResponse("text/plain", [](uint8_t *buffer, size_t maxLen, size_t index)-> size_t
    {
      size_t len=0;

      maxLen = maxLen>(2048+16) ? (2048):maxLen-16; // 16 bytes of "extra" space 

      while((len<(maxLen))  && history.tempCount>0)   // while there is room in the buffer and we have not sent it all
      {
        len+=snprintf((char*)buffer+len,8,"%hd,",(short)history.insideTemp.buffer[history.tempTail]); 
        len+=snprintf((char*)buffer+len,8,"%hd,",len); 
        ++history.tempTail &= histBufLen;
        --history.tempCount;
      }
      return len;
    });
    request->send(response); });
```

vortigont, 2023-02-02:
> maxLen = maxLen>(2048+16) ? (2048):maxLen-16; // 16 bytes of "extra" space

you have an error here. What if `maxLen` is less or equal to 16? You'll get a crash!

jordanfuerst, 2023-02-03:
True. But maxLen has never been less than 2K in all the testing I've ever done (always more) . If there is less than 16 bytes available, I can't send the data needed and returning 0 ends the process. There has to be at least 16 bytes available and at least in my testing it's always well above that. 

I'm starting to suspect that the bug is actually part of the webserver serving up .js and a .css file from flash in the background. If i move those elsewhere the crash stops.

vortigont, 2023-02-03:
@jordanfuerst but there is still a bug in your code anyway :) BTW, if you use only 2k out of 5k buffer you are just wasting memory that was allocated already.
I mentioned it because I actually had crashed like this long time ago with esp8266. Found that some times buffer len provided was much less than 1k, maybe not 16 bytes, but definitely less than 100-200 bytes as I remember.
Since my code was sending huge json there, I just set it to send a 1 byte space char if allocated buffer size was less than some predefined minimum that I need. With that approach I had no more crashes with chunked transfer.