me-no-dev/ESPAsyncWebServer #1254: Espressif SDK 2.0.6 seems to break AsyncWebServer operation (vs 2.0.4 & 2.0.5)

simkard69, 2022-12-23:
Dear Everyone,

I recently updated my Espressif SDK (in Arduino IDE 2.0.3) to latest 2.0.6 version and it seems that my AsyncWebServer is not working anymore.
Maybe it's due to AsyncWebServer directly ... or maybe it is more related to AsyncTCP library ... who knows // As I don't have time to investigate for now, I'll leave this issue opened here to hope for resolution.

My tests :
- Espressif SDK 2.0.4 : Working good
- Espressif SDK 2.0.5 : Working good
- Espressif SDK 2.0.6 : HTTP Error 404 while trying to browse main page on ESP32

Thanks

danbrummett, 2023-01-23:
I was also having a problem with the webserver simply failing to start and I was on framework-arduinoespressif32 (2.0.6) as you were.

I followed the directions to downgrade the framework version to 2.0.4 then did a clean all, then rebuild filesystem image, then build and upload my project.

Unfortunately even after downgrading I'm still not seeing the webserver start.

viktak, 2023-02-03:
for me it is working with this configuration in platformio.ini
```ini
[env:esp32dev]
platform = espressif32@3.0.0
board = esp32dev
framework = arduino
```

My problem is that I also want to use esp-now in the same project, and that does not work @3.0.0, only with the latest...

ivancmz, 2023-05-05:
Hi, any news on this, I've updated to 2.0.8 and it broke too