me-no-dev/ESPAsyncWebServer #1255: Running async tasks

VladoPortos, 2022-12-25:
Hello,
I know from the documentation I should not use delay() but what's the workaround... I have a POST event towards the ESP32 and I need a sequence of code to happen with various delays, but if it goes above 5s it will:

```
E (18170) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (18170) task_wdt:  - async_tcp (CPU 0/1)
E (18170) task_wdt: Tasks currently running:
E (18170) task_wdt: CPU 0: IDLE
E (18170) task_wdt: CPU 1: loopTask
E (18170) task_wdt: Aborting.
```

For example, I call some endpoint and need to have a separate "thread" if possible to do something... and if the endpoint is called the thread is killed and run again with new values...

Zhu-jiatong, 2023-01-02:
Maybe try using `std::thread` to start a new thread, where `delay()` would not be blocking other tasks.
Or you could try using the FreeRTOS functions.

Jakeduncan00, 2023-01-02:
if a predefined method returns a Task , you simply mark the calling method as async and put the await keyword in front of the method call. It's helpful to understand the control flow associated with the await keyword, but that's basically it.

Belleson, 2023-01-02:
I had a lot of trouble with watchdog timeout errors when I first created a web app with ESP32, PlatformIO and ESPAsyncWebServer library. The way I solved them was to avoid webserver.on(...) and instead run everything through web sockets. Then within the socket message callback handler, set (extern global) variables from web data, then trigger a Ticker timer to asynchronously process the data. So here is a socket message handler for a relay controller:

`
extern Preferences prefs; // Found in main.cpp
extern uint_fast16_t channel; // found in channel.cpp
extern uint_fast16_t relayYn; // found in channel.cpp
extern uint_fast16_t newBank; // found in channel.cpp
extern bool relayPulseTrigger;
extern uint_fast16_t capValue;
extern uint_fast16_t resValue;

AsyncWebSocket socket("/ws");
Ticker socketAction;

[...all the helper functions were here...]

void handleSocketMsg(void *arg, uint8_t *data, size_t len)
{
#if MONITOR_STATUS
	Serial.printf("\nENTERING handleSocketMessage()"); // WiFi.macAddress().c_str());
#endif

	AwsFrameInfo *info = (AwsFrameInfo *)arg;
	if (info->final && info->index == 0 && info->len == len && info->opcode == WS_TEXT)
	{
		const uint8_t size = JSON_OBJECT_SIZE(8);
		StaticJsonDocument<size> json;
		DeserializationError err = deserializeJson(json, data);
		if (err)
		{
			Serial.printf("\ndeserializeJson() failed with code %s", err.c_str());
			return;
		}

#if DEBUG_OUTPUT
		Serial.println("\nSocket message from client:\n");
		serializeJsonPretty(json, Serial);
#endif
		const char *action = json["action"];
		if (strcmp(action, "resetwifi") == 0)
		{
#if DEBUG_OUTPUT
			Serial.printf("\nPurging wifi credentials");
#endif
			wifiSaved = false;

			// AsyncTCP has a fast watchdog timer and will fail and reboot ESP32 before
			// some operations are complete. So set a Ticker to trigger a callback
			// some milliseconds later. This allows a socket message call to complete
			// and return before the watchdog barks and bites.

			socketAction.once_ms(1, wifiReset);
		}
		else if (strcmp(action, "setRly") == 0)
		{
			channel = json[CHANNEL];
			relayYn = json[RELAY];
			newBank = json[RELAY_BANK];
			socketAction.once_ms(1, channelSet);
		}
		else if (strcmp(action, "getChan") == 0)
		{
			// When a browser opens, it sends a message requesting the current channel
			// This replies back with it, so browser can set the channel knob to the
			// correct position
			socketAction.once_ms(1, sendChannel);
		}
		else if (strcmp(action, "pulse") == 0) // TODO get rid of this clause when pulse button is removed
		{
			relayPulseTrigger = true;
			socketAction.once_ms(1, sendPulse);
		}
		else if (strcmp(action, "setLoad") == 0)
		{
			if (strcmp(json["type"], CLOAD) == 0)
			{
				capValue = json["value"];
				socketAction.once_ms(1, setCLoad);
			}
			else //if (strcmp(json["type"], RLOAD) == 0)
			{
				resValue = json["value"];
				socketAction.once_ms(1, setRLoad);
			}
		}
		else if (strcmp(action, "getPhLd") == 0)
		{
			socketAction.once_ms(1, sendLoad);
		}
	}
}
`

The asynchronous nature of the whole thing causes me programming headaches because I'm not good at seeing every operation that can happen at once. So sometimes there are race conditions I have to debug. But those are problems I can solve, and the app stays functional for days with no reboots.

Belleson, 2023-01-02:
Also, it's worth noting that this code uses the same "socketAction" Ticker for all message processing, and any currently running Ticker stops and restarts when a new message retriggers the Ticker, So there will be times when you want to use multiple ticker instances.

Belleson, 2023-01-02:
And finally, here is corresponding javascript message processing code:

```
function onSocketMsg(evt) {
	var rotAng;
	var chNum;

	if (logStatus) console.log("\nENTERING rad.onSocketMsg()");

	let data = JSON.parse(evt.data);
	if (logStatus) console.log(data);

	if ("reloadLogin" in data) {
		reloadLoginPage();
	} else if ("chan" in data) {
		chNum = parseInt(data.chan);
		if (logStatus) console.log("Set channel in client(s)=" + chNum);

		rotAng = (chNum - 1.5) * 30;
		setKnobInd(rotAng, chNum, chInd, chIndShad);
	} else if ("capLd" in data) {
		rotAng = (parseInt(data.capLd) + 1) * 22.5 - 102.25;

		if (logStatus)
		{
			console.log("Setting Cload=" + parseInt(data.capLd) * 50);
			console.log("Ind " + capInd.id + " set rotAng=" + rotAng + " load=" + data.capLd);
		}
		setKnobInd(rotAng, parseInt(data.capLd), capInd, capIndShad);
	} else if ("resLd" in data) {
		rotAng = (parseInt(data.resLd) + 1) * 22.5 - 102.25;

		if (logStatus)
		{
			console.log("Setting Rload=" + parseInt(data.resLd) * 50);
			console.log("Ind " + resInd.id + " set rotAng=" + rotAng + " load=" + data.resLd);
		}

		setKnobInd(rotAng, parseInt(data.resLd), resInd, resIndShad);
	} else if (logStatus && "keepAlive" in data) {
		websocket.send(JSON.stringify({ action: clientID + " is alive" }));
	} else if ("chanIncr" in data) {
		chNum = parseInt(data.chanIncr);
		incrementChan(chNum);
	}
}

```

Jakeduncan00, 2023-01-03:
Let's take a look at a simple example of Task.Run, to get an idea of the syntax:

async void OnButtonClick()
{
  await Task.Run(() => /* your code here*/);
}
csharp
Task.Run accepts an Action (or a Func<T> in the event you need to return a value), so it is very flexible. You can write your code in line, e.g.:

await Task.Run(() => DoExpensiveOperation(someParameter));
csharp
...or within a block, e.g.:

await Task.Run(() => 
{
  for (int i = 0; i < int.MaxValue; i++)
  {
    ...
  }
});
csharp
In the case of a single method with no parameters, you simply pass the name of the method:

await Task.Run(MyMethod);