me-no-dev/ESPAsyncWebServer #1256: SPIFFS Editor file open error

mtibi75, 2022-12-25:
When the first load editor, it show the index.htm content. But i clicked other file, it not show file content, not delete, not download.
Browser console error: GET http://192.168.1.23/edit?edit=index.css 404 (Not Found). But, i open this link a new tab and modify the path, insert a "/" :http://192.168.1.23/edit?edit=/index.css, it shows file content. I use Arduino 1.8.19, LittleFs file system and the new 2.0.6 board SDK. Thanks for help.

andykaaa, 2022-12-28:
„When the first load editor, it show the index.htm content. But i clicked other file, it not show file content, not delete, not download.”

The same problem

Jakeduncan00, 2022-12-28:
The files you want to upload to the ESP32 filesystem should be placed in a folder called data under the project folder. ...
Then, select the newly created data folder and create the files you want to upload by clicking on the New File icon. ...
Write some random text inside that .

Jakeduncan00, 2022-12-29:
The files you want to upload to the ESP32 filesystem should be placed in a folder called data under the project folder. ...
Then, select the newly created data folder and create the files you want to upload by clicking on the New File icon. ...
Write some random text inside that