me-no-dev/ESPAsyncWebServer #1257: Use HTML colorpicker input in C code

WhatWouldJoeDo, 2022-12-29:
Hi guys, that's probably a no brainer but i'm really not into HTML and that stuff... Anyways, how can i use the input from a html color picker in my C code? I am using an ESP8266 as a server host and want to control single addressable LEDs (WS2812). The value from the color picker (HTML code) should be accessible in my C code which controls the LEDs.

The color values should be updated on change event or periodically.
Thanks for your help 🙏

Jakeduncan00, 2022-12-30:
The <input type="color"> defines a color picker. The default value is #000000 (black). The value must be in seven-character hexadecimal notation. Tip: Always add the <label> tag for best accessibility practices!

WhatWouldJoeDo, 2022-12-30:
Thanks @Jakeduncan00 for you answer. In fact i already have the color picker and i'm also able to display it's HEX value on the webpage (updated on change event). I'm still not sure how to use this value in my C code.

closed 2023-01-28 by WhatWouldJoeDo