me-no-dev/ESPAsyncWebServer #1262: Esp32 reboots after restarting the websocket client for a long number of times.

miguelfernando55, 2023-01-07:
Hi,

There are 2 cases that I believe they should need atention: 
1. The first is related to async web sockets. When reconnecting a certain long number of clients, the Esp32 reboots.

2. Async web server requests, when sending a long number of xmlhttp requests from the browser, the esp32 reboots.

I think that both are related to the AsyncTCP, because the given error is the same: [1045143][W][AsyncTCP.cpp:949] _poll(): pcb is NULL
The number inside the brackets keeps rising until the Esp32 reboots, by giving a Guru meditation error.

I am performing simple examples codes.

[image]

miguelfernando55, 2023-01-09:
After some debugging with Exception tool from me-no dev, it appears that the reason for the Guru meditation error is caused by: 
[image]

I already decreased the amount of server sends for 6 seconds interval, where each server send is sent with an interval of 100 ms from the previous one.

Still, after a long time, the AsyncTCP _poll error happens. And after a random long time, the esp32 reboots.

Jakeduncan00, 2023-01-10:
unsigned long currentMillis = millis();
// if WiFi is down, try reconnecting
if ((WiFi.status() != WL_CONNECTED) && (currentMillis - previousMillis >=interval)) {
  Serial.print(millis());
  Serial.println("Reconnecting to WiFi...");
  WiFi.disconnect();
  WiFi.reconnect();
  previousMillis = currentMillis;
}
Don’t forget to declare the previousMillis and interval variables. The interval corresponds to the period of time between each check in milliseconds (for example 30 seconds):

unsigned long previousMillis = 0;
unsigned long interval = 30000;

miguelfernando55, 2023-01-10:
Those logs are about the disconnection of the client.  The esp32 is connected and working.