me-no-dev/ESPAsyncWebServer #1266: [Qestion]: Usage of AsyncWebserver

hasenradball, 2023-01-15:
Hi,
I found you repo what looks quite interesting for me.
I would do the following and I am not sure if this is possible with you AysncWebserver.

What I would do is to use a LED Matrix (usage of PxMatrix lib) and show some information depending on 
what the ESP receives over wifi.

Lets say if a car is connected to the wallbox  it should show the actual charging state, power , etc...
If no car connected it should show the time or something else.

Is it possible to call a function if something is received via the wifi?
This function then takes care what the Matrix is showing. 

Best regards Frank

StefanoTesla, 2023-01-27:
It's the basic propouse of this library running a function when something is recived via wifi...

MacDada, 2023-03-12:
Yes.