me-no-dev/ESPAsyncWebServer #1271: AsyncWebSocket undefined reference to `SHA1Init'

luyang14, 2023-01-28:
`c:/users/iotts/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\nodemcu-32s\lib342\libESP Async WebServer.a(AsyncWebSocket.cpp.o):(.literal._ZN22AsyncWebSocketResponseC2ERK6StringP14AsyncWebSocket+0x10): undefined reference to `SHA1Init'
c:/users/iotts/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\nodemcu-32s\lib342\libESP Async WebServer.a(AsyncWebSocket.cpp.o):(.literal._ZN22AsyncWebSocketResponseC2ERK6StringP14AsyncWebSocket+0x18): undefined reference to `SHA1Update'
c:/users/iotts/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\nodemcu-32s\lib342\libESP Async WebServer.a(AsyncWebSocket.cpp.o):(.literal._ZN22AsyncWebSocketResponseC2ERK6StringP14AsyncWebSocket+0x1c): undefined reference to `SHA1Final'
c:/users/iotts/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\nodemcu-32s\lib342\libESP Async WebServer.a(AsyncWebSocket.cpp.o): in function `AsyncWebSocketResponse::AsyncWebSocketResponse(String const&, AsyncWebSocket*)':
G:\esp_program\esp32_websocket/.pio/libdeps/nodemcu-32s/ESP Async WebServer/src/AsyncWebSocket.cpp:1269: undefined reference to `SHA1Init'
c:/users/iotts/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: G:\esp_program\esp32_websocket/.pio/libdeps/nodemcu-32s/ESP Async WebServer/src/AsyncWebSocket.cpp:1270: undefined reference to `SHA1Update'
c:/users/iotts/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: G:\esp_program\esp32_websocket/.pio/libdeps/nodemcu-32s/ESP Async WebServer/src/AsyncWebSocket.cpp:1271: undefined reference to `SHA1Final'`
How can I solve this problem

BackInFiveMinutes, 2023-02-08:
I'm getting exactly the same issue.  I pared a sketch to the minimum and  this error occurs when the variable AsyncWebSocket ws("/ws"); is declared.  

Any help would be appriciated, and thanks for proving a great library.

nsbender, 2023-02-20:
Same issue here. I get the above warnings followed by this vague linker error.

`collect2.exe: error: ld returned 1 exit status`

Good catch. Took me a couple days to find this thread but Im glad I did. Too bad this library hasnt been updated in a year.

felixls, 2023-02-24:
guys, re-apply this one on master and it works
https://github.com/me-no-dev/ESPAsyncWebServer/pull/999

FreekBosPrive, 2023-08-02:
I'm also having the same issue...

mpearon, 2024-03-23:
> I'm also having the same issue... I fixed it by adding: lib_deps = https://github.com/me-no-dev/ESPAsyncWebServer.git
> 
> I needed to delete .pio folder, clean the project and restart VS Code. After that the project build successful

This was the fix for me!  Thank you, @FreekBosPrive!