me-no-dev/ESPAsyncWebServer #1273: something about url

yueda-han, 2023-01-29:
hi everyone:
I use the master branch of ESPAsyncWebServer,open simple_server.ino with ESP32 in Arduino.Everything is ok.
I change some code as follow:
````
server.on("/hello", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(200, "text/plain", "You will see Hello, world\r\n");
});

server.on("/hello/test", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send(200, "text/plain", "Hello, world, test\r\n");
});
````
When i wisit the url of "/hello/tets" from my broswer i can not see "Hello, world, test" How can i do? Or i have to change "/hello/test" to other one witout "/hello"?

Pablo2048, 2023-01-29:
:D just swap the definitions. It's because `/hello` matches `/hello/test` so the first definition is always fired...

yueda-han, 2023-01-30:
Thank you very much for your reply, solved the problem perfectly.

But I encountered another problem with URLs：
`server.on("/hello/test", HTTP_GET, [](AsyncWebServerRequest *request){
        request->send(200, "text/plain", "Hello, world, test\r\n");
    });

    server.on("/hello", HTTP_GET, [](AsyncWebServerRequest *request){
        request->send(200, "text/plain", "You will see Hello, world\r\n");
    });`
The  question is:
When I visit"/hello/testhello, return "You Will see Hello, world"     
When I visit"/hello/test/ha, return "Hello, world, test"       
---->>Actually, I don't define "/hello/tethello"  or "/hello/test/ha"
I mean, can URLs match them all ?

Pablo2048, 2023-01-30:
I don't think so, but you can use `onNotFound` handler and handle them by yourself.