me-no-dev/ESPAsyncWebServer #1275: 400 Bad Request Error for Large JSON Requests

ali80, 2023-02-01:
## Description

I am encountering a 400 Bad Request error for JSON requests that are larger than a certain size. The code never reaches the callback and it seems to throw 400 error before that, probably in the request JSON parsing step.

<details>
  <summary>Here is a sample JSON that causes the issue:</summary>
```json
  {
  "id": 0,
  "src": "50B4472BFE1F2AF5",
  "params": {
    "mid": "50B4472BFE1F2AF5",
    "t": "config",
    "rc": 4,
    "relays": [
      {
        "name": "out1",
        "thr": 25,
        "hys": 1,
        "ac": 2,
        "inv": 0,
        "ic": 15
      },
      {
        "name": "out2",
        "thr": 0,
        "hys": 0,
        "ac": 0,
        "inv": 0,
        "ic": 10
      },
      {
        "name": "out3",
        "thr": 0,
        "hys": 0,
        "ac": 0,
        "inv": 0,
        "ic": 8
      },
      {
        "name": "ou4",
        "thr": 0,
        "hys": 0,
        "ac": 0,
        "inv": 0,
        "ic": 12
      },
      {
        "name": "ou5",
        "thr": 0,
        "hys": 0,
        "ac": 0,
        "inv": 0,
        "ic": 12
      },
      {
        "name": "out6",
        "thr": 0,
        "hys": 0,
        "ac": 0,
        "inv": 0,
        "ic": 12
      },
      {
        "name": "out7",
        "thr": 0,
        "hys": 0,
        "ac": 0,
        "inv": 0,
        "ic": 12
      },
      {
        "name": "out8",
        "thr": 0,
        "hys": 0,
        "ac": 0,
        "inv": 0,
        "ic": 12
      }
    ],
    "sc": 1,
    "type": 7,
    "nodeName": "node",
    "v": "2.92+00",
    "network": 1,
    "tz_off": 12600
  }
}
```
</details>

The JSON is sent in compressed format and i**ts size is 613 bytes**. For smaller requests, there is no error.

## Code
Here is the relevant code:
```cpp
server.addHandler(Routes::rpc_post());
...
AsyncCallbackJsonWebHandler* Routes::rpc_post() { return _rpc_post; }

AsyncCallbackJsonWebHandler* _rpc_post
    = new AsyncCallbackJsonWebHandler(ROUTE_RPC, [](AsyncWebServerRequest* request, JsonVariant& json) {
          auto jsonObj = json.as<JsonObject>();
          AsyncJsonResponse* response = new AsyncJsonResponse(false, 2048);
          if (!response) {
              request->send(500, "application/json", "{\"error\":{\"message\":\"aloc error\"}}");
              log_c(2, MODULE "aloc error");
              return;
          }
          auto root = response->getRoot();
          root["res"] = "ok";
          response->setLength();
          request->send(response);
      });
```
## Error message
The error message is: "400 Bad Request Error"
## Expected outcome
I expect the JSON request to be processed successfully without encountering a 400 Bad Request error.

1cue-jrupp, 2023-04-19:
We're having the same issue. Are there any news on this?

ali80, 2024-05-16:
In AsyncCallbackJsonWebHandler, there are two main limits to be aware of:
1. HTTP Content Length: This defines the maximum size of the HTTP request content that the server will accept. The default value is 16384 bytes (16 KB).
2. JSON Buffer Size: This defines the maximum size of the JSON payload that can be parsed. The default value is 1024 bytes (1 KB), which might be too low for larger JSON requests.
To handle larger JSON payloads, you need to increase the JSON buffer size. Here’s how you can do it:
```cpp
const size_t jsonBufferSize = 4096;
AsyncCallbackJsonWebHandler* _rpc_post
    = new AsyncCallbackJsonWebHandler(ROUTE_RPC, [](AsyncWebServerRequest* request, JsonVariant& json) {
          auto jsonObj = json.as<JsonObject>();
          AsyncJsonResponse* response = new AsyncJsonResponse(false, 2048);
          if (!response) {
              request->send(500, "application/json", "{\"error\":{\"message\":\"aloc error\"}}");
              log_c(2, MODULE "aloc error");
              return;
          }
          auto root = response->getRoot();
          root["res"] = "ok";
          response->setLength();
          request->send(response);
      }, jsonBufferSize );
```

closed 2024-05-16 by ali80