me-no-dev/ESPAsyncWebServer #1279: Microcontroller crashing after few days

bruno-slv, 2023-02-08:
Hello, I've been using the library for about 4 months by now and I'm facing the same issue ever since: My MCU crashes after a few days of running code. I'm using a chinese ESP32 DevKit to watch the level of a water tank and send the current measurement to a web page with WebSockets. The main problem consists on the MCU crashing after 4 days executing the software, and it won't display my web page (stored on SPIFFS) ever again, only if I resend the code to the ESP.

I'm using Platform.IO with Visual Studio Code.

```
#include <Arduino.h>
#include <Arduino_JSON.h>
#include <ESPAsyncWebServer.h>
#include <ESPmDNS.h>
#include <SPIFFS.h>
#include <WiFi.h>

/**
 *
 * Indicador de Nível Control-Liq
 *
 * Autor: Bruno
 * Descrição: Correção do erro de reset da versão anterior
 * usando websockets ao invés de eventos enviados pelo servidor
 *
 */

/**
 * Definição das variáveis responsáveis por verificar falhas nos níveis
 */

bool sinalZero = false;
bool sinalEnviadoBaixo = false;
bool sinalEnviadoMedio = false;
bool sinalEnviadoAlto = false;
bool sinalEscape = false;

bool iniciarPulso = false;

bool falhaBaixo = false;
bool falhaMedio = false;
bool falhaBaixoMedio = false;
bool falhaAlto = false;
bool nivelAltoFalhaBaixo = false;
bool nivelMedioFalhaBaixo = false;
bool nivelAltoFalhaMedio = false;
bool nivelBaixoFalhaMedio = false;
bool falhaSomenteBaixo = false;
bool falhaSomenteMedio = false;

/**
 * Definição das variáveis do sinal sonoro:
 *
 * Tempo Atual
 * Tempo Anterior
 * Intervalo entre as ações
 */

unsigned long millisAtualPulso = 0;
unsigned long millisAnteriorPulso = 0;
unsigned long intervalo = 100;
int estadoSinaleiro = LOW;

/**
 * Definição de um IP fixo para maior comodidade
 * ao conectar-se ao microcontrolador
 */

IPAddress local_IP(192, 168, 0, 149);
IPAddress gateway(192, 168, 0, 1);
IPAddress subnet(255, 255, 0, 0);

/**
 * Credenciais de acesso ao WiFi
 */
[... 314 lines cut ...]
      falhaMedio && !falhaBaixoMedio) {
    ws.textAll("100");
    falhaBaixoMedio = false;
    falhaMedio = false;
    digitalWrite(ESCAPE, LOW);
  }

  // Falha sensores nivel baixo e medio
  if (sensor_alto_acionado && !sensor_medio_acionado &&
      !sensor_baixo_acionado && !falhaBaixoMedio) {
    ws.textAll("100 baixo_medio");
    falhaBaixoMedio = true;
  }

  if (!sensor_alto_acionado && !sensor_medio_acionado &&
      !sensor_baixo_acionado && falhaBaixoMedio) {
    ws.textAll("0");
    falhaBaixoMedio = false;
    falhaSomenteBaixo = false;
    falhaSomenteMedio = -false;
    digitalWrite(ESCAPE, HIGH);
  }
}

```

barbiani, 2023-02-08:
Keep an eye on the free heap size and save the crash logs.

Websockets and SSE are known to cause crashes. I'd change the code to use fetch/api.

bruno-slv, 2023-02-08:
I'm currently using the fetch API on the Javascript side of the project. I will be paying more attention the the free heap size

collin80, 2023-02-09:
This happens to me too. And, sure enough, I'm using websockets. It most certainly seems to be a heap issue for me. But, oddly enough, the actual heap has storage. I record a low heap over all time of about 25k with the current biggest chunk being about 57k. Now, that reading is a few seconds removed from the crash but still it seems to suggest that actual heap memory was available. I get crashes that are somewhat random but they still point to heap corruption. With heap poisoning turned on I can sometimes see the poison value getting used as an address during calculations which suggests there are out of bounds memory accesses somewhere. It really only happens with web sockets. I can make it better sometimes but it never goes away.

bruno-slv, 2023-02-09:
That doesn't sound the best for me, I probably should quit websockets for my application.

Obter o Outlook para Android<https://aka.ms/AAb9ysg>

closed 2023-02-15 by bruno-slv