me-no-dev/ESPAsyncWebServer #1280: using AsyncWebserver inside my class

tobiasfaust, 2023-02-10:
Hi, 
i want use th AsyncWebserver inside my class. But it dont works :(
My painpoint is here:
<code> this->webserver.on("/", HTTP_GET, this->PageIndex); </code>

Can anyone help me the figure out the right syntax?

<pre>
//
// A simple server implementation showing how to:
//  * serve static messages
//  * read GET and POST parameters
//  * handle missing pages / 404s
//

// derived from: https://github.com/me-no-dev/ESPAsyncWebServer/issues/755

#include <Arduino.h>
#ifdef ESP32
#include <WiFi.h>
#include <AsyncTCP.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#include <ESPAsyncTCP.h>
#endif
#include <ESPAsyncWebServer.h>

const char* ssid = "YOUR_SSID";
const char* password = "YOUR_PASSWORD";

const char* PARAM_MESSAGE = "message";

//##################################### Class definition

class WebHandler {
    public:
    WebHandler(AsyncWebServer & webserver);
    void begin(void);
    
    void PageIndex(AsyncWebServerRequest *request);
    
    private:
    AsyncWebServer webserver;

    void setHandlers(void);
};

WebHandler::WebHandler(AsyncWebServer & webserver) : webserver(webserver){}

void WebHandler::begin() {
    this->webserver.begin();
    this->setHandlers();
    this->webserver.rewrite("/", "/index.htm");
    
    Serial.println("Webhandler begin");

    this->webserver.begin();
}

void WebHandler::setHandlers() {
    Serial.println("SetHandler");
    
 //   this->webserver.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
 //       request->send(200, "text/plain", "Hello, world");
 //   });

    this->webserver.on("/", HTTP_GET, this->PageIndex);
    
    
}

void WebHandler::PageIndex(AsyncWebServerRequest *request) {
  request->send(200, "text/plain", "Hello, world");
}

//##################################### Class definition

AsyncWebServer webserver(80);
WebHandler webHandler(webserver);

void setup() {

    Serial.begin(115200);
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid, password);
    if (WiFi.waitForConnectResult() != WL_CONNECTED) {
        Serial.printf("WiFi Failed!\n");
        return;
    }

    Serial.print("IP Address: ");
    Serial.println(WiFi.localIP());
}

void loop() {
}
</pre>

`In member function 'void WebHandler::setHandlers()':
test_AsycWebServer_simple:61:54: error: invalid use of non-static member function 'void WebHandler::PageIndex(AsyncWebServerRequest*)'
     this->webserver.on("/", HTTP_GET, this->PageIndex);`

MENIER, 2023-02-10:
You can use lambda and this pointer :

tobiasfaust, 2023-02-10:
thanks for your response.
I found a solution too :)
<code>this->webserver.on("/", HTTP_GET, std::bind(&WebHandler::PageIndex, this, std::placeholders::_1));</code>

It works, but i dont know what "std::placeholders::_1" does exactly

well, its good to write about a problem to find a solution quickly by myself ;)

MENIER, 2023-02-10:
WebHandler is static?

MENIER, 2023-02-10:
Tnaks for your response too! You are Lucky you have juste one argument in your function to pass 6 arguments function The number of the placeholder is not here to match the position of the args, but rather to choose which arguments to send to the original function in which position so 
you must write:
```
this->onEvent(std::bind(&Derived::onEventDerived, this,
				std::placeholders::_1,
				std::placeholders::_2,
				std::placeholders::_3,
				std::placeholders::_4,
				std::placeholders::_5,
				std::placeholders::_6));
```

tobiasfaust, 2023-02-11:
thanks, it works perfectly :)

closed 2023-02-11 by tobiasfaust