me-no-dev/ESPAsyncWebServer #1282: Is this repo not supported anymore by anyone?

amid0, 2023-02-16:
Last update year ago.
Huge list of useful pull requests...

bmedici, 2023-02-24:
Seems dead :(

proddy, 2023-03-01:
yeah

MacDada, 2023-03-12:
I have the same question.

I could make a few PRs with fixes/features, but I will not waste my time, if there is nobody to merge/reject them…

MacDada, 2023-03-12:
@me-no-dev Is this repo [abandoned](https://en.wikipedia.org/wiki/Abandonware)? If so, could you at least mention that in the `README.md`?

me-no-dev, 2023-03-13:
It is in effect on hold. I was able to arrange some time for it last year, but unfortunately my health had another idea, so I spent the time in the hospital instead. Currently me and the rest of the ESP32 Arduino team are working on major upgrade to ESP-IDF 5.1. Once that is a reality, I plan to come back and rework many things to make them specifically more ESP32 and FreeRTOS friendly. Work and family are my priority, but management is willing to allow us time for personal project, as long as that does not interfere much with work schedule. This is as much as I can say on the subject. I love this project and hope to have it in decent state soon.

bmedici, 2023-03-13:
Thanks for this needed update for the project.
Wish you the best about what matters the most in life, and a speedy recovery as much as possible.
Thanks for the great contribution!

MacDada, 2023-03-13:
@me-no-dev Thank you for the answer and all the work you've done for this project and the ESP community in general!

Can we help somehow?

I'm a uC hobbyist, with not a biggest knowledge of C++, but in my journey I've found a few issues with the lib and I'm willing to help to improve it. Does it make sense to start with issues and maybe do some PRs?

Baptou88, 2023-03-13:
@me-no-dev hope your health improves

MassiPi, 2023-04-03:
> I spent the time in the hospital instead

so sad to know this.
Take care of yourself mate (but not too much, we need this to work nicely with esp32 also :)

ayaghini, 2023-04-15:
This is is a great project, wish you fast recovery and thanks for all you do!

MacDada, 2023-06-18:
up

MacDada, 2023-06-18:
> [STALE_SET] This issue has been automatically marked as stale because it has not had recent activity. It will be closed in 14 days if no further activity occurs. Thank you for your contributions.

Actually, you are stale!

> The stale app is deprecated and this repository is no longer maintained
> Please use [the stale action](https://github.com/actions/stale) instead.

https://github.com/apps/stale

zekageri, 2023-07-11:
@me-no-dev I hope you are doing well and your health is good.
With great respect, can we expect a continuation on this repo sometime in the near future?

DRSDavidSoft, 2024-09-06:
I seriously hope @me-no-dev is in good health and shape, and for sure, family and work comes before personal projects ❤️ 

With that being said, there are already great forks mentioned in this issue: #1017 and the most complete continuation seems to be this:

https://github.com/mathieucarbou/ESPAsyncWebServer

Which implemented many of the great missing features, while addressing all the annoying bugs. I've been using this for a while now, and I'm excited to see it working amazingly so far 😄 👍🏻