me-no-dev/ESPAsyncWebServer #1287: ESPAsyncWebServer -> ESP32's native IDF Http/s Server anyone?

proddy, 2023-03-01:
I've been using ESPAsyncWebServer and the sister AsyncTCP quite happily for a number of years now, first ESP8266 and now exclusively all variants of the ESP32. No issues. It's perfect.

But...with the ESP-IDF becoming more mature, has anyone successfully migrated over to the native [HTTP Server](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_http_server.html) library? There's an example of https [here](https://github.com/espressif/esp-idf/tree/master/examples/protocols/https_server/simple).

https://github.com/me-no-dev/ESPAsyncWebServer/issues/1282

proddy, 2023-03-01:
trawling through the issues I found this [post](https://github.com/me-no-dev/ESPAsyncWebServer/issues/1183#issuecomment-1426780734) by @justbendev recommending https://github.com/khoih-prog/HTTPS_Server_Generic which I'll check out.

dzungpv, 2023-10-16:
@proddy I have some projects to convert, do you found an easy way?

proddy, 2023-10-16:
> @proddy I have some projects to convert, do you found an easy way?

No I haven't. Still on my todo list.

proddy, 2023-12-08:
> @proddy check out https://github.com/hoeken/PsychicHttp
> 
> Its a wrapper for the ESP-IDF library with ESPAsyncWebserver syntax. Actively developed and reliable / stable (thanks Espressif!)

Thanks @hoeken - looks promising. I'll check it out for sure.