me-no-dev/ESPAsyncWebServer #1296: The program crashes when requesting serveStatic() files

butaikis, 2023-03-26:
Hi. I have a problem with ESPAsyncWebServer. Previously I used the arduino-esp32 framework in PlatformIO.

```
platform=espressif32
framework = arduino
```
Used the LittleFS file system, along with it

`WebServer.serveStatic("/", FS_WebUI,"/").setDefaultFile("index.html");`

Everything worked fine.

Then I needed to move the project to the framework
`framework = esp-idf, arduino`
_ESP-IDF ver 4.4.4_

Now an error occurs, LittleFS is mounted ok. But now the program crashes when requesting `serveStatic ` files
In the debugger here is this error:

```
Thread 
7 "async_tcp" received signal SIGTRAP, Trace/breakpoint trap.
AsyncWebServerRequest::_removeNotInterestingHeaders (this=0x3ffd6d8c) at lib/ESPAsyncWebServer-master/src/WebRequest.cpp:184
184	      if(!_interestingHeaders.containsIgnoreCase(header->name().c_str())){
```
I tried changing the file system to SPIFFS, the result is the same. 
Tell me maybe some configs affect my new **sdkconfig.h** ?

butaikis, 2023-03-30:
solving the problem
`CONFIG_HEAP_CORRUPTION_DETECTION` set `HEAP_POISONING_LIGHT`

closed 2023-03-30 by butaikis

StefanRvO, 2023-07-30:
I also ran into this issue, do you know why this config option is needed and why it makes a difference? It seems a bit strange to me.